esp8266/Arduino #7705: Unable to call String::indexOf with char* for a needle (implicit conversion)

paulocsanz, 2020-11-13:
Hey I'm trying to avoid allocating+copying to search for a needle in a haystack. So it would be great if String provided a indexOf method that receives a char*.

Ideally it would provide a method for __FlashStringHelper, but since it uses asm/intrinsics to search I don't know if there is a way to do the search without ignoring that it's from PROGMEM. Is it possible?

Anyway I'm happy with just char*.

paulocsanz, 2020-11-13:
I've went ahead and implemented it for `__FlashStringHelper*` too to prevent people from calling F() on the char* and actually converting it to String without realizing. So it prevents implicit allocation+copying.

paulocsanz, 2020-11-13:
I've went ahead and provided a `indexOf` for `__FlashStringHelper` pointers. And am calling `strstr_P` instead, as @earlephilhower showed me how.

closed 2020-11-13 by earlephilhower

paulocsanz, 2020-11-13:
@earlephilhower @d-a-v 

https://arduino-esp8266.readthedocs.io/en/latest/PROGMEM.html#functions-to-read-back-from-progmem

In the documentation (second paragraph of this section) it says it's UB to call `strstr_P` with a needle that's not from `PROGMEM`. So is the documentation wrong or is my code UB?

Should I restore the code to have the `strstr_P` path only for `__FlashStringHelper` and keep using `strstr` for the rest?

earlephilhower, 2020-11-13:
If the documentation says that, it's wrong and we should fix it.  Anything using PROGMEM is also safe to use RAM (but not vice-versa, of course!).  The ESP8266 isn't like the AVRs where reads from PROGMEM require a special instruction.  On the 8266 they just need to be 4-byte aligned and only 32-bits in size.  The `_P` calls handle this behind the scenes.

PR #7706 (2020-11-13): Provide String::indexOf for a char* needle
Fixes #7705