esp8266/Arduino #7718: ESP8266EX can support 32MB?

AMBULATUR, 2020-11-22:
Hello,recently i recieve these package [https://aliexpress.ru/item/1005001290915248.html](url)
Seller said that board have ESP8266EX + 32MB(M-Byte?) but in datasheet[https://www.espressif.com/sites/default/files/documentation/0a-esp8266ex_datasheet_en.pdf](url)
ESP8266EX can "uses external SPI flash to store user programs, and supports up to 16 MB
memory capacity theoretically."  
And under a magnifying glass I saw that i have 
[ESP8266EX ]						  [G AH1450 ]
|  462017  | + CH340G + flash Memory  |25Q32BSIG|        
[TUB0PGFX76]						  [	E5P376  ]

Datashit of flash memory says that it 32M-BIT not 32-M-Byte
Datasheet - [https://www.winbond.com/resource-files/w25q32fv%20revi%2010202015.pdf](url).

So, is i'm right or not and plate realy ESP8266 + 32MB(M-Byte)?

Which settings should i use? Is my settings right(sry 4 rus lang)?
[image]

d-a-v, 2020-11-22:
`esptool.py flash_id` will tell you the true size. It's probably 4MBytes = 32Mbits.
(I don't think 32MBytes flash chips are supported).

earlephilhower, 2020-11-22:
Good eyes!

10 seconds on a search engine shows that "25Q32BSIG" is a 32Mbit*, and hence 4M*byte* chip.  Nothing to do here, closing.

closed 2020-11-22 by earlephilhower

AMBULATUR, 2020-11-23:
By the way, sorry for being annoying, what am i doing wrong?
[image]
In code i set Serial.Begin(115200) as all guides request, but when i set in Serial Monitor baud to 115200 i receive wrong data, or nothing, but when i set 74880 baud in Serial Montior, and press "RST" button i get "Error?" and data goes correct, what does this error mean?

d-a-v, 2020-11-23:
ESP boot-up babbling works at 115200/40*26 baud.
Then in your sketch you can set 115200 and ESP will speak at 115200.
That is it for everybody.
If that's not the case for you, you can open a new issue and this time don't avoid to fill the issue template.