esp8266/Arduino #7719: ESP8266-WROOM-02D EEPROM issue

pravinzende12, 2020-11-24:
[latest flash erase problem files given by QDNET.zip](https://github.com/esp8266/Arduino/files/5588030/latest.flash.erase.problem.files.given.by.QDNET.zip)
[latest flash erase problem files for ESP8266.zip](https://github.com/esp8266/Arduino/files/5588040/latest.flash.erase.problem.files.for.ESP8266.zip)

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				ESP-WROOM-02D 
- Core Version:			[latest git hash or date]
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			ESP-WROOM-02D 
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

After writing on memory by SSID & Password for Wi-fi 
when there is power off module gets reset & flsh gets
erased with SSID & Password any one having idea for this
 

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

Code is attached 

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

devyte, 2020-11-24:
@pravinzende12 please don't attach zip files and add your MCVE directly to your post above.

pravinzende12, 2020-11-25:
ok attaching MCVE file

pravinzende12, 2020-11-25:
> ok attaching MCVE file

Tech-TX, 2020-11-28:
Two issues in 2 days with corrupt EEPROM on a WROOM-02D, and neither of them apparently read https://github.com/esp8266/Arduino/issues/6551

Sounds like a school project.

closed 2021-01-13 by earlephilhower