esp8266/Arduino #7720: EEPROM Issue

aldo4127, 2020-11-24:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:			ESP8266-WROOM-02D
- Core Version:			[latest git hash or date]
- Development Env:		Arduino IDE 1.8.12
- Operating System:		Windows 7

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			DOUT(compatible)
- Flash Size: 		        2MB
- lwip Variant:                   v2 Lower Memory
- Reset Method:		dtr(aka nodemcu)
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:                115200

### Problem Description

The following program will initialize a local IP in AP_MODE and will wait for credentials from the user. When received, the credentials will be saved in ESP using EEPROM.put function and will connect to the saved credentials in STA_MODE and will connect to the same credentials even after power reset.
After multiple resets, out of blue, the credentials were erased from EEPROM and whole 512 bytes were 0xFF (255).
This issue is coming randomly.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

#include <EEPROM.h>

#include <BearSSLHelpers.h>
#include <CertStoreBearSSL.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiAP.h>
#include <ESP8266WiFiGeneric.h>
#include <ESP8266WiFiGratuitous.h>
#include <ESP8266WiFiMulti.h>
#include <ESP8266WiFiScan.h>
#include <ESP8266WiFiSTA.h>
#include <ESP8266WiFiType.h>
#include <WiFiClient.h>
#include <WiFiClientSecure.h>
#include <WiFiClientSecureAxTLS.h>
#include <WiFiClientSecureBearSSL.h>
#include <WiFiServer.h>
#include <WiFiServerSecure.h>
#include <WiFiServerSecureAxTLS.h>
#include <WiFiServerSecureBearSSL.h>
#include <WiFiUdp.h>

#include <ArduinoJson.h>

void ConnectToRouter(void);
void ReadCredentialsFromEeprom(void);
void SetHotspot(void);
void WaitForCredentials(void);
void SetCredentialsToEeprom(void);

#define CREDENTIALS_FLAG               1
#define CREDENTIALS_START_ADDRESS      2

bool credentialsAvailableFlag = false;

typedef struct{
  char ssid[32];
  char password[64];
  char qrCode[20];
}UserCredentials_t;

UserCredentials_t wifiCredentials;

WiFiServer server(8080);      
WiFiClient client;

void setup()
{
  EEPROM.begin(512);                                                            //EEPROM initialized
  Serial.begin(11520);                                                          //Serial port for debug
  if(EEPROM.read(CREDENTIALS_FLAG) == 1)                                        //will read credentials flag at power on 
  {                                                                               //if 1 then credentials available
    credentialsAvailableFlag = true;                                              // if not 1 then credentials not available
  } 
  else
  {
    credentialsAvailableFlag = false;
  }
}

void loop()
{
  /*
   * if credentials available then read the credentials from EEPROM
   * if not available then initialize hotspot mode and wait for credentials from user through TCP
   */
  if(credentialsAvailableFlag)                    
  {
    ReadCredentialsFromEeprom();                                                
    ConnectToRouter();
  }
  else
  {
    SetHotspot();
    WaitForCredentials();
  }
}

void ConnectToRouter(void)
{
   WiFi.mode(WIFI_STA);                                           //will connect to the router whose credentials have been read from EEPROM
   WiFi.begin(wifiCredentials.ssid,wifiCredentials.password);
   Serial.print("Connecting to ");
   Serial.println(wifiCredentials.ssid);
}

void ReadCredentialsFromEeprom(void)
{
  EEPROM.get(CREDENTIALS_START_ADDRESS,wifiCredentials);    // reading credentials from EEPROM and savind it in struct
}

void SetHotspot(void)
{
    WiFi.mode(WIFI_AP);                       //Initialize Hotspot with crendentials as SSID: HotspotName | Paasword: password 
    IPAddress local_IP(192, 168, 1, 1);
    IPAddress gateway(192, 168, 1, 1);
    IPAddress subnet(255, 255, 255, 0);
    WiFi.softAPConfig (local_IP, gateway, subnet);
    WiFi.softAP("HotspotName", "password" , 1, false , 1);
    server.begin();
}

void WaitForCredentials(void)
{
  while(client.connected())     //if client connected at 192.168.1.1:8080
  {
    if (client.available())
    {
      String tempBuffWifiDirect = client.readStringUntil('\r\n\r\n');              //will read data printed on local IP
      Serial.println(tempBuffWifiDirect);
                                                                                // packet recieved {"SSID":"userssid","Password":"userpassword","QRcodeSerial":"12345"}
      StaticJsonBuffer<1000> jsonBuffer;                                          //parse the received JSON
      JsonObject& root = jsonBuffer.parseObject(tempBuffWifiDirect);
      
      if (root.success())          //if credentials JSON  parsing is Successful
      {
        const char* ssidTempBuff = "";
        const char* pwdTempBuff = "";
        const char* qrCodeSerial = "";
        ssidTempBuff = root["SSID"];
        pwdTempBuff = root["Password"];
        qrCodeSerial = root["QRcodeSerial"];
        Serial.println("Received Credentials:");
        Serial.println("SSID: " + (String)ssidTempBuff);
        Serial.println("Password: " + (String)pwdTempBuff);
        Serial.println("QRCode: " + (String)qrCodeSerial);
        memset(wifiCredentials.ssid, 0, sizeof(wifiCredentials.ssid));
        memset(wifiCredentials.password, 0, sizeof(wifiCredentials.password));
        memset(wifiCredentials.qrCode, 0 , sizeof(wifiCredentials.qrCode));
        strcpy(wifiCredentials.ssid, ssidTempBuff);
        strcpy(wifiCredentials.password, pwdTempBuff);
        strcpy(wifiCredentials.qrCode, qrCodeSerial);
        SetCredentialsToEeprom();              //save credentials to EEPROM
        credentialsAvailableFlag = true;        // set flad to read from EEPROM and connect to router
      }  
    }
  } 
}

void SetCredentialsToEeprom(void)
{
  EEPROM.put(CREDENTIALS_START_ADDRESS,wifiCredentials); //save credentials
  EEPROM.write(CREDENTIALS_FLAG,(int)1);                  //save credentials flag 1 so it connecets to router from next power on

  EEPROM.commit();    //commit to EEPROM
}

zztiger123, 2020-11-27:
Hi,
In this case, you may be trying to read and write EEPROM when the power supply is insufficient. Can you try to write the data you need into two different EEPORMs at the same time (this can prevent an error in one block from causing the entire program to be abnormal)

aldo4127, 2020-11-27:
@zztiger123 
Hi, I have tried writing the credentials in two different places, but when this issue happens everything gets erased to 255.

Tech-TX, 2020-11-28:
Two issues in 2 days with corrupt EEPROM on a WROOM-02D, and neither of them apparently read https://github.com/esp8266/Arduino/issues/6551 - try dropping the flash speed to 26MHz.

earlephilhower, 2021-01-13:
Dup of #7719.

closed 2021-01-13 by earlephilhower

pravinzende12, 2021-01-14:
> Dup of #7719.
Dear earlephilhower have you resolved problem?