esp8266/Arduino #7725: WIFI + AVRISP + avrdude: can't upload hex file

Saschko2000, 2020-11-27:
Hello

I am trying to use the sketch Arduino_Wifi_AVRISP, together with avrdude without success.

My setup:

Windows 10
Avrdude 6.3

Wemos D1 mini connected to Arduino Pro Mini:

ESP <-> ProMini
MISO (GPIO 12) - MISO (12)
MOSI (GPIO 13) - MOSI (11)
SCK (GPIO 14) - SCK (13)
RST (GPIO 5) - RST (RST)
3.3V - VCC
GND - GND

All copied in one directory.

Avrdude command:

avrdude -v -C avrdude.conf -c arduino -p atmega328 -P net:192.168.0.48:328 -U flash:w:firmware.hex:i

Result:

avrdude: Version 6.3, compiled on Feb 17 2016 at 09:25:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "avrdude.conf"

         Using Port                    : net:192.168.0.48:328
         Using Programmer              : arduino
avrdude: ser_drain(): read error: Falscher Parameter.

avrdude: ser_drain(): read error: Falscher Parameter.

avrdude: ser_drain(): read error: Falscher Parameter.

         AVR Part                      : ATmega328
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PC2
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
           lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14

avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
         Hardware Version: 4669920
         Firmware Version: 0.2
         Topcard         : STK502
         Vtarget         : 1.8 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10

avrdude done.  Thank you.

The serialoutput of the ESP:

[AVRISP] connection pending
[AVRISP] programming mode
[AVRISP] now idle

Any Ideas?

Thank you very much

   Sascha

Saschko2000, 2020-11-27:
Is it true, that "-P net" does not work with AVRDUDE under Windows?
Is that my Problem?

Tech-TX, 2020-11-28:
Presuming you're using a Pro Mini 3.3V (you didn't say) and [this](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266AVRISP/examples/Arduino_Wifi_AVRISP/Arduino_Wifi_AVRISP.ino), it looks like it ought to work.  I don't see anything set up incorrectly.
I haven't tried that before, but @d-a-v has experience with it.

The only note I see is here: https://www.nongnu.org/avrdude/user-manual/avrdude_4.html
>For programmers that attach to a serial port using some kind of higher level protocol (as opposed to bit-bang style programmers), port can be specified as net:host:port. In this case, instead of trying to open a local device, a TCP network connection to (TCP) port on host is established. The remote endpoint is assumed to be a terminal or console server that connects the network stream to a local serial port where the actual programmer has been attached to. The port is assumed to be properly configured, for example using a transparent 8-bit data connection without parity at 115200 Baud for a STK500.

>_This feature is currently not implemented for Win32 systems._

So, looks like the network port won't work with 32-bit Win10, but maybe will with 64-bit Win10

Saschko2000, 2020-11-28:
I use a Pro Mini 5V, but power it with 3.3V to VCC. I testet it with the PoluluUSB AVR Programmerv 2.1, and it worked flawlessly with 3.3V. hmmm... Can I test it in any other way?

Saschko2000, 2020-11-30:
I have tested it without any connection to the ProMini, The result is exact the same. :-(

Tech-TX, 2020-12-05:
Sorry, I can't help.  I haven't tried to use avrdude like that, I just use the AVRISP.  The only thing I could find on using that 'example' is this:
https://arduino.stackexchange.com/questions/44702/esp8266-as-isp-program-for-atmega16

My only suggestion is to try an old (pre-2017) version of avrdude to see if that works, based on the comment I see at stackexchange.  They may have done something with that stk500 driver that renders it inoperable with the example file.  Since avrdude isn't part of this repo, it's likely nobody here has tested it in quite a long time.

jandrassy, 2020-12-05:
so is
```
avrdude -v -C avrdude.conf -c stk500 -p atmega328 -P net:192.168.0.48:328 -U flash:w:firmware.hex:i
```
the solution?

Saschko2000, 2020-12-05:
OK, I tested it with stk500v1, stk500v2, stk500. All 3 the same result :-(

In the serial monitor I can see that the esp reacts to the command:

    [AVRISP] connection pending
    [AVRISP] programming mode
    [AVRISP] now idle

Here the Result of one test :

    d:\001\AVRDude>avrdude -v -C avrdude.conf -c stk500v1 -p atmega328 -P net:192.168.0.48:328 -U flash:w:firmware.hex:i
    
    avrdude: Version 6.3, compiled on Feb 17 2016 at 09:25:53
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch
    
             System wide configuration file is "avrdude.conf"
    
             Using Port                    : net:192.168.0.48:328
             Using Programmer              : stk500v1
    avrdude: ser_drain(): read error: Falscher Parameter.
    
    avrdude: ser_drain(): read error: Falscher Parameter.
    
    avrdude: ser_drain(): read error: Falscher Parameter.
    
             AVR Part                      : ATmega328
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PC2
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :
    
                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom        65    20     4    0 no       1024    4      0  3600  3600 0xff 0xff
               flash         65     6   128    0 yes     32768  128    256  4500  4500 0xff 0xff
               lfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               efuse          0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               lock           0     0     0    0 no          1    0      0  4500  4500 0x00 0x00
               calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
    
             Programmer Type : STK500
             Description     : Atmel STK500 Version 1.x firmware
    
    avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x14
    
    avrdude: stk500_getparm(): (a) protocol error, expect=0x14, resp=0x10
             Hardware Version: 4669920
             Firmware Version: 0.2
             Topcard         : STK502
             Vtarget         : 1.8 V
             Varef           : 0.0 V
             Oscillator      : Off
             SCK period      : 0.1 us
    
    avrdude: stk500_initialize(): (b) protocol error, expect=0x10, resp=0x01
    avrdude: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.
    
    avrdude: stk500_disable(): protocol error, expect=0x14, resp=0x10
    
    avrdude done.  Thank you.

Pablo2048, 2020-12-05:
-P net: is not supported in Windows system AFAIK - see https://www.nongnu.org/avrdude/user-manual/avrdude_4.html

jandrassy, 2020-12-05:
even on Linux you have to copy the "Arduino as ISP" upload avrdude command for ESP8266AVRISP, not the over bootloader upload command.
for upload over Serial and bootloader[ a simple telnet sketch](https://github.com/jandrassy/lab/blob/master/AvrDudeTelnet/AvrDudeTelnet.ino) on esp8266 will work. (from Linux because of the network pipe support for avrdude)