esp8266/Arduino #7734: The need to have a macro or method to obtain full path to source during runtime

WeaVaer, 2020-12-01:
Since the change in \_\_FILE\_\_ macro which doesn't contain path information anymore,
there is a need for a macro or method to obtain full path to source during runtime for various usecases.

Thus one of the methods below as a solution would be very useful and much apreciated and sought after.
- a \_\_PATH\_\_ macro that emits path information
- another macro that emits full path + filename as before
- another method that you will propose

Thanks in advance

earlephilhower, 2020-12-01:
There's no simple way of doing what you're asking as this is part of the GCC internals.  GCC adds internal special-case representations to its parse tree for these macros which makes it doable, but involving lots of little changes through the code and intermediate parser language.  There was a patch proposed by the Debian guys (I think) which would strip a user-supplied prefix (required so you can build binary-exact copies of code on different systems for reproducible builds) but it wasn't taken up and I think that project has their own fork of GCC to do it now.

As you asked in another closed issue, the sources/patches are all in the https://github.com/earlephilhower/esp-quick-toolchain repo.  You can drop the shortname patch and rebuild with a simple `make` on a Linux system for your specific use case.  Good luck!

closed 2020-12-01 by earlephilhower

WeaVaer, 2020-12-02:
Thanks a ton