esp8266/Arduino #7735: Any software implementation available for ESP8266 in Slave Mode?

rajhlinux, 2020-12-01:
ESP8266 has no hardware I2C, but few people says it can be done using software for the ESP8266 to be in Slave Mode.

I'm not trying to make this into another project, hoping someone has done this and there is some kind of "library" ready to be downloaded.

Thanks.

fsommer1968, 2020-12-01:
Hi  rajhlinux,

does this https://github.com/esp8266/Arduino/tree/master/libraries/Wire/examples/slave_sender
and this https://github.com/esp8266/Arduino/tree/master/libraries/Wire/examples/slave_receiver

help?

earlephilhower, 2020-12-01:
Thanks @fsommer1968.  There is an I2C library with slave mode implemented with examples.  It's SW based, so timings can't be too tight (i.e. expect errors at higher clock rates).

For help using the SW I2C library, this issue tracker is not appropriate.  Use https://esp8266.com or https://gitter.im/esp8266/Arduino .  Closing.

closed 2020-12-01 by earlephilhower

Tech-TX, 2020-12-07:
@rajhlinux due to the interrupt processing overhead in the SDK, the fastest a software I2C slave can run is approximately 50KHz, and you must compile with 160MHz CPU clock to get that.  The slave implementation is buggy.  I identified several problems with it, but haven't finished working out a solution.  Even a completely re-written slave cannot run faster than ~50KHz.