esp8266/Arduino #7742: `XCHAL_HAVE_ADDX` config is virtually ineffective against GCC 10.1 or later

jjsuwa-sys3175, 2020-12-04:
`XCHAL_HAVE_ADDX` == 1 in `xtensa-config.h` when building GCC, we expect that:
```
unsigned int addsubx_test(unsigned int a, unsigned int b) {
    return a * 2 + b;
}
```
will be compiled:
```
addsubx_test:
	addx2	a2, a2, a3
	ret.n
```
using GCC 4.8.2 (`https://github.com/earlephilhower/esp-quick-toolchain/releases/download/2.5.0-4/i686-w64-mingw32.xtensa-lx106-elf-b40a506.1563313032.zip`) satisfies the expectation.

but using 10.1.0 (`https://github.com/earlephilhower/esp-quick-toolchain/releases/download/3.0.0-gnu12/i686-w64-mingw32.xtensa-lx106-elf-0474ae9.200706.zip`) or 10.2.0 (`https://github.com/earlephilhower/esp-quick-toolchain/releases/download/3.0.0-gnu13/i686-w64-mingw32.xtensa-lx106-elf-2f311fd.200918.zip`), the expectation is betrayed:
```
addsubx_test:
	slli	a2, a2, 1
	add.n	a2, a2, a3
	ret.n
```
degradation is obvious.

it seems the RTL instruction combination pass does not consider the equivalence of "left-shift by a constant" and "multiplication with the constant power-of-2": a simple patch [attached](https://github.com/esp8266/Arduino/files/5640844/gcc-repair-TARGET_ADDX.zip) will solve that.

d-a-v, 2020-12-04:
Nice !
You should consider proposing your patch on [esp-quick-toolchain](https://github.com/earlephilhower/esp-quick-toolchain/tree/master/patches/gcc10.2) repository. 
At least.

jjsuwa-sys3175, 2020-12-04:
thanks for your reply, @d-a-v. i will do so later.

ps. i know some other inefficiencies for Xtensa GCC backend, eg. 64bit constant load always blindly emits a pair of `L32R`+litpool :)

closed 2020-12-08 by jjsuwa-sys3175