esp8266/Arduino #7743: HTTP UPDATE SWITCH RET NOT RETURNING

sebosfato, 2020-12-04:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

i have a question my http updates are working but i get no response for the sketck ok updated

i tried all here and there but it seems that is interfering with something 

when the sketch finish uploading the serial monitor prints some blank spaces...

i googled it but didnt found solution

problem is that i would like to update eprom before reset and is reseting before the switch ret so i have no chance to save..

Im not sure if is related but i updated the ide to the last version recently 

it was working two days ago but now it works only on spiffs i mean it print the switch ret 

anyone could help? 

i tried the example from the arduino and it does the same for me... it reboot before the switch ret

Please 

thanks

sebosfato, 2020-12-04:
I just managed to force the switch by seting ESPhttpUpdate.rebootOnUpdate(false);

but i would like to know what is happening since it was working well before... is there any chance it give me problems ahead?

im using this code 

 ESPhttpUpdate.rebootOnUpdate(false);

    Serial.println("Update Sketch...");
   WiFiClient client;
    ESPhttpUpdate.setLedPin(LED, LOW);
    t_httpUpdate_return Ret = ESPhttpUpdate.update(client, "updatelink");

    switch (Ret) {
      case HTTP_UPDATE_FAILED:
        Serial.printf("HTTP_UPDATE_FAILD Error (%d): %s", ESPhttpUpdate.getLastError(), ESPhttpUpdate.getLastErrorString().c_str());
        break;

      case HTTP_UPDATE_NO_UPDATES:
        Serial.println("HTTP_UPDATE_NO_UPDATES");
        break;

      case HTTP_UPDATE_OK:
          versao=versaofs;
          variavel="versao";
          valor=versao;
          POSTDATA();
          digitalWrite(LED, LOW);
          Serial.println("HTTP_UPDATE_OK");
          Serial.println("Updated Sketch ooooookkk...");
          EEPROM.put(198, versao);  // long - so 8 bytes (next address would be '12')
          digitalWrite(LED, LOW);
          // commit (write) the data to EEPROM - only actually writes if there has been a change
          boolean ok43 = EEPROM.commit(); Serial.println((ok43) ? "43 commit OK" : "Commit 43 failed");
           EEPROM.put(170, userid);  // long - so 8 bytes (next address would be '12')
          boolean ok49 = EEPROM.commit(); Serial.println((ok49) ? "43 commit OK" : "Commit 43 failed");
          Serial.println("HTTP_UPDATE_OK");
       //    delayMicroseconds(5000000);
       Serial.println("versao atualizada"+ String(versao));
        break;
 //ESP.restart();
          } } Serial.println("A versao era igual a fonte");
    }else{
      Serial.println("falhou");
 }

sebosfato, 2020-12-04:
can anyone telme if the lib changed to reboot automaticaly when sketch upload finish?

earlephilhower, 2020-12-06:
Closing due to template begin ignored, MCVE missing.  Use https://esp8266.com or https://gitter.im/esp8266/Arduino for general questions.

closed 2020-12-06 by earlephilhower