esp8266/Arduino #7744: Error message "fatal: not a git repository" when using the GIT snapshot

M-Reimer, 2020-12-06:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				other AI-Thinker ESP8266MOD
- Core Version:			47e7b2d
- Development Env:		Arduino IDE
- Operating System:		Arch Linux

### Settings in IDE

- Module:  			Wemos D1 mini r2
- Flash Mode:			?
- Flash Size: 		4MB/1MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		?
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   921600

### Problem Description

If I create a toolchain based on the GIT snapshot (ZIP file), then I always get the following message prior compiling:

    fatal: not a git repository (or any of the parent directories): .git

Reason is that this script tries to fetch some info using GIT:

https://github.com/esp8266/Arduino/blob/master/tools/makecorever.py

The message does not cause any problems. It just looks ugly.

How do you plan to fix this? Hide the message? I could create a pull request if I know how you want this to be fixed.

d-a-v, 2020-12-06:
Can you try with:
```
diff --git a/tools/makecorever.py b/tools/makecorever.py
index e7bd3486..659591b5 100755
--- a/tools/makecorever.py
+++ b/tools/makecorever.py
@@ -26,7 +26,7 @@ def generate(path, platform_path, git_ver="ffffffff", git_desc="unspecified"):
     def git(*args):
         cmd = ["git", "-C", platform_path]
         cmd.extend(args)
-        proc = subprocess.Popen(cmd, stdout=subprocess.PIPE, universal_newlines=True)
+        proc = subprocess.Popen(cmd, stdout=subprocess.PIPE, universal_newlines=True, stderr=DEVNULL)
         return proc.stdout.readlines()[0].strip()
 
     try:
```

If that suits your need, would you make a pull request ?

M-Reimer, 2020-12-06:
Done: https://github.com/esp8266/Arduino/pull/7745

d-a-v, 2020-12-10:
closed by #7745

commit 45664001 (2020-12-10): makecorever.py: Hide GIT error messages. Fixes #7744 (#7745)

* makecorever.py: Hide GIT error messages. Fixes #7744
closed 2020-12-10 by d-a-v