esp8266/Arduino #7746: LittleFS: code restarting by slow serial baudrate of 110

HakamSaffour, 2020-12-06:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:			[ESP-12]
- Core Version:			[framework-arduinoespressif8266 2.7.4]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[WeMos D1 R2 and mini]
- Flash Mode:			[dio]
- Flash Size: 		        [4MB]
- lwip Variant:                    [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[clk]
- Flash Frequency:	        [40Mhz]
- CPU Frequency:	        [80Mhz]
- Upload Using:		        [SERIAL]
- Upload Speed:                [115200] (serial upload only)

### Problem Description

I want to use LittleFS to save some data coming from serial port at baudrate 110, N81
I tested the provided example in:
https://github.com/esp8266/Arduino/tree/master/libraries/LittleFS/examples/LittleFS_Timestamp
 and it works fine at baudrate 115200
but when changing baudrate to 110, the code seems to retart (exception) 

I need to have a running FS at baudrate of 110, because I can not change the baudrate of the device I am connected to.

### Sketch

```cpp

https://github.com/esp8266/Arduino/blob/master/libraries/LittleFS/examples/LittleFS_Timestamp/LittleFS_Timestamp.ino

```

### Debug Messages

@ Baudrate 110
[image]

@Baudrate 115200
[image]

HakamSaffour, 2020-12-06:
add platformio.ini file

devyte, 2020-12-06:
Note: probably yield() needed in this while:
https://github.com/esp8266/Arduino/blob/8b662ed3b345d2ded3c2ac484949b03e10bdc43c/cores/esp8266/uart.cpp#L477

devyte, 2020-12-06:
or this one (might be lighter):
https://github.com/esp8266/Arduino/blob/8b662ed3b345d2ded3c2ac484949b03e10bdc43c/cores/esp8266/uart.cpp#L509

earlephilhower, 2020-12-07:
Adding `yield()s` in the UART code needs to be careful about IRQ and SYS context safety.  I believe both can perform UART writes.

HakamSaffour, 2020-12-07:
@devyte 
so you mean it is related to uart code?
why then the code restart at:

````cpp
if(!LittleFS.begin())
````

d-a-v, 2020-12-07:
@HakamSaffour
110 bauds is 11 printed chars per second.
`yield()` or `loop()` calling frequency must be much more higher than once per second.
You may try and add a call to `yield()` after every `Serial.print`.
Or you may try and use an alternative print function:
```
void myPrint (const String& toPrint) {
    for (int i = 0; i < toPrint.length(); i++) {
        Serial.print(toPrint[i]);
        yield();
    }
}

...
    myPrint("I hope this works better); // Serial.println("I hope this works");
```

HakamSaffour, 2020-12-07:
> 
> 
> @HakamSaffour
> 110 bauds is 11 printed chars per second.
> `yield()` or `loop()` calling frequency must be much more higher than once per second.
> You may try and add a call to `yield()` after every `Serial.print`.
> Or you may try and use an alternative print function:
> 
> ```
> void myPrint (const String& toPrint) {
>     for (int i = 0; i < toPrint.length(); i++) {
>         Serial.print(toPrint[i]);
>         yield();
>     }
> }
> 
> ...
>     myPrint("I hope this works better\n"); // Serial.println("I hope this works");
> ```

it seems to be working, no crash any more

d-a-v, 2020-12-07:
> But I need to write a wrapper for each print type, ex: Print, PrintIn, print(int, etc.)

True.

Would it be wise to do that in https://github.com/esp8266/Arduino/blob/8b662ed3b345d2ded3c2ac484949b03e10bdc43c/cores/esp8266/uart.cpp#L509

```cpp
    while (size--)
        uart_do_write_char(uart_nr, pgm_read_byte(buf++));
```
`=>`
```cpp
    if (can_yield() && 100*size/(baud/10) > 0) {
        // it takes more than 0.01s to print this string => yielding between chars
        // ( <=> (1000*size)/baud>0 <~> (size<<10)/baud > 0 )
        while (size--) {
            uart_do_write_char(uart_nr, pgm_read_byte(buf++));
            yield();
        }
    } else {
        while (size--) {
            uart_do_write_char(uart_nr, pgm_read_byte(buf++));
        }
    }
```

HakamSaffour, 2020-12-08:
source code of 'Espressif 8266' platfotm is not available in PlatformIO, I have to switch to Ardiono IDe to implement the above proposed changes

HakamSaffour, 2020-12-08:
I replaced 'baud' with 'baud_rate' but I did not find 'can_yield', any hint?

```CPP
if (can_yield() && 100*size/(uart->baud_rate/10) > 0) {
````

d-a-v, 2020-12-08:
> any hint?

In `coredecls.h`, sorry for the lack of clarity

HakamSaffour, 2020-12-09:
Hello @d-a-v 

Thank you for your solution.

I made the modifications in uart.cpp - as mentioned by you - and I tried this code

Sketch
```cpp
https://github.com/esp8266/Arduino/blob/master/libraries/LittleFS/examples/LittleFS_Timestamp/LittleFS_Timestamp.ino
````

and it looks to be working without any problem at baudrate = 110 and also at baud rate = 115200

[image]

shall I make a merge request to uart.cpp ?

earlephilhower, 2020-12-09:
@HakamSaffour sure, please throw a PR our way.  I'm not sure the exact changes, but this might also fix a problem w/MDNS debugging at 9600 baud (another issue open to document it) where I think it has the same overflow due to its chattiness.

Tech-TX, 2020-12-10:
Wow... welcome to the 1970s!  I haven't seen 110baud since I used an [ASR33 Teletype](https://en.wikipedia.org/wiki/Teletype_Model_33) over an acoustic-coupled modem time-sharing on an old Sperry Univac 9000 computer.

Two tin cans and a piece of string is a more efficient method of data transfer...  no wonder the issue never came up until now.

TD-er, 2020-12-10:
Last week I had an user report WDT crashes when running software serial at 1200 baud.
Maybe also add some similar fix there?

d-a-v, 2020-12-10:
> Maybe also add some similar fix there?

cc @dok-net

HakamSaffour, 2020-12-10:
A PR is created.

dok-net, 2020-12-29:
@d-a-v I've provided a ready-to-merge PR #7799 to fix this. Is should call yield() at a lower frequency in case where not only the low bitrate, but a large buffer is the culprit causing a long interval without wdt reset before. Please check it off if you agree.

PR #7799 (2020-12-31): UART write loop could lead to wdt with low bitrates or large buffers
EspSoftwareSerial successfully uses optimistic_yield for the same purpose already.
Fixes #7746
commit 84260dcb (2020-12-31): Write loop could lead to wdt with low bitrates or large buffers. Fixes #7746 (#7799)
closed 2020-12-31 by earlephilhower