esp8266/Arduino #7747: Question about ESP8266HTTPClient & direct access to Stream

enjoyneering, 2020-12-08:
Hi,

I made a simple ring buffer class and am unable to access the ESP8266HTTPClient stream in order to reduce the stream.read() overhead.  What am I doing wrong?

P.S. The ring buffer class function declaration - **uint16_t read(uint8_t *dstBuffer, uint16_t length);**

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266HTTPClient.h>

#include <RingBuffer.h>

#define RING_BUFFER_SIZE 256
#define URL_ADDRESS      "http://address"

RingBuffer ringBuffer(RING_BUFFER_SIZE);
WiFiClient client;
HTTPClient http;

void setup(){
  if (http.begin(client, URL_ADDRESS) == true)
  {
    int16_t httpCode = http.GET();

    if (httpCode == HTTP_CODE_OK)
    {
      Serial.println(F("ESP CLIENT..............connected to server"));
    }
    else
    {
      Serial.printf_P(PSTR("ESP CLIENT..............connection to server failed, %s\n"), http.errorToString(httpCode).c_str());
    }
  }
}

void loop() {
  WiFiClient *stream = http.getStreamPtr();

  readBytes = stream->available();

  readBytes = readBytes < ringBuffer.freeSpace() ? readBytes : ringBuffer.freeSpace();

  ringBuffer.write(reinterpret_cast<uint8_t*>(stream), readBytes); //read directly from HTTP client stream to ring buffer
}
```

earlephilhower, 2020-12-08:
The template was ignored and this is a “How do I?” question.  Closing.  Use esp8266.com for general things like this.

closed 2020-12-08 by earlephilhower

themindfactory, 2020-12-08:
"how do I" is for esp8266.com :-)