esp8266/Arduino #7749: Latest commit https://github.com/esp8266/Arduino/pull/7060 prevent compilation under windows with arduino IDE

luc-github, 2020-12-09:
- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ x] I have tested that the issue is present in current master branch (aka latest git).
- [x ] I have searched the issue tracker for a similar issue.
- [ NA] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12|
- Core Version:			latest git hash or date]https://github.com/esp8266/Arduino/pull/7060
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

- Module:  			Nodemcu
- Flash Mode:			dio
- Flash Size: 		4MB
- lwip Variant:   v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		160MHz
- Upload Using:		NA
- Upload Speed:   NA

### Problem Description

https://github.com/esp8266/Arduino/pull/7060 prevent compilation as it use mkdir.exe and cp.exe which do not exist under windows:  mkdir is part of Microsoft shell not and exe and cp is unix tool not present on windows

when linking the compilation failed with following error message 
```
exec: "mkdir": executable file not found in %PATH%
```
if mkdir is workrounded the a new message pop up
 
```
exec: "cp": executable file not found in %PATH%
```

[image]

Workaround 1:    
revert https://github.com/esp8266/Arduino/pull/7060 allows compillation

Workaround 2:   
 add mkdir.exe and cp.exe from coreutils (http://gnuwin32.sourceforge.net/packages/coreutils.htm)  allows compillation

earlephilhower, 2020-12-09:
@mhightower83, it should be a simple fix for this.  Use the same form of `cmd /c` wrappers used elsewhere in the `platform.txt` calls and it should resolve w/o much grief.

luc-github, 2020-12-09:
additonnaly is that normal the build pass in travis/ci where there is a window image used? 
is the ci reliable

earlephilhower, 2020-12-09:
Yeah, the Windows CI is behaving as expected.  Unfortunately, to run the CI GH needs certain tools installed.  I think they have almost a full mingw running (to allow their scripting/control/etc.).

luc-github, 2020-12-10:
I close issue as fixed - thank you

closed 2020-12-10 by luc-github