esp8266/Arduino #7751: The system_update_cpu_freq() function has no effect on ESP8266 2.7.4 core version

Th3maz1ng, 2020-12-10:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12F]
- Core Version:			[2843a5a core version 2.7.4 ]
- Development Env:		[Arduino IDE 1.8.13]
- Operating System:		[Windows & Ubuntu]

### Settings in IDE

- Module:  			[Nodemcu V1.0]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [921600]

### Problem Description

Hi, first of all thank you for your amazing work on this API.

I lately noticed that the **system_update_cpu_freq(uint8_t frequency)** function had no effect at all with the esp8266 core version 2.7.4.
It worked nicely with the version 2.6.3 of the core (I didn't test with the versions in-between though).

Thank you for your time.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include <user_interface.h>

unsigned long ts(0);

void setup() {
   Serial.begin(115200);
   Serial.printf("\n\nChange cpu freq : %d\n",system_update_cpu_freq(160));
}

void loop() 
{
  if(millis() - ts > 5000)
  {
    Serial.printf("Current freq : %u\n",ESP.getCpuFreqMHz());
    ts = millis();
  }
}

```

### Debug Messages
```
No error or debug messages even with core debug activated.
system_update_cpu_freq(160) returns 1 - success.
```

rndme, 2020-12-13:
Seems not to work on 2.7.1 or 2.7.0 either, but works on 2.6.3. It does work faster at 160 on 2.6.3, so I don't think it's a problem with getCpuFreqMHz()

Th3maz1ng, 2020-12-13:
Hi,
I don't think that it's a problem with getCpuFreqMHz() either because if I flash the esp with the 160MHz config, it reports just fine.
The issue is that I'm unable to change the frequency on the fly. Did the NONos SDK version used in the 2.7.0 version of the core change compared to the version used in the 2.6.3 version ?
Is there something I could look into in order to understand why system_update_cpu_freq has no effect ?
Thank you

devyte, 2020-12-13:
On the fly changing of the cpu freq is currently not supported in the core. There are Reasons for this.
There are some efforts to move in that direction, but there's nothing testable. 
Closing due to not supported.

closed 2020-12-13 by devyte