esp8266/Arduino #7759: in mDNS, board property is not correct and Arduino cannot display variant in description

luc-github, 2020-12-12:
- [x ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x ] I have tested that the issue is present in current master branch (aka latest git).
- x[ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			[latest git ]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Generic ESP8266 ]
- Flash Mode:			[dio]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[160MHz]
- Upload Using:		[OTA]
- Upload Speed:   [115200|] 

### Problem Description
OTA description is incomplete - variant is missing  in description in  Arduino OTA menu unlike for ESP32

[image]
it is due to property board which should contain the variant name, but ESP8266 put another string 
[image]
 string is put here (thanks @tobozo to find out)
https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/src/LEAmDNS.cpp#L1278
and also https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266mDNS/src/LEAmDNS2Host.cpp#L792
So arduino IDE cannot match the variant with one present in board.txt

if I hardcode "generic" for the variant on board property - the result is correct

[image]
[image]

mcspr, 2022-07-12:
Slight correction: "board" from mDNS (for some reason) should be equal to the ID of the board, not the whatever value we set to either board= or variant=
This is true for both the old IDE and the new one; see `arduino-cli board list` with mDNS discovery enabled
https://github.com/arduino/Arduino/blob/57a931c9c4c084057417837239ad8136f8a7b1aa/arduino-core/src/cc/arduino/packages/discoverers/NetworkDiscovery.java#L113-L119
https://github.com/arduino/arduino-cli/blob/63b53c0fa6830ce77cfb5508f06b2b4e601c90b5/cli/board/list.go#L134
(where identification prop map of this specific board is `{board: d1_mini}`)

For example, if we select `LOLIN(WEMOS) D1 R2 & mini`, mDNS should return `d1_mini`. No relation to the similarly named `build.variant=`, just the root of the property.
https://github.com/esp8266/Arduino/blob/a2982f96a49f40197a56207d6a6a0a4a34fcf90a/boards.txt#L3848

esp32 seems to use a different scheme btw, by introducing `ARDUINO_VARIANT` flag and using it's value

mcspr, 2022-07-13:
> (where identification prop map of this specific board is {board: d1_mini})

...which we could find out normally, without going too far into the arduino-cli code
```console
$ arduino-cli.exe board details -b $board_fqbn
```
for example, huzzah
```console
$ arduino-cli.exe board details -b esp8266com:esp8266:huzzah
Board name:                Adafruit Feather HUZZAH ESP8266
FQBN:                      esp8266com:esp8266:huzzah
Board version:             3.1.0-dev

Identification properties: board=huzzah
...
```

PR #8634 (2022-07-27): Board ID as build flag
Use `{_id}` in mDNS response instead of the `{build.board}`, adds extra build flag `-DARDUINO_BOARD_ID="{_id}"`

For example
```
$ arduino-cli.exe board list
Port          Protocol Type              Board Name                FQBN                       Core
xxx.xxx.x.xxx network  Network Port      LOLIN(WEMOS) D1 R2 & mini esp8266com:esp8266:d1_mini esp8266com:esp8266
```

resolve #7759
commit 06f34ed5 (2022-07-27): Board ID as build flag (#8634)

* Board ID as build flag

When using IDE or `arduino-cli board list`, show full board name and FQBN
```
$ arduino-cli.exe board list
Port          Protocol Type              Board Name                FQBN                       Core
xxx.xxx.x.xxx network  Network Port      LOLIN(WEMOS) D1 R2 & mini esp8266com:esp8266:d1_mini esp8266com:esp8266
```
Also add `description` field to the mDNS response containing the
original `{build.board}` string value.

resolve #7759

* does not work with recent ide (?)

* fixup! does not work with recent ide (?)
closed 2022-07-27 by d-a-v