esp8266/Arduino #7762: ESP8266 - WiFi connection with WPA2 Enterprise

ziogianni, 2020-12-13:
Hello everyone, I've been trying to establish a connection between a [NodeMCU 12E v3] and an AP relying on the WPA2 Enterprise authentication method, but unfortunately something is not working as it should.

I found two main issues:
- The First one is linked to a typo in the 2.2.1 Espressif SDK, in the API they forgot to replace the hard-coded Username used for their testing\validation purposes with a variable. Indeed, the API always sends to the RADIUS server the following user "anonymous@espressif.com" no matter what you specify with the method `wifi_station_set_enterprise_username()`.
A workaround for this is pretty easy, you need to add that specific Username in your RADIUS server.

 - Second issue, once the "anonymous@espressif.com" has been added in the RADIUS, the nodemcu is able to send a connection request, MSCHAPv2 challenge starts and it is finally accepted. 
Anyway after authentication occurs, nothing happens... it doesn't get an IP address and the connection is reset and start the whole process over like in a loop.

Looking at online resources, I got that this issue may be tied with older SDK releases like the 2.2.1 which should be provided in the current Arduino IDE Boards package. It's very likely that the EspressIf 3.0 SDK release solves the WPA2 authentication issue.

Below I'm going to add the sketch I used  for testing.
I also tried a similar implementation with ESP32, using proper ESP32 Arduino libraries (DOIT ESP32 DEVKIT V1), and it worked like a charm.

```
#include  <ESP8266WiFi.h>
#define STATION_IF      0x00

extern "C" {
#include "user_interface.h"
#include "wpa2_enterprise.h"
#include "c_types.h"
}

// SSID to connect to
char ssid[] = "your SSID";
char username[] = "anonymous@espressif.com";
char identity[] = "anonymous@espressif.com";
char password[] = "your password";

void setup() {

  WiFi.mode(WIFI_STA);
  Serial.begin(115200);
  delay(1000);
  Serial.setDebugOutput(true);
  Serial.printf("SDK version: %s\n", system_get_sdk_version());
  Serial.printf("Free Heap: %4d\n",ESP.getFreeHeap());

  // Setting ESP into STATION mode only (no AP mode or dual mode)
   wifi_set_opmode(STATION_MODE);

  struct station_config wifi_config;

  memset(&wifi_config, 0, sizeof(wifi_config));
  strcpy((char*)wifi_config.ssid, ssid);
  strcpy((char*)wifi_config.password, "router RADIUS password");
  

  wifi_station_set_wpa2_enterprise_auth(1);

  // Clean up to be sure no old data is still inside
  wifi_station_clear_cert_key();
  wifi_station_clear_enterprise_ca_cert();
  wifi_station_clear_enterprise_identity();
  wifi_station_clear_enterprise_username();
  wifi_station_clear_enterprise_password();
  wifi_station_clear_enterprise_new_password();

  wifi_station_set_enterprise_identity((uint8*)identity, strlen(identity));
  wifi_station_set_enterprise_username((uint8_t *)username, strlen(username));                
  wifi_station_set_enterprise_password((uint8_t *)password, strlen(password));
  
wifi_station_set_config(&wifi_config);
wifi_station_connect();

  while (WiFi.status() != WL_CONNECTED) {
    delay(5000);
    Serial.print(".");
    Serial.print(wifi_station_get_connect_status());
  }

  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
} 
```

devyte, 2020-12-20:
WPA2 is currently not supported.
Closing.

closed 2020-12-20 by devyte

hutje, 2021-02-01:
So WPA2 Enterprise is not being supported for esp8266?

d-a-v, 2021-02-01:
Currently not with Arduino. You may try with RTOS-SDK and native API.

ziogianni, 2021-02-01:
As far as I know Wpa2 Enterprise works if you use the Espressif SDK.

You can even achieve TLS auth with client/server certificates.

But the Arduino IDE board includes an outdated firmware version that
doesn't implement that.

hutje, 2021-02-01:
Will this work?
https://gist.github.com/Matheus-Garbelini/2cd780aed2eddbe17eb4adb5eca42bd6

naimulh247, 2022-09-06:
> Will this work? https://gist.github.com/Matheus-Garbelini/2cd780aed2eddbe17eb4adb5eca42bd6

Were you able to test if that works?

ziogianni, 2022-09-06:
As far as I remember, for certificate auth you need to convert certificates
as unsigned char arrays and then you must be sure that the server accept
TLS 1.0 with seclevel=1.

Server should need the same configuration also if you decide to use
credentials.