esp8266/Arduino #7765: It is a bad idea to allow using GPIO16/D0 as PWM.

rin67630, 2020-12-14:
With the Arduino IDE you can use ESP8266's GPIO16/D0 as PWM.
Expressif's IDE does not allow it. 

For a ggod reason: GPIO16 is part of the RTC section, so writing to it is much slower than writing to any other GPIO as the PWM core has to use API calls instead of GPOC/GPOS to set the pin high or low. 

Arduino compiled sketches using GPIO16/D0 as PWM may work, but:
-it will be much slower than other regular pins.
-unexpected resets and erratic behaviour are experienced randomly.
-the whole system slows down.

I have lost _months_ of weird debugging, until I came to the very problem of D0 not really being suitable for use as PWM.

IMHO It is a very bad idea to allow using GPIO16/D0 as PWM.
Please disable it or at least issue a _strong_ compiler warning.

Tech-TX, 2020-12-14:
Using the Board Manager version may have been a problem as that code was quite old, but the git version from at least April? has had improvements that should have worked OK with GPIO16.  I've been testing PWM on GPIO16 since yesterday morning, and once I cleaned up my source tree haven't seen any issues with it.  I wouldn't try running it over 25KHz, but it's working OK down in the range that most people will be using it.

earlephilhower, 2020-12-14:
Actually, the core doesn't need to use any calls to set the GPIO pins (see the 2 different versions in source).  It's just a memory-mapped register on the peripheral bus.  Slower than an internal core I/O register, yes, but not insanely so.

We're not aware of any issues other than the soft limit @Tech-TX mentioned above (25kHz is still above what the Espressif PWM implementation [which we do not use] allows IIRC).  In fact, we have two completely different PWM/waveform implementations in the core now to choose from, both enabling GPIO16 as a first-class citizen.

If you have any specific reproducible issue, please do file an issue with MCVE so we can look at it.  Again, please make sure you're using the latest GIT version as we've moved on significantly from 2.7.4 in this pre-release version.

Otherwise, since there's nothing concrete to look at or do here, closing.

closed 2020-12-14 by earlephilhower

rin67630, 2020-12-14:
I just wanted to help others, give a feed-back.
I will not dig into past erratic issues, I just can report that I have definitively got daily problems over months with the ESP8266 being not responsive for several seconds or refusing to OTA and now, that I run PWM at another pin than GPIO16, everything runs smoothly.
For my part, I can live with it...