esp8266/Arduino #7783: Daylight saving time zone adjust

Delanjp949, 2020-12-20:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.
I line in Adelaide Australia.  GMT +9:30 or GMT +10:30.  Setup as per intruction.  The clock at the moment is 1 hour fast.  Is there any way to adjust this.  Is there a Option available to add for this.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|Platformio|Make|other]
- Operating System:		[Windows|Ubuntu|MacOS]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}

```

### Debug Messages

```
Debug messages go here
```

d-a-v, 2020-12-20:
Check the [NTP-TZ-DST example](https://github.com/esp8266/Arduino/blob/f3521da173e7995f3d7e4a657fed5ab081de2ff6/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino#L42) available from the Arduino IDE (`Files>examples>esp8266>NTP-TZ-DST`).
Your timezone would be [`TZ_Australia_Adelaide`](https://github.com/esp8266/Arduino/blob/f3521da173e7995f3d7e4a657fed5ab081de2ff6/cores/esp8266/TZ.h#L319) (instead of `TZ_Europe_London` in the example).

devyte, 2020-12-20:
Closing due to not compliant.

closed 2020-12-20 by devyte