esp8266/Arduino #7792: time.cpp:27:14: fatal error: sys/_tz_structs.h: No such file or directory

TD-er, 2020-12-24:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12|ESP-01|ESP-07|ESP8285 device|other]
- Core Version:			[latest git]
- Development Env:		[Platformio]
- Operating System:		[Windows|Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module|Wemos D1 mini r2|Nodemcu|other]
- Flash Mode:			[dio]
- Flash Size: 		[4MB/1MB]
- lwip Variant:   [v1.4|v2 Lower Memory|Higher Bandwidth]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|160MHz]
- Upload Using:		[OTA|SERIAL]
- Upload Speed:   [115200|other] (serial upload only)

### Problem Description

In my builds where I use the latest git of esp8266/Arduino, I get this build error:

```
C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\time.cpp:27:14: fatal error: sys/_tz_structs.h: No such file or directory
```

This is when building in Windows as well as building in Linux.

d-a-v, 2020-12-24:
The toolchain needs an update
(it's `cd tools; ./get.py` when using git and the Arduino IDE)

mcspr, 2020-12-24:
> Development Env: [Platformio]

PIO needs to be explicitly told about the toolchain-xtensa package version when on `master` branch. This repo has no way of controlling it, since it is bound to the [platform-espressif8266](https://github.com/platformio/platform-espressif8266) config that locks the version :/

Some discussion here:
https://github.com/platformio/platformio-core/issues/3612
I'll publish the packages for my own repo some time today to be available via `mcspr/toolchain-xtensa`

Right now though it's something like
```ini
platform_packages = 
    framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
    toolchain-xtensa @ https://github.com/earlephilhower/esp-quick-toolchain/releases/download/3.0.0-newlib4.0.0-gnu21/x86_64-linux-gnu.xtensa-lx106-elf-dba3cc8.201223.tar.gz
```
(changes per platform, not really portable)

Documentation needs an update, both https://docs.platformio.org/en/latest/platforms/espressif8266.html and README. idk about referring to the `mcspr/...` package though, but that's the only way atm without pinging PIO team each toolchain version bump

Jason2866, 2020-12-24:
@mcspr  The Linux package seems to have wrong permissions in the package.
Platformio build fails

proddy, 2020-12-25:
same here
```
Tool Manager: Installing mcspr/toolchain-xtensa @ 5.100200.201223
Error: Could not find the package with 'mcspr/toolchain-xtensa @ 5.100200.201223' requirements for your system 'linux_x86_64'
```

Jason2866, 2020-12-25:
@proddy it is NOT a Arduino ESP8266 issue. It is a PlatformIO issue. mcspr already opened a issue in platformio github

earlephilhower, 2021-01-03:
I'm going to close this since it seems the consensus is it's a PIO issue and might already have been taken care of.  If there's a core-proper problem we can reopen and figure it out.

closed 2021-01-03 by earlephilhower

TD-er, 2021-01-03:
> and might already have been taken care of.

Not really....
But I will try to find the right issue for it.
Just to give an impression of "not being fixed":

```
In file included from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:39,
                 from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\Stream.cpp:23:
C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/twi.h:24:10: fatal error: C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h: No such file or directory
   24 | #include "Arduino.h"
      |          ^~~~~~~~~~~
compilation terminated.
In file included from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\StreamString.cpp:23:
C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:37:10: fatal error: binary.h: No such file or directory

****************************************************************
* Looking for binary.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:binary.h"
* Web  > https://platformio.org/lib/search?query=header:binary.h
*
****************************************************************

compilation terminated.
In file included from c:\github\td-er\espeasy\src\espeasy_common.h:35,
                 from src\src\Globals\../Helpers/Scheduler.h:4,
                 from src\src\Globals\ESPEasy_Scheduler.h:4,
                 from src\src\Globals\ESPEasy_Scheduler.cpp:1:
C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:37:10: fatal error: binary.h: No such file or directory

****************************************************************
* Looking for binary.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:binary.h"
* Web  > https://platformio.org/lib/search?query=header:binary.h
*
****************************************************************

compilation terminated.
In file included from c:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\libraries\esp8266sdfat\src\common\fsapiconstants.h:58,
                 from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266SdFat\src\ExFatLib\ExFatFile.h:36,
                 from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266SdFat\src\ExFatLib\upcase.h:27,
                 from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266SdFat\src\ExFatLib\upcase.cpp:25:
C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/FS.h:25:10: fatal error: C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h: No such file or directory
compilation terminated.
*** [.pio\build\test_beta_ESP8266_16M_LittleFS\FrameworkArduino\Stream.cpp.o] Error 1
*** [.pio\build\test_beta_ESP8266_16M_LittleFS\FrameworkArduino\StreamString.cpp.o] Error 1
In file included from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/ESP8266WiFi.h:39,
                 from src\src\Helpers\../../ESPEasy_common.h:82,
                 from src\src\Helpers\../../ESPEasy-Globals.h:6,
                 from src\src\Helpers\WebServer_commandHelper.cpp:3:
C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\libraries\ESP8266WiFi\src/WiFiClient.h:25:10: fatal error: C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h: No such file or directory
   25 | #include "Arduino.h"
      |          ^~~~~~~~~~~
compilation terminated.
In file included from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:285,
                 from c:\github\td-er\espeasy\src\espeasy_common.h:35,
                 from src\src\Globals\../Globals/../DataStructs/Caches.h:5,
                 from src\src\Globals\../Globals/Cache.h:4,
                 from src\src\Globals\Cache.cpp:1:
C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/WString.h:29:10: fatal error: C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/pgmspace.h: No such file or directory
   29 | #include <pgmspace.h>
      |          ^~~~~~~~~~~~
compilation terminated.
In file included from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\Updater.cpp:1:
C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\Updater.h:4:10: fatal error: Arduino.h: No such file or directory

*****************************************************************
* Looking for Arduino.h dependency? Check our library registry!
[... 89 lines cut ...]
* Web  > https://platformio.org/lib/search?query=header:mmu_iram.h
*
******************************************************************

compilation terminated.
compilation terminated.
compilation terminated.
compilation terminated.
compilation terminated.
*** [.pio\build\test_beta_ESP8266_16M_LittleFS\src\src\Globals\Cache.cpp.o] Error 1
*** [.pio\build\test_beta_ESP8266_16M_LittleFS\lib169\ESP8266SdFat\ExFatLib\upcase.cpp.o] Error 1
*** [.pio\build\test_beta_ESP8266_16M_LittleFS\FrameworkArduino\Updater.cpp.o] Error 1
*** [.pio\build\test_beta_ESP8266_16M_LittleFS\src\ESPEasy_common.cpp.o] Error 1
[... cut ...]
*** [.pio\build\test_beta_ESP8266_16M_LittleFS\src\src\ESPEasyCore\ESPEasyWiFi_credentials.cpp.o] Error 1
*** [.pio\build\test_beta_ESP8266_16M_LittleFS\lib9d7\ESP8266WiFi\ESP8266WiFiSTA.cpp.o] Error 1
xtensa-lx106-elf-g++: error: C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\WMath.cpp: No such file or directory
xtensa-lx106-elf-g++: fatal error: no input files
compilation terminated.
*** [.pio\build\test_beta_ESP8266_16M_LittleFS\FrameworkArduino\WMath.cpp.o] Error 1
```

earlephilhower, 2021-01-04:
Fair enough, thanks for the pointer for others.  But if you're getting something like...

````
In file included from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h:39,
                 from C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\Stream.cpp:23:
C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/twi.h:24:10: fatal error: C:\users\gijsn\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Arduino.h: No such file or directory
   24 | #include "Arduino.h"
      |          ^~~~~~~~~~~
````

then there's definitely nothing we can do here to fix that.  Hopefully PIO will be able to fix things...

maxgerhardt, 2021-01-30:
I can compile this repo in C++17 mode while when commenting out line 27 of the `time.cpp` with the offending `#include <sys/_tz_structs.h>` line just fine.

This include file is nowhere to be found in the Xtensa GCC 10.1.0. package. Since it still compiles it must have been moved into headers that are already included (otherwise there would be a build failure). 

So I guess the Arduino core shouldn't include a compiler and compiler-version specific / internal `_tz_structs.h` file? The underscore in the filename also makes it "reserved for internal use" doesn't it?  

Full `platformio.ini` for refernence, working on Windows 10, should work on all OSes.

```
[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
platform_packages = 
    toolchain-xtensa@~2.100100.0
    framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
```

Jason2866, 2021-01-30:
@maxgerhardt the toolchain provided from Platformio you mentioned is to old to be used with actual Arduino stage. The needed toolchain is here https://github.com/earlephilhower/esp-quick-toolchain/releases/tag/3.0.0-newlib4.0.0-gnu22

maxgerhardt, 2021-01-30:
Thanks, confirmed working with that direct toolchain and a new clean clone of the current repo.

```
[env:esp01_1m]
platform = espressif8266
board = esp01_1m
framework = arduino
; windows specific download link straight from git
platform_packages = 
    toolchain-xtensa@https://github.com/earlephilhower/esp-quick-toolchain/releases/download/3.0.0-newlib4.0.0-gnu22/x86_64-w64-mingw32.xtensa-lx106-elf-48f7b08.210128.zip
    framework-arduinoespressif8266 @ https://github.com/esp8266/Arduino.git
```

Since I did not see that a PlatformIO issue was opened between 24th December and now regarding the toolchain update, I opened it at https://github.com/platformio/platform-espressif8266/issues/242. 

The version in the current repo is the second-latest availalbe version in relation to the repo you posted above.

crwhite57, 2021-04-02:
> @proddy it is NOT a Arduino ESP8266 issue. It is a PlatformIO issue. mcspr already opened a issue in platformio github

I am using Arduino to program my esp8266 and I am getting the same thing so how is it not an Arduino ESP8266 issue?

Jason2866, 2021-04-03:
@crwhite57  you have to install as described here https://github.com/esp8266/Arduino#using-git-version

crwhite57, 2021-04-03:
So are you saying to install the latest version of Arduino? If so, I have the newest version you can get. If not, Please explain.

maxgerhardt, 2021-04-03:
In what sketch are you getting this error? Any, like also the blinky sketch? 

That error should only appear if you are using a very recent Arduino-ESP8266 core version but an older compiler that doesn't have this file (in the right place).

crwhite57, 2021-04-03:
I am getting in any sketch I do a check on. I can't find where to get that
file.

maxgerhardt, 2021-04-03:
And did you install Arduino-ESP8266 in the Arduino IDE via the [board manager URL](https://github.com/esp8266/Arduino#installing-with-boards-manager) or somehow else?

crwhite57, 2021-04-03:
Yes, I put that in and set my board to the lion Wemos d1 mini. I have also
tried it by setting it as a Wemos D1 R2.

d-a-v, 2021-04-03:
This issue is caused y a mismatching toolchain and source code.

There are several ways to install this core and the matching toolchain:
- arduino board manager will install the right toolchain
- git version gets its toolchain via the script `get.py` in `tools/`
- [PlatformIO](https://github.com/esp8266/Arduino#using-platformio)

Overwriting an installation with another one's source code _is unsupported and lead to issues_.
When using the Arduino IDE, ehe issue often disappears by completely removing the esp8266/arduino installation and restarting from scratch. By completely is meant that `arduino15/package/esp8266` directory must be is removed too.

crwhite57, 2021-04-03:
Ok, thanks.

calebleab, 2024-10-25:
I am having the same problem but i did note that i could not find the time.h file