esp8266/Arduino #7796: SPISlave.onData never get triggered

AnthonyPapageo, 2020-12-26:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ x] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ x] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[latest git hash or date]
- Development Env:		[Arduino IDE|]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio|dio|other]
- Flash Size: 		[4MB]
- lwip Variant:   [v1.4|v2 Lower Memory]
- Reset Method:		[ck|nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz|]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Detailed problem description goes here.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include "SPISlave.h"
volatile uint8_t interruptCounter = 0;
uint8_t data_to_send[4] = {110,24,30,40};

void ReceiveData(uint8_t * pdata, size_t len) 
{
  interruptCounter++;
}
void setup() {
  Serial.begin(74880);
  Serial.setDebugOutput(true);

  SPISlave.onData(ReceiveData);

  SPISlave.onDataSent([]() {
        Serial.println("Answer Sent");
    });

  // Setup SPI Slave registers and pins
  SPISlave.begin();

  SPISlave.setData(data_to_send, 4);
}

void loop() 
{
  Serial.print("interrupt counter : ");
  Serial.println(interruptCounter);
  delay(1000);
 }
```
Hi,

I'm using an ARM board  as a Spi master and i'm trying to communicate with a NodeMcu board configured as a slave. With a logic analyser I was able to see the communication working (the master produced a correct clock and the correct data was sent). From the ESP side, on the serial I'm able to see an "Answer sent" and check the data on the Logic Analyser but ``` interruptCounter``` always stays at 0 and I can't understand why. I've checked the wiring 10 times and as I said, the slave can correctly send data but ReceiveData never get called.

Thanks,

Anthony

AnthonyPapageo, 2020-12-27:
I've searched how this onData stuff work and I've gone through the source code, in ```hspi_slave.c``` I've added some counter to see where the code goes and where it doesn't. in ``` void ICACHE_RAM_ATTR _hspi_slave_isr_handler(void *arg) ``` the ``` rxtCounter``` never gets incremented.

 
```
void ICACHE_RAM_ATTR _hspi_slave_isr_handler(void *arg)
{
    uint32_t status;
    uint32_t istatus;
    istatus = SPIIR;

    if(istatus & (1 << SPII1)) { //SPI1 ISR
        status = SPI1S;
        SPI1S &= ~(0x3E0);//disable interrupts
        SPI1S |= SPISSRES;//reset
        SPI1S &= ~(0x1F);//clear interrupts
        SPI1S |= (0x3E0);//enable interrupts
        interruptCounter = 25;
        if((status & SPISRBIS) != 0 && (_hspi_slave_tx_data_cb)) {
            _hspi_slave_tx_data_cb(arg);
            txtCounter++;
        }
        if((status & SPISRSIS) != 0 && (_hspi_slave_tx_status_cb)) {
            _hspi_slave_tx_status_cb(arg);
        }
        if((status & SPISWSIS) != 0 && (_hspi_slave_rx_status_cb)) {
            uint32_t s = SPI1WS;
            _hspi_slave_rx_status_cb(arg, s);
             rxtCounter++;
        }
        if((status & SPISWBIS) != 0 && (_hspi_slave_rx_data_cb)) {
            uint8_t i;
            uint32_t data;
            _hspi_slave_buffer[32] = 0;
            for(i=0; i<8; i++) {
                data=SPI1W(i);
                _hspi_slave_buffer[i<<2] = data & 0xff;
                _hspi_slave_buffer[(i<<2)+1] = (data >> 8) & 0xff;
                _hspi_slave_buffer[(i<<2)+2] = (data >> 16) & 0xff;
                _hspi_slave_buffer[(i<<2)+3] = (data >> 24) & 0xff;
            }
            _hspi_slave_rx_data_cb(arg, &_hspi_slave_buffer[0], 32);
            rxtCounter++;
        }
    } else if(istatus & (1 << SPII0)) { //SPI0 ISR
        SPI0S &= ~(0x3ff);//clear SPI ISR
    } else if(istatus & (1 << SPII2)) {} //I2S ISR
}
```

Here is a example of a SPI transaction :
[image]

JiriBilek, 2020-12-30:
Could the problem be in incorrect usage of the proprietary SPI protocol used in ESP8266?
You sent the command **3**, that is receive data. For sending data you should use **2**. It's a long time I made a SPISlave communication application, so I might miss something, you can check the code at 
https://github.com/esp8266/Arduino/blob/a3a7a23a92508ce60c4ec0491dc3e0318978c8f5/libraries/SPISlave/examples/SPISlave_Master/SPISlave_Master.ino#L58

AnthonyPapageo, 2020-12-30:
Oh god I never knew there was some sort of command to send to start the communication. As I needed only a one way communication (master -> slave) to workaround I coded a simple soft spi reception. As it's near New Year's Eve I can't test that right now but I'll test it when I get my hands back on the system. If anyone got stuck in the same position that I was, here's my code that proved to work (I attached an interrupt on the SS pin). This works with a SPI (500kHz) mode 3 :

```
volatile int prv_CLK = 1;
volatile int prv2_CLK = 1;
volatile int prv_SI = 1;
volatile uint8_t received_data[4] = {0,0,0,0};
volatile uint8_t rec_data = 0;
volatile int bit_cnt = 0;
volatile int byte_cnt = 0;
void ICACHE_RAM_ATTR ReceiveData() 
{
  byte_cnt =0;
  do{
    do{
      do{
        prv2_CLK = prv_CLK;
        prv_SI = digitalRead(SIPin);
        prv_CLK = digitalRead(CLKPin);
        counter++;
      } 
      while(!((prv2_CLK == LOW) && (prv_CLK == HIGH)));
      rec_data |= prv_SI;
      bit_cnt++;
      if (bit_cnt<8)
        rec_data = rec_data << 1;
    } while(bit_cnt<8);
    received_data[byte_cnt] =rec_data;
    bit_cnt = 0;
    rec_data = 0;
    byte_cnt++;
  } while(byte_cnt <4);
  rec_data =0;
  bit_cnt =0;

} 
```

Thanks Jiri and Happy New Year !

yxf1234567, 2022-04-02:
Hello sir, my MCU master can send data to the NODEMCU slave, and the slave can receive it, but when the slave sends data to the master, it will be garbled, can you give me some help? I will be very grateful.