esp8266/Arduino #7798: Use IPv6 Router Advertisement options for DNS Configuration (RFC 8106)

MarcusWichelmann, 2020-12-28:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				ESP-12
- Core Version:			2.7.3-3-g2843a5ac
- Development Env:		Platformio
- Operating System:		Fedora

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Size: 		4MB
- lwip Variant:   IPv6+STABLE-2_1_2_RELEASE
- Reset Method:		manual
- Upload Using:		serial

### Problem Description

When connecting the ESP to an ipv6 enabled WiFi network with a router, that sends the RDNSS option in its router advertisements, the received DNS server is not respected and not visible when querying `WiFi.dnsIP()`. This feature is supported by `lwIP`, but seems to be disabled:
https://github.com/esp8266/Arduino/blob/5c29517d119393f3197871957ff18adbf918bac9/tools/sdk/lwip2/include/lwipopts.h#L2679

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>

void setup()
{
  Serial.begin(74880);
  Serial.setDebugOutput(ENABLE_DEBUG);
  Serial.println(ESP.getFullVersion());

  WiFi.mode(WIFI_STA);
  WiFi.hostname(SYSTEM_HOSTNAME);
  WiFi.begin(WIFI_STA_SSID, WIFI_STA_PASSWORD);

  // Wait for IPv4 & IPv6 addresses other than link-local
  bool hasV4 = false, hasv6 = false;
  while (!hasV4 || !hasv6)
  {
    for (auto entry : addrList)
    {
      IPAddress addr = entry.addr();
      if (addr.isLocal())
        continue;

      if (!hasV4)
        hasV4 = addr.isV4();
      if (!hasv6)
        hasv6 = addr.isV6();
    }

    Serial.print('.');
    delay(500);
  }

  Serial.println();

  // Print dns server list
  Serial.print("DNS servers:");
  for (int i = 0; i < DNS_MAX_SERVERS; i++)
  {
    IPAddress dns = WiFi.dnsIP(i);
    if (dns.isSet())
      Serial.printf(" %s", dns.toString().c_str());
  }
  Serial.println();
}

void loop()
{
}

```

### Debug Messages

```
SDK:2.2.2-dev(38a443e)/Core:2.7.3-3-g2843a5ac=20703003/lwIP:IPv6+STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
....
DNS servers: 192.168.11.1```

d-a-v, 2020-12-28:
Did you try the change ?
You can do it by yourself if you use the git version of this core with linux (osx, wsl2 maybe)
```
cd tools/sdk/lwip2
make download
vi builder/glue-lwip/arduino/lwipopts.h
make install
```

MarcusWichelmann, 2020-12-30:
Thank you for your response.
I just tried that, as you described, and it works. Now it shows me an IPv6 DNS server entry, but **only** that:

```
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3616, room 16 
tail 0
chksum 0x07
csum 0x07
va3a7a23a
~ld
SDK:2.2.2-dev(38a443e)/Core:2.7.2-207-ga3a7a23a=20702207/lwIP:IPv6+STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:28bebad
scandone
......scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 1
cnt 

connected with Wichelmann-IoT, channel 6
dhcp client start...
ip:192.168.11.153,mask:255.255.255.0,gw:192.168.11.1
.ip:192.168.11.153,mask:255.255.255.0,gw:192.168.11.1
ip:192.168.11.153,mask:255.255.255.0,gw:192.168.11.1
.
DNS servers: fdef:b:11::1
ip:192.168.11.153,mask:255.255.255.0,gw:192.168.11.1
ip:192.168.11.153,mask:255.255.255.0,gw:192.168.11.1
pm open,type:2 0
```

I've changed `DNS_MAX_SERVERS` to 4 and `LWIP_ND6_RDNSS_MAX_DNS_SERVERS` to `DNS_MAX_SERVERS`.

Do you have an idea, why the IPv4 DNS server thats received via DHCPv4 disappears in this case?

d-a-v, 2021-04-15:
We first should try with the latest version of lwIP (we use the the latest stable 2.1.2 and [latest git is stable too](http://lwip.100.n7.nabble.com/What-LwIP-version-to-use-td36095.html)).

d-a-v, 2022-06-11:
lwIP-2.1.3 is in master, @MarcusWichelmann did you make a try with it ?

MarcusWichelmann, 2022-06-12:
Thanks, will take a look when I have time, but not sure, when that will be.