esp8266/Arduino #7806: configTime() new signature does not influence mktime() under the tested circumstances

6v6gt-duino, 2021-01-03:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:			[ESP-12]
- Core Version:			[2.7.4]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Wemos D1 mini r2]
- Flash Mode:			[Unknown]
- Flash Size: 		        [4MB]
- lwip Variant:                   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	        [Unknown]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:               [921600] (serial upload only)

### Problem Description
0. configTime() new signature does not appear to influence mktime() 
1. I'm integrating an RTC module which will run in UTC. It needs "broken down" time (struct tm).
2. I need to get from a time_t value (UTC) to a tm struct (UTC) to set the RTC then, to read the RTC I need to convert
     from a tm struct (UTC) back to a time_t (UTC).
3. The first part is simply gmtime() and works. For the second part, there is no reverse of gmtime() such as say timegm() on this platform.
4. The gnu c library documentation suggests a work around of reverting the timezone/DST temporarily from local time to UTC,
     using mktime() to perform the tm to time_t conversion, then restoring the original timezone/DST.
5. This work around fails with either (a) setenv("TZ", "", 1) and tzset() together or (b) timeConfig( <new signature>),
   however, it works with timeConfig( <old signature> )

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

/*
   configTime_issue_v0_02

   new format configTime() signature fails to change the behaviour of mktime(). 
   The old format signature appear, however, to work

   OLD:
   void configTime(int timezone_sec, int daylightOffset_sec, const char* server1, const char* server2, const char* server3)

   NEW:
   void configTime(const char* tz, const char* server1, const char* server2, const char* server3)

   
   This example starts with a UTC time_t variable, converts it to a tm object in UTC time, then converts
   it back to a UTC time_t variable.  
   
   It does this twice. Once with with the new format configTime() signature, and once with the old
   and compares the results.

   Converting the tm object back to time_t is done using mktime(), but with attempts to temporarily set the
   timezone to UTC before hand.

   The only method which works for setting the TZ to UTC is configTime() with the "OLD" format signature.

   The behaviour is the same if I set up a WLAN connection and obtain the times from an NTP server.

   Bits from:

   https://github.com/SensorsIot/NTP-time-for-ESP8266-and-ESP32/blob/master/NTP_Example/NTP_Example.ino
   https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino
   http://www.catb.org/esr/time-programming/   (timegm() function)
   https://forum.arduino.cc/index.php?topic=719576.0

   Tested with Arduino ESP8266 2.7.4

   Version history (date format DD.MM.20YY) :
   v0_01  20.12.2020  comlete restructure.
   v0_02  20.12.2020  strip
*/

#include <ESP8266WiFi.h>
#include <time.h>
#include <sntp.h>       // sntp_servermode_dhcp() (from NTP-TZ-DST.ino )

/*******************************************************************************/

const char* NTP_SERVER = "ch.pool.ntp.org";  // *****set your own

// enter your time zone (https://remotemonitoringsystems.ca/time-zone-abbreviations.php)
const char* TZ_INFO    = "CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00";  // ***** set your own (not UTC!!)

/*******************************************************************************/

void setup() {

  Serial.begin(115200);
  Serial.println("\n\nNTP Time Test\n");

  WiFi.mode(WIFI_OFF);  

  setenv("TZ", TZ_INFO, 1);
  tzset();

  time_t timeNowUTC ;
  tm timeStruc ;
  char *tz ;
  time_t ret ;
  time_t sampleTime ;
  timeval tv;

  //sampleTime = 1590998400 ; // UTC  Monday, June 1, 2020 8:00:00 AM   (in DST)
  sampleTime = 1577865600 ; //  Wednesday, January 1, 2020 8:00:00 AM

  configTime( TZ_INFO, NTP_SERVER);  // new format

 // Part 1 ****************************************************************

  Serial.println("new ConfigTime() Signature") ;
  tv.tv_sec = sampleTime ;
  settimeofday(&tv, nullptr);
  timeNowUTC = time( nullptr ) ;
  Serial.printf( "time UTC (3) = %lu \n",  timeNowUTC ) ;

  // UTC time_t to tm struct  using gmtime()
  gmtime_r( &timeNowUTC, &timeStruc ) ;
  // showTime( timeStruc ) ;

  tz = getenv("TZ");  // use own buffer instead ??
  if (tz) {
    tz = strdup(tz);
    // configTime( 0 , 0, sntp_getservername(0), sntp_getservername(1), sntp_getservername(2) ) ; // Old format Works here
    configTime( "" , sntp_getservername(0), sntp_getservername(1), sntp_getservername(2) ) ; // New format fails to clean
    Serial.println( tz ) ;
  }
  else Serial.println( "TZ env var not set" ) ;
  
  // set TZ to UTC in preparation for  mktime()
  setenv("TZ", "", 1);
  tzset();

  // tm struct to time_t using mktime()
  ret = mktime( &timeStruc );

  // restore TZ to original
  if (tz) {
    configTime( tz , sntp_getservername(0), sntp_getservername(1), sntp_getservername(2) ) ; // v0_03
    setenv("TZ", tz, 1);
    free(tz);
  } else unsetenv("TZ");
  tzset();

  if ( ret == sampleTime ) Serial.println( "=== SUCCESS ===" ) ; 
  else Serial.println( "*** FAILURE ***" ) ; 
  Serial.printf( "time UTC (4) [should be the same as time UTC (3)] = %lu \n\n\n",  ret ) ;

  // Part 2 ****************************************************************

  Serial.println("OLD ConfigTime() Signature") ;
  tv.tv_sec = sampleTime ;
  settimeofday(&tv, nullptr);
  timeNowUTC = time( nullptr ) ;
  Serial.printf( "time UTC (3) = %lu \n",  timeNowUTC ) ;

  // UTC time_t to tm struct  using gmtime()
  gmtime_r( &timeNowUTC, &timeStruc ) ;
  // showTime( timeStruc ) ;

  tz = getenv("TZ");  // use own buffer instead ??
  if (tz) {
    tz = strdup(tz);
    configTime( 0 , 0, sntp_getservername(0), sntp_getservername(1), sntp_getservername(2) ) ; // Old format Works here
    // configTime( "" , sntp_getservername(0), sntp_getservername(1), sntp_getservername(2) ) ; // New format fails to clean
    Serial.println( tz ) ;
  }
  else Serial.println( "TZ env var not set" ) ;
  
  // set TZ to UTC in preparation for  mktime()
  setenv("TZ", "", 1);
  tzset();

  // tm struct to time_t using mktime()
  ret = mktime( &timeStruc );

  // restore TZ to original
  if (tz) {
    configTime( tz , sntp_getservername(0), sntp_getservername(1), sntp_getservername(2) ) ; // v0_03
    setenv("TZ", tz, 1);
    free(tz);
  } else unsetenv("TZ");
  tzset();

  if ( ret == sampleTime ) Serial.println( "=== SUCCESS ===" ) ; 
  else Serial.println( "*** FAILURE ***" ) ; 
  Serial.printf( "time UTC (4) [should be the same as time UTC (3)] = %lu \n\n\n",  ret ) ;
}

void loop() {
}

```

### Debug Messages

```
NTP Time Test

new ConfigTime() Signature
time UTC (3) = 1577865600 
CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00
*** FAILURE ***
time UTC (4) [should be the same as time UTC (3)] = 1577862000 

OLD ConfigTime() Signature
time UTC (3) = 1577865600 
CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00
=== SUCCESS ===
time UTC (4) [should be the same as time UTC (3)] = 1577865600 

```

d-a-v, 2021-01-04:
Here's my go:
```cpp

#include <ESP8266WiFi.h>
#include <time.h>
#include <sntp.h>       // sntp_servermode_dhcp() (from NTP-TZ-DST.ino )

#include <TZ.h>

/*******************************************************************************/

const char* NTP_SERVER = "ch.pool.ntp.org";  // *****set your own

// enter your time zone (https://remotemonitoringsystems.ca/time-zone-abbreviations.php)
//const char* TZ_INFO    = "CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00";  // ***** set your own (not UTC!!)

#define TZ_INFO "UTC-1"

/*******************************************************************************/

#define PTM(w) \
  Serial.printf(" " #w "=%d", (int)(tm->tm_##w));

void printTm(const char* what, const tm* tm) {
  Serial.print(what);
  PTM(isdst); PTM(yday); PTM(wday);
  PTM(year);  PTM(mon);  PTM(mday);
  PTM(hour);  PTM(min);  PTM(sec);
}

void test (int v)
{
  time_t timeNowUTC ;
  tm timeStruc ;
  char *tz ;
  time_t ret ;
  time_t sampleTime ;
  timeval tv;

  //sampleTime = 1590998400 ; // UTC  Monday, June 1, 2020 8:00:00 AM   (in DST)
  sampleTime = 1577865600 ; //  Wednesday, January 1, 2020 8:00:00 AM

  Serial.printf("--------------------\n\n%s ConfigTime() Signature\n", v==0? "Old ": "New ") ;
  tv.tv_sec = sampleTime ;
  tv.tv_usec = 0 ; // <----- I've had a hard time because this was missing
  settimeofday(&tv, nullptr);
  timeNowUTC = time( nullptr ) ;
  Serial.printf( "time UTC (1) = %llu - sampletime = %llu (diff=%lld)\n",  timeNowUTC, sampleTime, timeNowUTC - sampleTime) ;
  Serial.printf( "ctime(sampletime)=%s\n", ctime(&sampleTime));

  // UTC time_t to tm struct  using gmtime()
  gmtime_r( &timeNowUTC, &timeStruc ) ;
  printTm( "timeNowUTC", &timeStruc ) ;
  Serial.println();
  Serial.printf( "asctime(   gmtime(time UTC (1)))=%s\n", asctime(&timeStruc));
  Serial.printf( "asctime(localtime(time UTC (1)))=%s\n", asctime(localtime(&timeNowUTC)));

  tz = getenv("TZ");  // use own buffer instead ??
  if (tz) {
    tz = strdup(tz);
    if (v==0) configTime( 0 , 0, sntp_getservername(0), sntp_getservername(1), sntp_getservername(2) ) ; // Old format Works here
    else      configTime( "" , sntp_getservername(0), sntp_getservername(1), sntp_getservername(2) ) ; // New format fails to clean
    Serial.printf( "Now in UTC, TZ was '%s'\n", tz ) ;
  }
  else Serial.println( "TZ env var not set" ) ;
  
  // set TZ to UTC in preparation for  mktime()
  //setenv("TZ", "", 1);
  //tzset();
  // ^^^^^^^^^^^^ configTime is supposed to take care of this, at least the "new" one
  Serial.printf("irrelevant: between configTime() calls: TZ is '%s'\n", getenv("TZ"));

  // tm struct to time_t using mktime()
  ret = mktime( &timeStruc );

  // restore TZ to original
  if (tz) {
    configTime( tz , sntp_getservername(0), sntp_getservername(1), sntp_getservername(2) ) ; // v0_03
    //setenv("TZ", tz, 1);
    // ^^^^^^^^^^^^^^^^^^^ irrelevant, same reason
    free(tz);
  } else unsetenv("TZ");
  //tzset(); // irrelevant

  Serial.printf("now TZ is '%s'\n", getenv("TZ"));

  if ( ret == sampleTime ) Serial.println( "=== SUCCESS ===" ) ; 
  else Serial.println( "*** FAILURE ***" ) ; 
  Serial.printf( "time UTC (2) [should be the same as time UTC (1)] = %llu (diff=%lld)\n\n\n",  ret, sampleTime - ret ) ;

}

void setup() {

  Serial.begin(115200);
  Serial.println("\n\nNTP Time Test\n");

  WiFi.mode(WIFI_OFF);  

  setenv("TZ", TZ_INFO, 1);
  tzset();

  configTime( TZ_INFO, NTP_SERVER);  // new format

  // Part 1 ****************************************************************
  test(0);

  // Part 2 ****************************************************************
  test(1);
}

void loop() {
}
```

Note: I'm using current git master branch

```
01:18:46.677599802 SDK:2.2.2-dev(38a443e)/Core:2.7.2-211-g5b42e73b=20702211/lwIP:IPv6+STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:28bebad
01:18:46.687216713 NTP Time Test
01:18:46.701591130 
01:18:46.716476764 --------------------
01:18:46.724071794 Old  ConfigTime() Signature
01:18:46.732595311 time UTC (1) = 1577865600 - sampletime = 1577865600 (diff=0)
01:18:46.741415564 ctime(sampletime)=Wed Jan  1 09:00:00 2020
01:18:46.749082978 timeNowUTC isdst=0 yday=0 wday=3 year=120 mon=0 mday=1 hour=8 min=0 sec=0
01:18:46.757750064 asctime(   gmtime(time UTC (1)))=Wed Jan  1 08:00:00 2020
01:18:46.766628308 asctime(localtime(time UTC (1)))=Wed Jan  1 09:00:00 2020
01:18:46.775577102 Now in UTC, TZ was 'UTC-1'
01:18:46.783872984 irrelevant: between configTime() calls: TZ is 'UTC-1'
01:18:46.791544797 now TZ is 'UTC-1'
01:18:46.798680234 === SUCCESS ===
01:18:46.806532325 time UTC (2) [should be the same as time UTC (1)] = 1577865600 (diff=0)
01:18:46.813444577 --------------------
01:18:46.820559621 New  ConfigTime() Signature
01:18:46.828045348 time UTC (1) = 1577865600 - sampletime = 1577865600 (diff=0)
01:18:46.836425935 ctime(sampletime)=Wed Jan  1 08:00:00 2020
01:18:46.843248407 timeNowUTC isdst=0 yday=0 wday=3 year=120 mon=0 mday=1 hour=8 min=0 sec=0
01:18:46.850202913 asctime(   gmtime(time UTC (1)))=Wed Jan  1 08:00:00 2020
01:18:46.857390269 asctime(localtime(time UTC (1)))=Wed Jan  1 08:00:00 2020
01:18:46.865704729 Now in UTC, TZ was 'UTC-1'
01:18:46.872741502 irrelevant: between configTime() calls: TZ is ''
01:18:46.879504827 now TZ is 'UTC-1'
01:18:46.886802673 === SUCCESS ===
01:18:46.893296954 time UTC (2) [should be the same as time UTC (1)] = 1577865600 (diff=0)
```

6v6gt-duino, 2021-01-04:
Thanks for getting back so quickly with your test results and suggestion.
As it is written, it appears at first sight to work. The presentation of the 64bit numbers is not too pretty (I guess that could be  because I'm using 2.7.4 and you made an explicit reference to the current git master branch ) but that is a cosmetic issue. 

However, if the "new format configTime()" part of the test is run on its own , that is test(0) is commented out in Setup(), then it fails. Clearly, the test(1) is dependent on some residual environment, side effects etc., left by test(0). I'll try by a binary chop and trial and error to isolate it. It took me some time to get this far so I am taking a break now. 
I'm keen, however, to get this working in a reliable way so I can cut a huge amount of code out of an existing application which uses the Arduino TimeLib.h, TimeZone.h, some big lump of UDP code for talking to an NTP server etc.

6v6gt-duino, 2021-01-06:
I've got a bit further in demonstrating the issue but it took some time because a second issue has appeared in the meantime. So it is not only the matter of mktime() not respecting the change of the timezone to UTC, but also that, after restoring the timezone, localtime() may not work correctly. However, I have developed a sketch which demonstrates these issues and a "solution". The "solution" is to use both types of configTime() signatures.
I've had to slightly restructure the sketch because it is not possible to combine everything in one run due to mutual interference between the test cases. The instructions are at the top of the sketch. Basically, you have to enter the test number (1, 2, or 3) in the serial monitor, restarting the ESP8266 module between tests. Only test 3 appears to work.

``` 
/*  configTimeIssue.ino 7806
 *  version 06-01-2021  (tested with 2.7.4)
 *
 *  https://github.com/esp8266/Arduino/issues/7806  // issue report
 *
 *  Enter Test Mode in serial monitor
 *
 *  Test 1 - Pure old format configTime() signature
 *    Result: mktime appears to work but old time zone not restored properly so localtime()
 *      appears to treat the current time zone as UTC
 *
 *  Test 2 - Pure new format configTime() signature
 *    Result: mktime fails to accept the prior timezone change to UTC work but old time zone is
 *      restored properly so localtime() does appear to work afterwards.
 *
 *  Test 3 - Hybrid, using both formats of the configTime() signature
 *    Result: this appears to work for the applied tests.
 *
 *
 */

#include "Arduino.h"
#include <ESP8266WiFi.h>
#include <time.h>
#include <sntp.h>       // sntp_servermode_dhcp() (from NTP-TZ-DST.ino )

#include <TZ.h>

/*******************************************************************************/

const char* NTP_SERVER = "ch.pool.ntp.org";  // *****set your own

// enter your time zone (https://remotemonitoringsystems.ca/time-zone-abbreviations.php)
//const char* TZ_INFO    = "CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00";  // ***** set your own (not UTC!!)

#define TZ_INFO "UTC-1"

/*******************************************************************************/

#define PTM(w) \
  Serial.printf(" " #w "=%d", (int)(tm->tm_##w));

void printTm(const char* what, const tm* tm) {
  Serial.print(what);
  PTM(isdst); PTM(yday); PTM(wday);
  PTM(year);  PTM(mon);  PTM(mday);
  PTM(hour);  PTM(min);  PTM(sec);
}

void test ( char v)
{
  time_t timeNowUTC ;
  tm timeStruc ;
  char *tz ;
  time_t ret ;
  time_t sampleTime ;
  timeval tv;

  //sampleTime = 1590998400 ; // UTC  Monday, June 1, 2020 8:00:00 AM   (in DST)
  sampleTime = 1577865600 ; //  Wednesday, January 1, 2020 8:00:00 AM

  Serial.printf("--------------------\n\nConfigTime() Signature Test #%c\n", v) ;
  tv.tv_sec = sampleTime ;
  settimeofday(&tv, nullptr);
  tv.tv_usec = 0 ; // <----- I've had a hard time because this was missing [ OP / sorry !]
  timeNowUTC = time( nullptr ) ;
[... 106 lines cut ...]
>>Send to COM14: "3"<<
Test 3 in progress--------------------

ConfigTime() Signature Test #3
time UTC (1) = 1577865600 - sampletime = 1577865600 (diff=0)
ctime(sampletime)=Wed Jan  1 09:00:00 2020

timeNowUTC isdst=0 yday=0 wday=3 year=120 mon=0 mday=1 hour=8 min=0 sec=0
asctime(   gmtime(time UTC (1)))=Wed Jan  1 08:00:00 2020

asctime(localtime(time UTC (1)))=Wed Jan  1 09:00:00 2020

Now in UTC, TZ was 'UTC-1'
irrelevant: between configTime() calls: TZ is ''
now TZ is 'UTC-1'
=== SUCCESS ===                                                  <<<<<<<<<<<<<<<<<<<<<<<<<<<<
time UTC (2) [should be the same as time UTC (1)] = 1577865600 (diff=0)

time_t timeNowUTC = 1577865601
asctime(   gmtime(time UTC (1))) = Wed Jan  1 08:00:00 2020

asctime(localtime(time UTC (1))) = Wed Jan  1 09:00:01 2020            <<<<<<<<<<<<<<<<<<<<<<<<<<<<<

ctime(&timeNowUTC) = Wed Jan  1 09:00:01 2020
```

6v6gt-duino, 2021-01-07:
In the meantime, I have developed another workaround for my particular application which does not rely on the "on the fly" changing of time zones to use mktime in a "time zone neutral" way. The time zone switch method is recommended here, in the GNU c library documentation, for platforms which don't support timegm() https://www.gnu.org/software/libc/manual/html_node/Broken_002ddown-Time.html with the quote :

´´´
Portability note: mktime is essentially universally available. timegm is rather rare. For the most portable conversion from a UTC broken-down time to a simple time, set the TZ environment variable to UTC, call mktime, then set TZ back.
´´´ 

Anyway, it looks like switching time zones risks bringing its own problems so I am, at the moment, considering solving my particular problem with a "home made" implementation of timegm(), mainly copied from the Arduino TimeLib.h with a few adjustments for the different epoch dates, enumerations of array items etc. for c time library time.h compatibility. I've added it below incase anyone finds it useful in a similar situation.

That means that I don't particularly need to keep this issue ticket open but I also imagine that the issues raised should be addressed anyway. 

```
void timegm_rA( time_t *tOut, const struct tm *tme) {

	/*
	 *  void timegm_rA( time_t *tOut, const struct tm *tme)
	 *
	 *  arguments:
	 *  tOut : output into reference to time_t
	 *  tme  : input struct reference (broken down time)
	 *
	 *  converts struct tme (broken down time) into time_t (seconds from 01.01.1970)
	 *  Returns -1 if tme represents a date between 01.01.1900 upto (but not including) 01.01.1970
	 *    which is supported by tm but not time_t
	 *  this is the reverse of gmtime_r
	 *  It is like mktime (but treats the input and output as UTC.
	 *    That is, it does not skew the results, depending on the curent time zone, DST rules etc.
	 *
	 *  Based heavily on the Arduino TimeLib.h / Time.cpp by Paul Stoffregen
	 *    https://github.com/PaulStoffregen/Time but changed for different epochs, enumerations etc.
	 *
	 *  Named with a suffix A (timegm_rA) to make a clash with any existing library
	 *    name less likely.
	 *
	 *  Tested on an ESP8266 core 2.7.4
	 *
	 *
	 *  see:
	 *  https://github.com/espressif/arduino-esp32/blob/master/tools/sdk/include/newlib/time.h
	 *
	 */

	const time_t SECS_PER_MIN = 60UL ;
	const time_t SECS_PER_HOUR = 3600UL ;
	const time_t SECS_PER_DAY = 86400 ;

	//leap year calculator expects year argument as years offset from 1970
        #define LEAP_YEAR_rA(Y)     ( ((1970+(Y))>0) && !((1970+(Y))%4) && ( ((1970+(Y))%100) || !((1970+(Y))%400) ) )

	static  const uint8_t monthDays[]={31,28,31,30,31,30,31,31,30,31,30,31}; // this array starts from 0

	*tOut = -1 ;

	int i;
	uint32_t seconds;
	int year_ = tme->tm_year - 70 ;   // tm_year is years from 1900, time_t epoch from 1970.

	if ( year_ < 0 )  return ;  // tm structure is years from 1900 , time_t from 1970;

	// seconds from 1970 till 1 jan 00:00:00 of the given year
	seconds = year_*(SECS_PER_DAY * 365);
	for (i = 0; i < year_ ; i++) {
		if (LEAP_YEAR_rA(i)) {
			seconds += SECS_PER_DAY;   // add extra days for leap years
		}
	}

	// add days for this year, months start from 1
	for (i = 0; i < tme->tm_mon; i++) {
		if ( (i == 1) && LEAP_YEAR_rA(year_)) {
			seconds += SECS_PER_DAY * 29;
		} else {
			seconds += SECS_PER_DAY * monthDays[ i ];  // monthDay array starts from 0  //fdm
		}
	}

	seconds+= (tme->tm_mday-1) * SECS_PER_DAY;
	seconds+= tme->tm_hour * SECS_PER_HOUR;
	seconds+= tme->tm_min * SECS_PER_MIN;
	seconds+= tme->tm_sec;

	*tOut = seconds ;
}
```

d-a-v, 2021-01-07:
There's also this newlib (that we use) [proposal for `timegm()`](https://sourceware.org/legacy-ml/newlib/2018/msg00811.html
) (end of page, untested).

@earlephilhower Do you think it's worth adding it in your repo ?

@6v6gt-duino I'll try to understand what's happenning with your updated example, but I have to find a very quiet time for that :)

6v6gt-duino, 2021-01-08:
Thanks once again for looking at it.
As for a PR, I'm afraid I don't have enough of an overview over all the various bits of the C++ time infrastrucure, historical relics, vendor additions etc. etc. to have sufficient confidence that my implementation of timegm() would not simply contribute to a bigger mess. Therefore, I am reluctant to issue a PR for this and will regard it only as a "private" work around. However, if someone else wishes to adopt it, I have, of course, no objections. Unfortunately, I don't see this situation getting any better with year 2038 (Unix time DDay or 2036 for ntp DDay) approaching, prompting more changes in this area to attempt to minimise its impact, but with the accompanying instability. 
I won't be spending too much more time on this, but I think that the original matter should still be investigated because the problems may also manifest themselves elsewhere. Having said all that, I am very happy that, using only a few lines of code, it is possible to get a extremely accurate time/date stamp, corrected for local day light saving time and local time zone using a network connected ESP8266 (well at least for the next 16 to 18 years, that is).

6v6gt-duino, 2021-01-17:
I have done some further tests and now understand what the problem is.
It was my incorrect usage of configTime( ) in an  attempt to set the time zone to UTC ;
The following format of the time zone fail (silently) to set the time zone to UTC
```
configTime( "" , sntp_getservername(0), sntp_getservername(1), sntp_getservername(2) ) ; 
configTime( "TZ_Etc_UTC" , NTP_SERVER ) ; 
```
A time zone such as "UTC0" will work. However, it is best to use a form such as TZ_Etc_UTC (no quotes, chosen from TZ.h) so the compiler will reject invalid strings.
Apologies for not having seen this sooner. I will close this incident.

Here is a complete sketch which demonstrates that that configTime() does in fact work correctly:

```
/*  configTimeIssue.ino 7806  (closed)
 *
 *  Demonstrate successful use of configTime(<new signature>)
 *
 */

#include "Arduino.h"
#include <ESP8266WiFi.h>
#include <time.h>
#include <sntp.h>       // sntp_servermode_dhcp() (from NTP-TZ-DST.ino )

#include <TZ.h>

/*******************************************************************************/

const char* NTP_SERVER = "ch.pool.ntp.org";  // *****set your own

// enter your time zone (https://remotemonitoringsystems.ca/time-zone-abbreviations.php)
const char* TZ_INFO    = "CET-1CEST-2,M3.5.0/02:00:00,M10.5.0/03:00:00";  // ***** set your own (not UTC!!)

/*******************************************************************************/

#define PTM(w) \
		Serial.printf(" " #w "=%d", (int)(tm->tm_##w));

void printTm(const char* what, const tm* tm) {
	Serial.print(what);
	PTM(isdst); PTM(yday); PTM(wday);
	PTM(year);  PTM(mon);  PTM(mday);
	PTM(hour);  PTM(min);  PTM(sec);
}

void test_noTz( char v ) {

	Serial.printf("--------------------\n\nConfigTime() Signature Test #%c\n", v) ;

	// basic set up

	time_t sampleTime = 1577865600 ; // UTC  Wednesday, January 1, 2020 8:00:00 AM
	timeval tv = { sampleTime, 0 } ;
	settimeofday(&tv, nullptr);
	time_t timeNowUTC = time( nullptr ) ;

	tm bdtLoc ;
	localtime_r( &timeNowUTC, &bdtLoc ) ;

	Serial.printf( "PART1 sampleTime= %li , timeNowUTC= %li \n" , sampleTime, timeNowUTC )  ;
	printTm( "PART2" , &bdtLoc ) ; Serial.println() ;

	// now get a bdt from time_t UTC to simulate loading an RTC

	tm bdtUtc ;
	gmtime_r( &timeNowUTC, &bdtUtc ) ;
	Serial.printf( "PART3 sampleTime= %li , timeNowUTC= %li \n" , sampleTime, timeNowUTC )  ;
	printTm( "PART4" , &bdtUtc ) ; Serial.println() ;

	// now attempt to simulate recovery of timestamp from the RTC using mktime(). This clearly adds a
	//   time zone skew.

	time_t mkt1 = mktime( &bdtUtc ) ;
	Serial.printf( "PART5 timeNowUTC= %li , mkt1= %li , difference= %li \n" ,  timeNowUTC, mkt1, (timeNowUTC - mkt1) )  ;

	// now try again to simulate recovery of timestamp from the RTC but enclose mktime within a
	//  block which switches the time zone to UTC before and then retores it afterwards.

	configTime( TZ_Etc_UTC , NTP_SERVER ) ;

	delay(2000) ;

	time_t mkt2 = mktime( &bdtUtc ) ;
	Serial.printf( "PART6 timeNowUTC= %li , mkt2= %li , difference= %li \n" ,  timeNowUTC, mkt2, (timeNowUTC - mkt2) )  ;

	// now attempt to restore original time zone

	configTime( TZ_INFO, NTP_SERVER);  // new format

	delay(2000) ;

	// now test if localtime() accepts the resored time zone.

	tm bdtLoc2 ;
	localtime_r( &timeNowUTC, &bdtLoc2 ) ;
	Serial.printf( "PART7 sampleTime= %li , timeNowUTC= %li \n" , sampleTime, timeNowUTC )  ;
	printTm( "PART8" , &bdtLoc ) ; Serial.println() ;

}

void setup() {

	Serial.begin(115200);
	Serial.println("\n\nNTP Time Test\n");

	WiFi.mode(WIFI_OFF);

	configTime( TZ_INFO, NTP_SERVER);  // new format

	test_noTz( 1 ) ;

}

void loop() {
}

```

6v6gt-duino, 2021-01-17:
See the last entry for an explanation. It was a user error.
Many thanks for those who looked at this.

closed 2021-01-17 by 6v6gt-duino