esp8266/Arduino #7808: ESP8266 lags every 10-15min for a moment - evt: 7 error

ShineHunter, 2021-01-04:
#### Platform

- Hardware:				[ESP-12|
- Core Version:			[latest git date]
- Development Env:		[Arduino IDE|
- Operating System:		[Windows|

### Settings in IDE

- Module:  			|Nodemcu|
- Flash Mode:			[qio|
- Flash Size: 		[4MB]
- lwip Variant:   |v2 Lower Memory|
- Reset Method:		[ck]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] 

### Problem Description

Every10-15 time the ESP stops answering to my smartphone. Debugging on Android says "connection refused". Then, after 1-2min it starts spamming connection info to the serial port and answering to phone and after maybe 10s works normally. I am also getting wifi evt: 7 - WIFI_EVENT_SOFTAPMODE_PROBEREQRECVED on serial port.
According to https://github.com/esp8266/Arduino/issues/2735 -  it's a "[AP]err on recieved request"

... New data every 300ms ... evt: 7  ... Then updates very fast (every 50ms, not 300) for a moment.

The ESP is in AP mode.

Newest libraries.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

void setup()
{
  delay(10);
  
  Serial.begin(115200);
  Serial.println();

  WiFi.softAPConfig(ip,gateway,subnet);
  WiFi.mode(WIFI_AP);
  WiFi.softAP(ssid, password,10,false,1);
  IPAddress myIP = WiFi.softAPIP();
  Serial.print("AP IP address: ");
  Serial.println(myIP);

  server.begin();
}

void loop()
{
  WiFiClient client = server.available();
  // wait for a client (web browser) to connect
  if (client)
  {
    Serial.println("\n[Client]");
      
    while (client.connected())
    {
      Serial.println("\n[Client connected]");
      if (client.available() )
      {
        Serial.println("\n[Client available]");
        String line = client.readStringUntil('\r');
            
        line = line.substring(5);
        Serial.println(line);
                           
        String s = "HTTP/1.1 200 OK\r\nContent-Type: text/html\r\n\r\n";
        s += "@";
        s += 300+variable();
        s += "#";
        s +="200";
        s += "$";
        s += "\r\n<html>\r\n";
        //Serial.println(s);
        client.print(s);
        Serial.println("[Client disconnected]");   
        Serial.println(ESP.getFreeHeap(),DEC);     
        break;
      }     
    }
  }
}

```

### Debug Messages

```
[Client]
19:48:09.950 -> 0 HTTP/1.1
19:48:09.950 -> [Client disconnected]

19:48:58.765 wifi evt: 7 
19:49:00.627 wifi evt: 7 
19:49:00.627 wifi evt: 7 
19:49:12.179 wifi evt: 7

19:49:13.470 -> [Client]
19:49:13.470 -> 0 HTTP/1.1
19:49:13.470 -> [Client disconnected]

```

svdrummer, 2021-01-06:
As there is only part of the program here, it is hard to simulate.
A lot of problems show in the INCLUDE section and global setups.
It is hard to simulate your problem

d-a-v, 2021-01-30:
@ShineHunter please enable debug mode, and show the serial output when your issue occurs.
Also try to add `WiFi.setSleepMode(WIFI_NONE_SLEEP);`.

TD-er, 2021-02-19:
> @ShineHunter please enable debug mode, and show the serial output when your issue occurs.
> Also try to add `WiFi.setSleepMode(WIFI_NONE_SLEEP);`.

Should that make any difference?
The ESP is in AP mode according to the issue starter.

ShineHunter, 2021-02-19:
No, I tried this, but it isn't make any difference. But... I solved this problem by changed http request to standard TCP protocole.