esp8266/Arduino #7814: FS::read returning negative number on type size_t

fcapano, 2021-01-06:
in FS.cpp, function File::read should return an unsigned int (aka size_t), however it returns a negative value (-1).

https://github.com/esp8266/Arduino/blob/98125f88605cd7e46e9be4e1b3ad0600dd5d2b51/cores/esp8266/FS.cpp#L78

@igrr

themindfactory, 2021-01-06:
Seems OK to me, it denotes an error condition as there is no file pointer
defined, this would only occur if there is no open file and you tried to
read.

TD-er, 2021-01-08:
@themindfactory 
I guess @fcapano tries to make a point `-1` is not of the same type as `size_t`.
This means you may need to perform a bit more complex (or at least less intuitive) test to check the return value.

So you have 3 different return value states to check:
- return value < `size`: Not all expected data was read.
- return value == `size`: OK situation.
- return value > `size`: The error value (`-1`)

themindfactory, 2021-01-08:
True.. :-)

fcapano, 2021-01-08:
It's not a given the caller would check if the returned value is negative, given that it's returning an unsigned number.
Wouldn't it be more appropriate to return zero?

Just to drive the point further, this little program compiled with gcc will return 0:
`
#include <stddef.h>
size_t x(){ return -1; }

int main() {
    if (x()<0) return 1;
    return 0;
}
`

earlephilhower, 2021-01-08:
This is a case that's probably not hit ever since it only happens when you have a naked `File` and not one returned by a `{spiffs,sd,littlefs}.open()` call,  but you bring up a good point.

Returning `MAXINT` on a failed read is the wrong thing here IMO.  Either the `File::read` type return type should have been `ssize_t` (i.e. signed size, just like POSIX where `-1` indicates an error), or a `0` should be returned in this case.

I think I would just return 0 in this case, like in the `File::write()` calls.

themindfactory, 2021-01-08:
So now zero means error and a naked file? I think there was a reason POSIX
decided on ssize_t (-1).

earlephilhower, 2021-01-08:
0 means `EOF` for `read` which seems like a reasonable answer.  We don't have `errno` and I'm loathe to change so many function signatures for such a rare, uncommon case.

Again, this only happens if you have a fundamental, static source code problem (`File x; x.read();`) and not in the case of SPIFFS/LittleFS/SD file-not-found or other cases.  Those cases do have an implementation pointer and already return `0` when the open()ed file isn't valid.

PR #7817 (2021-01-09): Invalid read return value fix
Fixes #7814.

Return 0, not MAXINT, when a read is called on a File without a backing
instance of a SPIFFS/LittleFS/SD File.
commit 34545a16 (2021-01-09): Invalid read return value fix (#7817)

Fixes #7814.

Return 0, not MAXINT, when a read is called on a File without a backing
instance of a SPIFFS/LittleFS/SD File.
closed 2021-01-09 by earlephilhower