esp8266/Arduino #7818: Cannot stop ESP8266WebServer and reuse port

pvint, 2021-01-09:
#### Platform

- Hardware:					ESP8266
- Core Version:			2.7.4
- Development Env:		Arduino IDE
- Operating System:		Windows|Linux

### Settings in IDE

- Module:  			Generic ESP8266 Module
- Flash Mode:			DOUT
- Flash Size: 		1MB
- lwip Variant:   |v2 Lower Memory
- Reset Method:		nodemcu
- Flash Frequency:	40Mhz
- CPU Frequency:		80Mhz
- Upload Using:		SERIAL
- Upload Speed:   115200

### Problem Description

Cannot stop ESP8266WebServer and reuse port for a new webserver.

This issue first came up with a user of FauxmoESP who was using WiFiManger to set up a captive portal to configure WiFi. The issue was that once the webserver on the portal stopped, FauxmoESP was not able to use port 80 on its own webserver. (See https://github.com/vintlabs/fauxmoESP/issues/123 and https://github.com/tzapu/WiFiManager/issues/1142)

Testing today, I attempted simply using an example sketch for ESP8266WebServer, and attempting to stop it and start a second one. (Sketch below)

In this test and in the other scenario, there are no errors, but the "new" webserver on port 80 simply is not receiving any data on port 80.

This appears to be related to https://github.com/esp8266/Arduino/issues/686 

Is there a proper way to close the server that I haven't found?

Many thanks.
 
### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>

const char* ssid = "SSID";
const char* password = "PASS";

ESP8266WebServer server(80);
ESP8266WebServer server2(80);

void handleRoot() {
  server.send(200, "text/plain", "Response from server #1");
}

void handleRoot2() {
  server2.send(200, "text/plain", "Response from server #2");
}

void handleNotFound() {
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server.uri();
  message += "\nMethod: ";
  message += (server.method() == HTTP_GET) ? "GET" : "POST";
  message += "\nArguments: ";
  message += server.args();
  message += "\n";
  for (uint8_t i = 0; i < server.args(); i++) {
    message += " " + server.argName(i) + ": " + server.arg(i) + "\n";
  }
  server.send(404, "text/plain", message);
 
}

void setup(void) {
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  Serial.println("");

  // Wait for connection
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", handleRoot);

  server.begin();
  Serial.println("HTTP server started");

  while (millis() < 10000)
  {
    server.handleClient();
  }
  server.stop();

 
  server2.on("/", handleRoot2);
  server2.begin();
  Serial.println("HTTP SECOND server started");

  while (true)
  {
     server2.handleClient();
  }
}

void loop(void) { 
}
```

earlephilhower, 2021-01-09:
Can you retry with the current master?  When I run your example I get the expected output:
````
Connected to NOBABIES
IP address: 192.168.1.133
HTTP server started
HTTP SECOND server started
````

and on the webpage I get
````
Response from server #2
````

pvint, 2021-01-10:
I took a look at current master and didn't see anything that I thought would make it work, but I should really have tested anyway. ;)

Thanks for testing - sounds promising. (And really, it means a lot that you took the time to test with my code)

I'll test on current master and follow up.

Cheers
Paul

earlephilhower, 2021-01-10:
One other possibility is that it's timing related.  In many TCP stacks you can't re-use a port immediately after closing a listening socket on it.  The TCP stack wants to wait some time to ensure no old connections are still in flight.  `SO_REUSESOCKET`(sp?) , etc.  @d-a-v, is there anything in LWIP like this?

d-a-v, 2021-01-10:
> is there anything in LWIP like this?

I think [how-to-clear-tcp-pcbs-in-time-wait-state](https://arduino-esp8266.readthedocs.io/en/latest/faq/readme.html#how-to-clear-tcp-pcbs-in-time-wait-state) may help in this case.
It is no more needed for the "TCP eats my ram" issue, but will still close pcb in time-wait-state which is what happens in some circumstances when a tcp server is closed:
If a tcp server is closed when no client is connected, the trick above is useless. Otherwise, the tcp stack will keep it opened for a while to ensure the still-connected client get a proper response when they try to speak.

pvint, 2021-01-11:
Good news! I confirm that the issue no longer exists in the current master branch (34545a1)

Thanks so much for the info and your time.

Cheers!
Paul

closed 2021-01-11 by pvint

pvint, 2021-01-11:
Added note: @d-a-v : Good info there. Thanks!