esp8266/Arduino #7819: NodeMCU LittleFS implementation problem

firtinahg, 2021-01-11:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				NodeMCU v3 [ESP12E Module - 1.0]
- Core Version:			ESP8266 2.5.0
- Development Env:		Arduino IDE - 1.8.13 (Windows Store 1.8.42.0)
- Operating System:		Windows 10

### Settings in IDE

- Module:  			NodeMCU v3 [ESP12E Module - 1.0]
- Flash Mode:			??
- Flash Size: 		       4MB(1MB SPIFFS)
- lwip Variant:                  v2 Lower Memory
- Reset Method:		??
- Flash Frequency:	        40Mhz
- CPU Frequency:		80Mhz|
- Upload Using:		SERIAL
- Upload Speed:              115200

### Problem Description

I'm working on Arduino IDE to implement LittleFS file system.However I get the error message as "src\LITTLEFS.cpp:17:21: fatal error: vfs_api.h: No such file or directory #include "vfs_api.h" when I try to compile my program. I tried version 1.0.0 and 1.0.5 , there is no difference when I compiled.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>
#include "FS.h"
#include <LITTLEFS.h>

#define FORMAT_LITTLEFS_IF_FAILED true

void setup(){
  
}
void loop(){
  
}

```

### Debug Messages

```
src\LITTLEFS.cpp:17:21: fatal error: vfs_api.h: No such file or directory #include "vfs_api.h"
```

firtinahg, 2021-01-11:
You can follow the mention **_https://stackoverflow.com/questions/65669242/esp8266nodemcu-littlefs-library-problem_** 

Problem solved with this library : **https://github.com/esp8266/Arduino/tree/master/libraries/LittleFS** , I have been using the wrong library which is ESP32 library. The new library link is directs the ESP8266 library.

firtinahg, 2021-01-12:
Problem was about ESP8266 Library version. I was using 2.5.0 but to use this library you have to use 2.7.4. This solved my problem.

closed 2021-01-12 by firtinahg

Pulse-akhil, 2021-10-13:
How does one change the library version of ESP8266 inside Arduino IDE?

Akshaykushawaha, 2023-09-13:
same issue

Khalil-Daoud, 2024-06-21:
> How does one change the library version of ESP8266 inside Arduino IDE?

Tools --> Board --> Boards Manager, and you search for ESP8266