esp8266/Arduino #7829: Corruption of const char * variables

saman-taghipour, 2021-01-18:
Hi
I read data from .txt file and put this data in const char * variables. in other functions this variables is correct. but when I using client.connect("api.ipify.org", 80), all of "const char * variables" has been corrupted
this problem occur when it connect to somewhere.
when I connect to smtp server whit ESP-Mail-Client library, this problem occur too.

d-a-v, 2021-01-18:
The returned value given by `json16[]` is probably pointing to a temporary space later reused by internals.
You must either use a `String` or allocate space for your `var`.
Closing because this is probably a misuse of ArduinoJson.
Feel free to get back here if that's not the case.

closed 2021-01-18 by d-a-v

saman-taghipour, 2021-01-18:
@d-a-v 
My variable should be const char *
How can I allocate space for my const char * var?

devyte, 2021-01-18:
Your DynamicJsonBuffer is on the stack and is destroyed when the scope ends. Your var is a global, so you're trying to save a reference to a temporary, which then points to garbage.
Thisnis not the right place to ask such questions. Please request such help at a community forum like esp8266.com or stackoverflow.

saman-taghipour, 2021-01-19:
@d-a-v 
@devyte 
Thank you :-)
Your explanation was very helpful. I found problem in my code. I use String in one function and StaticJsonBuffer in other.