esp8266/Arduino #7836: wiring analogWrite() function does not respect previously set pinMode()

afdeprado, 2021-01-23:
### Basic Infos

- [X ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X ] I have tested that the issue is present in current master branch (aka latest git).
- [X ] I have searched the issue tracker for a similar issue.
- [X ] If there is a stack dump, I have decoded it.
- [X ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-12E]
- Core Version:			[master branch and 2.7.4 (3.20704.0)]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[do not know - not relevant]
- Flash Size: 		        [4MB]
- lwip Variant:                  [do not know - not relevant]
- Reset Method:		[nodemcu]
- Flash Frequency:	        [do not know - not relevant]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:              [115200] (serial upload only)

### Problem Description

The circuit I'm using the NodeMCU in requires PWM open-drain outputs. Even when setting the outputs to open-drain via pinMode(), passing OUTPUT_OPEN_DRAIN, the analogWrite() function resets the output to push-pull mode (OUTPUT).

The problem can be found in line 65 of the source file  Arduino/cores/esp8266/core_esp8266_wiring_pwm.cpp (in the current master branch), where the pinMode() is reset to OUTPUT, regardless of what was previously set via this method.

To reproduce the problem, the sketch below can be run, with the output pin connected via a pull-down resistor (10k) to ground, and an oscilloscope connected to the output. The output should stay at 0V all the time (as pwm should either connect the output to GND, or leave it open, where the pull-down resistor would keep it at GND potential, if open-drain was respected). The observed output transitions between 0V and 3V3, as corresponds to a push-pull output configuration.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <Arduino.h>

void setup() {
   pinMode(14, OUTPUT_OPEN_DRAIN);
   analogWrite(14, 128);
}

void loop() {
   // Nothing needed here
}

```

### Debug Messages

```
No debug messages
```

earlephilhower, 2021-01-24:
Interesting use case!

This is actually what the Arduino upstream versions do, so I suggest this is expected behavior:
https://github.com/arduino/ArduinoCore-avr/blob/master/cores/arduino/wiring_analog.c#L96-L104

You should be able to change the pinMode() immediately after your analogWrite(), or adjust your own fork of the core to remove the pinMode() command.

klugem, 2021-01-25:
Funnily enough, I had the exact same problem yesterday. A constant current source I want to control via PWM expects an open-drain PWM signal. 

I would suggest to add an optional parameter to analogWrite to allow setting the pin to OUTPUT_OPEN_DRAIN mode. I opened a pull request with that change.

afdeprado, 2021-01-25:
Michael, I think your solution is a very good alternative, as it doesn't disrupt the current working, nor compatibility with upstream Arduino versions, but solves a problem (and probably, an unintended bug in the original Arduino wiring libraries). I like the final proposed solution with a new analogWriteEx method. For my current project, I just duplicated the method and edited it to suit my purposes, as I was in a hurry.

Thanks everyone involved!

PR #7841 (2021-01-27): allow to set pin to OUTPUT_OPEN_DRAIN in analogWriteMode
I added an optional parameter in the analogWrite() function to allow open-drain PWM signals.
Also see #7836

*edit*: closes #7836
closed 2021-01-27 by earlephilhower