esp8266/Arduino #7851: client.connected() in AP mode problem.

Proamp, 2021-01-30:
client.connected() in AP mode always return 1 after the first connection.

Start code. All fine nobody connected to AP.
[image]

One user connected to AP
[image]

 User open ip:port
[image]

User disconnected from this AP(no one connected to this AP) , but client.connected() still return 1.
[image]

Tried on different ESP8266 boards, the result is still the same.

d-a-v, 2021-01-30:
You *must not* use `delay(500)` in the main loop.
Instead us polled timers:
```cpp
#include <PolledTImeout.h>
esp8266::polledTimeout::periodicMs every500ms(500);
void loop()
{
    if (every500ms) {
        Serial.print(...);
    }
    ... // no delays when network is used
}
```

> User disconnected from this AP(no one connected to this AP) , but client.connected() still return 1.

Did the client shut the TCP connection down before disconnecting from the AP ?
If that's not the case, the TCP stack is waiting for a proper shutdown or a timeout.
For example in the case of temporary loss of signal (too far or whatever reason), the TCP connection is able to go on once the link is reestablished. The TCP stack is behaving correctly.

Like you must shut the OS down before switching off your computer, you must properly close TCP connections before plugging the link off.

Proamp, 2021-01-30:
> You _must not_ use `delay(500)` in the main loop.
> Instead, use polled timers:
> 
> ```c++
> #include <PolledTimeout.h>
> esp8266::polledTimeout::periodicMs every500ms(500);
> void loop()
> {
>     if (every500ms) {
>         Serial.print(...);
>     }
>     ... // no delays when network is used
> }
> ```
> 
> > User disconnected from this AP(no one connected to this AP) , but client.connected() still return 1.
> 
> Did the client shut the TCP connection down before disconnecting from the AP ?
> If that's not the case, the TCP stack is waiting for a proper shutdown or a timeout.
> For example in the case of temporary loss of signal (too far or whatever reason), the TCP connection is able to go on once the link is reestablished. The TCP stack is behaving correctly.
> 
> Like you must shut the OS down before switching off your computer, you must properly close TCP connections before plugging the link off.

delay is for restrict console messages freq only... 

No, client is not shut down TCP connection before disconnecting from the AP.
I'm talking about the logic only. It's not possible to get any TCP connection without connection to the AP. Why this logic used in library?

Also, if i'll try to send anything via client.write(), when client is disconnected from the AP. what will happened?  This library allow me to do this.

d-a-v, 2021-01-30:
> delay is for restrict console messages freq only...

Of course.
I showed an example in my first answer with `PolledTimeout` about how to display messages every 500ms without introducing `delay(500)`.

> No, client is not shut down TCP connection before disconnecting from the AP.

Then that's the reason why the other side (ESP/AP) still believes the connection is opened.

> I'm talking about the logic only. It's not possible to get any TCP connection without connection to the AP. Why this logic used in library?

You have the same logic on your linux/windows computer.
Make a TCP connection, then abruptely shut the power on one side, the other side will patiently be waiting for answers or data, with a connected link, eventually closing after a (long to very long) timeout.

If you assume that without any link connection (AP, STA, Ethernet, PPP, ...) a TCP connection is impossible and consequently all logical connections should be closed, then that assumption is wrong.
*The TCP stack cannot know in advance whether the link layer won't be up later again*, as I explain in my first answer.

> Also, if i'll try to send anything via client.write(), when client is disconnected from the AP. what will happened? This library allow me to do this.

Every decent TCP library allows to do this. `client::write()` function returns the number of written bytes. The output buffer will be full at some point and the function will return 0. Later the TCP connection will get a timeout as described above.

Proamp, 2021-01-30:
> > delay is for restrict console messages freq only...
> 
> Of course.
> I showed an example in my first answer with `PolledTimeout` about how to display messages every 500ms without introducing `delay(500)`.
> 
> > No, client is not shut down TCP connection before disconnecting from the AP.
> 
> Then that's the reason why the other side (ESP/AP) still believes the connection is opened.
> 
> > I'm talking about the logic only. It's not possible to get any TCP connection without connection to the AP. Why this logic used in library?
> 
> You have the same logic on your linux/windows computer.
> Make a TCP connection, then abruptely shut the power on one side, the other side will patiently be waiting for answers or data, with a connected link (layer3), eventually closing after a (long to very long) timeout.
> 
> If you assume that without any link connection (layer1: AP, STA, Ethernet, PPP, ...) a TCP connection is impossible and consequently all logical connections should be closed, then that assumption is wrong.
> _The TCP stack cannot know in advance whether the link layer won't be up later again_, as I explain in my first answer.
> 
> > Also, if i'll try to send anything via client.write(), when client is disconnected from the AP. what will happened? This library allow me to do this.
> 
> Every decent TCP library allows to do this. `client::write()` function returns the number of written bytes. The output buffer will be full at some point and the function will return 0. Later the TCP connection will get a timeout as described above.

Regarding Arduino docs:
connected() 
Returns true if the client is connected, false if not.

But it's not true cause it returns true also while user is disconnected.

In my project I restrict client.write() function. This funcion send data to client only when client exist. It's right way I think. What reason to send data if nobody receive it? Better way to do it is to use the return data of client.connected(). But it doesn't work correctly. I'm using WiFi.softAPgetStationNum() instead, but it's not so good.

d-a-v, 2021-01-30:
> `connected()` returns true if the client is connected, false if not.

Do you know what walkies-talkies are ?
If your mate suddenly does not answer anymore for 5 minutes without having said "over and out",
will you think you'll never be able to talk to him/her again ?

I hope, for your mate, that you'd not assume that (s)he's dead and that you'd continue to "do you copy me?" until you get an answer.
That's ***exactly*** what your TCP link does.

And your mate may have run to the car to get the cellphone to reach you because walkies's batteries are dead.
TCP (layer3) relies on several layer1 (AP, STA, Ethernet - like walkies, cellphone).

To summarize, if you don't deliberately close your TCP connection,
***the connection is supposed to stay opened until "over and out" is heard***.

d-a-v, 2021-01-30:
You are free to implement a user-timeout in your sketch.
For example if a client hasn't sent any data for more than 10secs, then the connection is considered invalid, and can be unilaterally closed.

Proamp, 2021-01-30:
What I'm talking about, function client.connected() works incorrect cause it returns true w/o any TCP connection. That's all. Do you think it's normal behavior?

d-a-v, 2021-01-30:
Yes.

Proamp, 2021-01-30:
> Yes.

Function client.connected() returns true, when client is disconnected?

d-a-v, 2021-01-30:
Client is not disconnected.

Proamp, 2021-01-30:
> Client is not disconnected.

Why? It's disconnected from AP)))

d-a-v, 2021-01-30:
I think you are now ready to read the first answer and the following ones.
If thay's not sufficient I can repeat the whole thing in french if that can help.

closed 2021-01-30 by d-a-v