esp8266/Arduino #7852: get HEX data from server.arg()

saman-taghipour, 2021-01-30:
Hi,
I receive data from an Editbox in my web page with:
`server.arg("editBox1");`
But I think that it receives HEX data
for example:
When my input data is: 12345%6789
by server.arg() function
received data: 12345g89 (67 is HEX of 'g')
other example:
input data: 12345#6789
received data: 12345

d-a-v, 2021-01-30:
You deliberately deleted the issue template, and you are not giving any single detail on what is your "editbox" nor an MCVE for us to reproduce.

You are not helping us helping you. 

Closing.

closed 2021-01-30 by d-a-v

saman-taghipour, 2021-01-30:
Yes, I delete template, but I wrote my question.
So what should I do now?

saman-taghipour, 2021-01-31:
Here is my code for editBox:

`#editBox_eb { border:1px solid #CCCCCC; border-radius:4px; background-color:#FFFFFF; background-image:none; color:#000000; font-family:Arial; font-weight:normal; font-style:normal; font-size:13px; line-height:16px; padding:4px 4px 4px 4px; margin:0; text-align:left; }
#editBox_eb:focus { border-color:#66AFE9; box-shadow:inset 0px 1px 1px rgba(0,0,0,0.075), 0px 0px 8px rgba(102,175,233,0.60); outline:0; }
#editBox_eb { position:absolute; left:135px; top:35px; width:123px; height:16px; z-index:40; }

<input type="text" id="editBox_eb" name="editBox_eb" value="" tabindex="5" spellcheck="false">`

devyte, 2021-01-31:
The issue template that you deleted explains that this is not the right place to ask for general help for your particular project, because this is an issue tracker meant to track issues in our core code. Please ask for help at a community forum like esp8266.com or stackoverflow.

mathertel, 2021-01-31:
This is handled correctly in the URL parsing in this code.

using a '%' or a '#' in URLS has a special meaning.
'%' : starts specifying a character by it's code
'#': starts local parameters a.k.a. hash parameters (not send to server)
to pass the '12345%6789' value you have to encode the '%' character by '%25' so pass 12345%256789
 
See https://www.urlencoder.org/ and search for "url encoding"