esp8266/Arduino #7855: ESP8266 Restart Sometimes

laercionit, 2021-02-01:
----------------------------- Delete below -----------------------------

If your issue is a general question, starts similar to "How do I..", is related to 3rd party libs, or is related to hardware, please discuss at a community forum like esp8266.com.

INSTRUCTIONS
============
If you do not follow these instructions, your issue may be dismissed.

1. Follow the checklist under Basic Infos and fill in the [ ] spaces with an X.
2. Fill in all the fields under Platform and Settings in IDE marked with [ ] (pick the correct option for you in each case, delete the others).
3. If you haven't already done so, test your issue against current master branch (aka latest git), because it may have been already fixed.
4. Describe your problem.
5. If you have a STACK DUMP decode it:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html

6. Include a Minimal Complete Reproducible Example sketch that shows your issue. Do not include your entire project, or a huge piece of code.
7. Include debug messages:

https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/debugging.html 

8. Use markup (buttons above) and the Preview tab to check what the issue will look like.
9. Delete these instructions from the above to the below marker lines before submitting this issue.

----------------------------- Delete above -----------------------------

### Basic Infos

- [ X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ X] I have searched the issue tracker for a similar issue.
- [ X] If there is a stack dump, I have decoded it.
- [ X] I have filled out all fields below.

#### Platform

- Hardware:	[ESP-12]
- Core Version: [2.7.4]
- Development Env:[Arduino IDE]
- Operating System: [Windows]

### Settings in IDE

- Module:  		[Nodemcu 1.0]
- Flash Mode:	[qio|dio|other]
- Flash Size: 	[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:  [nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:	[80Mhz|]
- Upload Using:    [SERIAL]
- Upload Speed:   [115200]

### Problem Description

My project has been running for a long time, it has MQTT, WEBServer and Wifi Client. I've been investigating the reason for restarting at random for a long time. Works without restarting for weeks and restarts. I believe it is something related to a momentary internet or wifi failure. I made all possible code dealings but I did not identify the problem.  

The code hasn't changed, I just update the CORE, unfortunately because it's something I can't reproduce, I can't specify exactly which CORE the problem occurs in.

In the investigation I saw that it generates the problem in the line below:
#ifdef __GXX_EXPERIMENTAL_CXX0X__
void String :: move (String & rhs) {
    if (buffer ()) {
        if (capacity ()> = rhs.len ()) {
            memmove_P (wbuffer (), rhs.buffer (), rhs.length () + 1);
            setLen (rhs.len ());
            rhs.invalidate ();
            return;
        } else {
            if (! isSSO ()) {
                free (wbuffer ());
                setBuffer (nullptr);
            }
        }
    }
    if (rhs.isSSO ()) {
        setSSO (true);
        memmove_P (sso.buff, rhs.sso.buff, sizeof (sso.buff));
    } else {
        setSSO (false);
        setBuffer (rhs.wbuffer ());
    }
    setCapacity (rhs.capacity ());
    setLen (rhs.len ());
    rhs.setSSO (false);
    rhs.setCapacity (0);
    rhs.setLen (0);
    rhs.setBuffer (nullptr);
}
#endif

### Debug Messages

```
Decoding 13 results
0x40236644: String::move(String&) at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/WString.cpp line 230
0x4022c388: HTTPClient::disconnect(bool) at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 434
0x4022d264: HTTPClient::handleHeaderResponse() at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 1376
0x40211ad8: _Z14SystemRegisterb$part$305 at C:\Users\Laercio\AppData\Local\Temp\arduino_build_914177\sketch/System.h line 607
0x40100500: millis at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_wiring.cpp line 182
0x402377da: eboot_command_write at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_eboot_command.cpp line 59
0x40206bd3: Waiting(unsigned int) at C:\Users\Laercio\AppData\Local\Temp\arduino_build_914177\sketch/RTC.h line 33
0x40211bdc: SystemInit(bool) at C:\Users\Laercio\AppData\Local\Temp\arduino_build_914177\sketch/System.h line 625
0x40211cd5: SystemMonitor(bool) at C:\Users\Laercio\AppData\Local\Temp\arduino_build_914177\sketch/System.h line 652
0x402281a5: loop at D:\Save\Aquarino\Projects\Aquarino-ControllerBeta/Aquarino-ControllerBeta.ino line 98
0x402378bc: esp_yield at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_main.cpp line 119
0x4010118d: cont_wrapper at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/cont.S line 81

```

earlephilhower, 2021-02-02:
You need to paste in the entire crash dump to be useful.  Everything between the "----"s including the "...EPC1 xxxxx..." line is needed.  Without that, you just have a list of prior calls and not exactly where it crashed.

Also, without an MCVE, there's really not much that can be done even with the full PC decode.

laercionit, 2021-02-02:
@earlephilhower 

I just put what you requested. Thank you and I'm sorry I did it wrong.

earlephilhower, 2021-02-02:
Actually, we can't use the raw bytes you just added.  CAn you paste that into the exception decoder and update the output? It will have a "PC: xxxx <function-name>" line which will be the exact spot where a problem happened and a text description of the exception.  One more try, please!

laercionit, 2021-02-02:
Exception 9: LoadStoreAlignmentCause: Load or store to an unaligned address
Decoding 14 results
0x4023a974: WiFiClient::operator bool() at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266WiFi\src/WiFiClient.cpp line 349
0x40236644: String::move(String&) at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/WString.cpp line 230
0x4022c388: HTTPClient::disconnect(bool) at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 434
0x4022d264: HTTPClient::handleHeaderResponse() at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266HTTPClient\src/ESP8266HTTPClient.cpp line 1376
0x40211ad8: _Z14SystemRegisterb$part$305 at C:\Users\Laercio\AppData\Local\Temp\arduino_build_149574\sketch/System.h line 607
0x40100500: millis at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_wiring.cpp line 182
0x402377da: eboot_command_write at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_eboot_command.cpp line 59
0x40206bd3: Waiting(unsigned int) at C:\Users\Laercio\AppData\Local\Temp\arduino_build_149574\sketch/RTC.h line 33
0x40211bdc: SystemInit(bool) at C:\Users\Laercio\AppData\Local\Temp\arduino_build_149574\sketch/System.h line 625
0x40211cd5: SystemMonitor(bool) at C:\Users\Laercio\AppData\Local\Temp\arduino_build_149574\sketch/System.h line 652
0x402281a5: loop at D:\Save\Aquarino\Projects\Aquarino-ControllerBeta/Aquarino-ControllerBeta.ino line 98
0x402378bc: esp_yield at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_main.cpp line 119
0x4010118d: cont_wrapper at D:\SETUP\Programas\Desenvolvimento\ArduinoDesenv\portable\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/cont.S line 81

[image]

laercionit, 2021-02-02:
void SystemRegister(bool Boot = false) {
  Waiting();
  if (NetworkInternetStatus) {
    HTTPClient http;
    http.setTimeout(NetworkTimeout);
    http.begin (SystemRegisterURL);
    if (String(System.UserEmail) != "") http.addHeader(F("X_EMAIL"), System.UserEmail);
    http.addHeader( F("X_UUID"),           System.UUID     );
    http.addHeader( F("X_CHIP"),           System.Chip     );
    http.addHeader( F("X_HEAP"),    String(SystemGetFreeMemory()));
    http.addHeader( F("X_VERSION"),        SystemGetFullVersion());
    http.addHeader( F("X_HOSTNAME"),       System.Hostname );
    http.addHeader( F("X_FIRMWARE"), String(System.Firmware));
    http.addHeader( F("X_MANAGER"), String(System.Manager) );
    http.addHeader( F("X_LAN"),            NetworkGetIP()  );
    http.addHeader( F("X_STARTED"), String(System.Started) );
    int httpCode = http.POST(System.UUID);
    String ServerResponseBody;
    if (httpCode > 0) ServerResponseBody = http.getString();
    else              ServerResponseBody = http.errorToString(httpCode);
    http.end();
#ifdef SYSTEMDEBUG
    snprintf_P(Buffer, sizeof(Buffer), LangWebClientRequest, SystemRegisterURL, httpCode, ServerResponseBody.c_str());
    DebugSerial.println(Buffer);
#endif
    if ((httpCode > 0) && (Boot))  {
      if (httpCode == 200) { //Licensed
        SystemSetRegistered(true); SystemSetLicensed(true);
      } else if (httpCode == 201) { //Create
        SystemSetRegistered(true); SystemSetLicensed(false);
      } else if (httpCode == 202) { //Accept
        SystemSetRegistered(true); SystemSetLicensed(false);
      } else if (httpCode == 401) { //Unauthorized
        SystemSetRegistered(false); SystemSetLicensed(false);
      } else if (httpCode == 409) { //Conflict
        SystemSetRegistered(true); SystemSetLicensed(false);
      } else {
        LogSaveRecord (ServerResponseBody);
        if (Alert[AlertErrors].Status) EmailSend(Alert[AlertErrors].Name, FPSTR(LangEmailErrors), ServerResponseBody);
      }
    }
  }
}

laercionit, 2021-02-02:
[image]

d-a-v, 2021-02-02:
Try this:
```
    WiFiClient client; // <== added, before http
    HTTPClient http;
    http.setTimeout(NetworkTimeout);
    http.begin (client, SystemRegisterURL); // <== updated
    ...
````

laercionit, 2021-02-02:
@d-a-v

can I use the Wi-Fi Client that is instantiated for use in mqtt instead of instantiating a new one in the method?

d-a-v, 2021-02-02:
> can I use the Wi-Fi Client that is instantiated for use in mqtt instead of instantiating a new one in the method?

No. A WiFiClient matches with a connection. One would close the other.
If you need both connection at the same time, then you need two WiFiClient.

laercionit, 2021-02-02:
@d-a-v 
OK, I made the code change and I'll leave it running to see if it will resolve. Thanks for trying to help. I will inform you if it is stable or if the problem still persists.

laercionit, 2021-02-03:
Following the uptime, so far this is ok, but I wanted to understand what really changed in instantiating the WIFI Client before the HTTP Client could result in the request failing.

[image]

d-a-v, 2021-02-03:
> what really changed in instantiating the WIFI Client before the HTTP Client could result in the request failing.

Functionally speaking, that's hard to say.
That is anyway a deprecated way of using it in v2.7.4, and it is now removed in the soon-to-come v3 core. So it's better for your application to use the new API.
This parameter appeared in #4980, originally to nearly-transparently allow to switch to TLS where `WiFiClient` is used. If I remember well, management of the internal WiFiClient that needed to be created when using the old API was a bit complex and this complexity is removed with the new API.

laercionit, 2021-02-07:
Hello guys, just to inform you that the requested change worked. Thank you very much!!!!

closed 2021-02-07 by laercionit