esp8266/Arduino #7856: Crash occurs when I try to execute MDNS.removeService

whakru, 2021-02-02:
#### Platform

- Core Version:			[arduino 1.8.13 core esp8266 2.7.4]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

Crash occurs when I try to execute MDNS.removeService
If MDNS. end or MDNS.removeService has already been executed before, this causes a crash

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

if (WiFi.status() == WL_CONNECTED && lastStatus != WL_CONNECTED ){ 
      Serial.printf("Connected, IP is: %s\n", WiFi.localIP().toString().c_str());
      MDNS.removeService(DNS_NAME, "http", "tcp");
      MDNS.removeService(DNS_NAME, "ws", "tcp");
      MDNS.end(); //завершаем.
      delay(80);
      Serial.printf("MDNS begin 1 \n"); 
      if (!MDNS.begin(DNS_NAME,WiFi.localIP())) 
      {
        Serial.printf("Cannot restart mDNS responder\n");
      }
      else
      {
        Serial.printf("mDNS responder restarted\n");
      }
      Serial.printf("MDNS addService 2 \n"); 
      MDNS.addService("http", "tcp", 80); yield();
      MDNS.addService("ws", "tcp", 81); yield(); 
      

```

### Debug Messages

```
13:01:18.835 -> 
13:01:18.835 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
13:01:18.835 -> 
13:01:18.835 -> Exception (28):
13:01:18.835 -> epc1=0x40210ff3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
13:01:18.835 -> 
13:01:18.835 -> >>>stack>>>
13:01:18.835 -> 
13:01:18.835 -> ctx: cont
13:01:18.835 -> sp: 3ffffd00 end: 3fffffc0 offset: 0190
13:01:18.835 -> 3ffffe90:  3fffff40 3fffff10 00000008 0000000c  
13:01:18.835 -> 3ffffea0:  3ffefa90 00000008 3ffffec0 402133c0  
13:01:18.869 -> 3ffffeb0:  402207d8 00000000 0000000a 4021380c  
13:01:18.869 -> 3ffffec0:  534e444d 3fff1534 3ffef5b0 4021108a  
13:01:18.869 -> 3ffffed0:  3931203a 36312e32 31332e38 0a33382e  
13:01:18.869 -> 3ffffee0:  3fffff00 3fff1534 3fff1534 4020fa9d  
13:01:18.869 -> 3ffffef0:  000001f0 0000003e 00000000 4020cf22  
13:01:18.869 -> 3fffff00:  3fff0000 00000000 01010000 3f010001  
13:01:18.869 -> 3fffff10:  3fff0000 00000000 00000080 00000080  
13:01:18.903 -> 3fffff20:  3fff08e8 00000000 3fffff60 3ffef5b0  
13:01:18.903 -> 3fffff30:  3ffe8b40 3ffef5b0 3fff1534 4020c63f  
13:01:18.903 -> 3fffff40:  3ffe8b40 3ffefa90 3ffef5b0 4020c670  
13:01:18.903 -> 3fffff50:  3fffdad0 3ffefa90 3ffef889 40205a75  
13:01:18.903 -> 3fffff60:  3fff0400 000d000f 80000001 402207d8  
13:01:18.903 -> 3fffff70:  531fa8c0 00000000 3ffefb9c 3ffefbdc  
13:01:18.903 -> 3fffff80:  3fffdad0 00000000 3ffefb9c 3ffefbdc  
13:01:18.903 -> 3fffff90:  3fffdad0 00000000 3ffefb9c 40206b2c  
13:01:18.937 -> 3fffffa0:  3fffdad0 00000000 3ffefb9c 40215da0  
13:01:18.937 -> 3fffffb0:  feefeffe feefeffe 3ffe85b4 40101445  
13:01:18.937 -> <<<stack<<<
13:01:18.937 -> 
13:01:18.937 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
13:01:18.937 -> 
13:01:18.937 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
13:01:18.972 -> 
13:01:18.972 -> load 0x4010f000, len 3584, room 16 
13:01:18.972 -> tail 0
13:01:18.972 -> chksum 0xb0
13:01:18.972 -> csum 0xb0
13:01:18.972 -> v2843a5ac
13:01:18.972 -> ~ld
13:01:19.042 -> 
13:01:20.677 -> 
```

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40210ff3: esp8266::MDNSImplementation::MDNSResponder::_sendMDNSMessage_Multicast(esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 125
EXCVADDR: 0x00000000

Decoding stack results
0x402133c0: HardwareSerial::write(unsigned char const*, unsigned int) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/HardwareSerial.h line 165
0x4021380c: Print::printf(char const*, ...) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\Print.cpp line 74
0x4021108a: esp8266::MDNSImplementation::MDNSResponder::_sendMDNSMessage(esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 96
0x4020fa9d: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::stcMDNSSendParameter() at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS_Structs.cpp line 2375
0x4020cf22: esp8266::MDNSImplementation::MDNSResponder::_announceService(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService&, bool) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 1641
0x4020c63f: esp8266::MDNSImplementation::MDNSResponder::removeService(void const*) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS.cpp line 343
0x4020c670: esp8266::MDNSImplementation::MDNSResponder::removeService(char const*, char const*, char const*) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS.cpp line 362
0x40205a75: CheckConnect() at D:\YandexDisk\Нужное\Arduino\b1_CLOCK_v1.4NEW/a1_1wifi_init.ino line 398
0x40206b2c: loop() at D:\YandexDisk\Нужное\Arduino\b1_CLOCK_v1.4NEW/b1_CLOCK_v1.4NEW.ino line 314
0x40215da0: loop_wrapper() at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197

```

hreintke, 2021-02-02:
@d-a-v 
I will take a look at this

@whakru 
1/ Can you share the full mcve sketch, just to be sure I will be using the same 
2/ Can you run the mcve with git master ? There have been significant updates to mdns

whakru, 2021-02-02:
1. the sketch is large, I think you will not have so many modules to check.
Just try
execute MDNS.removeService(); after MDNS.end();
2.
[image]

hreintke, 2021-02-03:
@whakru : Can you check whether #7859 also works for you ?

whakru, 2021-02-04:
@hreintke hreintke> Can you check whether #7859 also works for you ?

sorry the check took time since I was installing the git version.
It doesn't work. An exception is thrown when I try to execute MDNS. removeService after MDNS.end().
I do not consider this a critical error, but I think it is worth processing the code.

     ```
     Serial.printf("Connected, IP is: %s\n", WiFi.localIP().toString().c_str());
     delay(150);
     MDNS.end(); 
     MDNS.removeService(DNS_NAME, "http", "tcp");//  <- crash
     MDNS.removeService(DNS_NAME, "ws", "tcp");
```

```
14:21:21.990 -> Connected, IP is: 192.168.31.83
14:21:22.126 -> 
14:21:22.126 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
14:21:22.126 -> 
14:21:22.126 -> Exception (28):
14:21:22.126 -> epc1=0x40210236 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
14:21:22.159 -> 
14:21:22.159 -> >>>stack>>>
14:21:22.159 -> 
14:21:22.159 -> ctx: cont
14:21:22.159 -> sp: 3ffffcf0 end: 3fffffc0 offset: 0190
14:21:22.159 -> 3ffffe80:  0000023e 0000023e 3ffe86a0 40100e7f  
14:21:22.159 -> 3ffffe90:  40220ae0 00000000 3ffffef0 3ffefda8  
14:21:22.159 -> 3ffffea0:  40220ae0 00000000 3fff17f4 4010105e  
14:21:22.159 -> 3ffffeb0:  3fff078c 00000001 3ffef710 402102ca  
14:21:22.159 -> 3ffffec0:  0000039a 00000000 3ffffef0 4020e8f6  
14:21:22.159 -> 3ffffed0:  3ffef710 00000000 3ffef710 4020e925  
14:21:22.193 -> 3ffffee0:  3ffef710 00000020 00000000 4020c0e6  
14:21:22.193 -> 3ffffef0:  3ffe0000 00000000 01010000 40010001  
14:21:22.193 -> 3fffff00:  fb000000 00000000 3fff0aa4 4022aa9c  
14:21:22.193 -> 3fffff10:  3ffeeb30 00000000 3fff0aa4 3ffe8d46  
14:21:22.193 -> 3fffff20:  3ffef710 3fff078c 3ffef710 4020ba6b  
14:21:22.193 -> 3fffff30:  3ffef710 3ffef190 3ffef710 4020ba9c  
14:21:22.193 -> 3fffff40:  3ffef188 3ffef190 3ffef9e9 40205937  
14:21:22.193 -> 3fffff50:  00000000 000d000f 00000000 40220ae0  
14:21:22.227 -> 3fffff60:  531fa8c0 db440d81 3ffefb00 3ffefda8  
14:21:22.227 -> 3fffff70:  3ffe8d8e 3ffefbf4 3ffefbd4 3ffefda8  
14:21:22.227 -> 3fffff80:  3fffdad0 3ffefbd4 3ffef88c 3ffefda8  
14:21:22.227 -> 3fffff90:  3fffdad0 3ffefbd4 3ffefd94 40206e34  
14:21:22.227 -> 3fffffa0:  3fffdad0 3ffefbd4 3ffefd94 40215f38  
14:21:22.227 -> 3fffffb0:  feefeffe feefeffe 3ffe869c 40101319  
14:21:22.227 -> <<<stack<<<
14:21:22.261 -> 
14:21:22.261 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
14:21:22.261 -> 
14:21:22.261 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
14:21:22.261 -> 
14:21:22.296 -> load 0x4010f000, len 3616, room 16 
14:21:22.296 -> tail 0
14:21:22.296 -> chksum 0x07
14:21:22.296 -> csum 0x07
14:21:22.296 -> v20413f81
14:21:22.296 -> ~ld
14:21:22.366 -> 
14:21:23.990 -> 

```

```
Exception 28: LoadProhibited: A load referenced a page mapped with an attribute that does not permit loads
PC: 0x40210236
EXCVADDR: 0x00000000

Decoding stack results
0x40100e7f: umm_free_core(umm_heap_context_t*, void*) at D:\YandexDisk\������\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 549
0x4010105e: free(void*) at D:\YandexDisk\������\Arduino\hardware\esp8266com\esp8266\cores\esp8266\umm_malloc\umm_malloc.cpp line 595
0x402102ca: esp8266::MDNSImplementation::MDNSResponder::_sendMDNSMessage(esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at D:\YandexDisk\������\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 101
0x4020e8f6: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::clear() at D:\YandexDisk\������\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Structs.cpp line 2412
0x4020e925: esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter::stcMDNSSendParameter() at D:\YandexDisk\������\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Structs.cpp line 2375
0x4020c0e6: esp8266::MDNSImplementation::MDNSResponder::_announceService(esp8266::MDNSImplementation::MDNSResponder::stcMDNSService&, bool) at D:\YandexDisk\������\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS_Control.cpp line 1632
0x4022aa9c: operator delete[](void*) at /workdir/repo/gcc-gnu/libstdc++-v3/libsupc++/del_opv.cc line 35
0x4020ba6b: esp8266::MDNSImplementation::MDNSResponder::removeService(void const*) at D:\YandexDisk\������\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS.cpp line 262
0x4020ba9c: esp8266::MDNSImplementation::MDNSResponder::removeService(char const*, char const*, char const*) at D:\YandexDisk\������\Arduino\hardware\esp8266com\esp8266\libraries\ESP8266mDNS\src\LEAmDNS.cpp line 280
0x40205937: CheckConnect() at D:\YandexDisk\Нужное\Arduino\b1_CLOCK_v1.4NEW/a1_1wifi_init.ino line 399
0x40206e34: loop() at D:\YandexDisk\Нужное\Arduino\b1_CLOCK_v1.4NEW/b1_CLOCK_v1.4NEW.ino line 314
0x40215f38: loop_wrapper() at D:\YandexDisk\������\Arduino\hardware\esp8266com\esp8266\cores\esp8266\core_esp8266_main.cpp line 200

```

hreintke, 2021-02-04:
@whakru 
Did you also apply the patch op top of the git master.
The PR is not yet committed.

@d-a-v 
Can you add this patch to the alpha version so it can be tested with boardmanager too ?

whakru, 2021-02-04:
> Did you also apply the patch op top of the git master.
> The PR is not yet committed.

[image]
Did I do everything right?

hreintke, 2021-02-04:
You should add the PR to your workspace
https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/checking-out-pull-requests-locally

In summary : 
git fetch origin pull/7859/head:mdnsbranch
git checkout mdnsbranche

whakru, 2021-02-04:
> You should add the PR to your workspace
> https://docs.github.com/en/github/collaborating-with-issues-and-pull-requests/checking-out-pull-requests-locally
> 
> In summary :
> git fetch origin pull/7859/head:mdnsbranch
> git checkout mdnsbranche

Sorry I've never encountered this before
[image]

hreintke, 2021-02-04:
@whakru Typo from my side

git checkout mdnsbranch

The `mdnsbranch` should be the same in both commands
The git fetch... creates a branch, the git checkout... lets you use the newly created branch

whakru, 2021-02-04:
> git checkout mdnsbranch
> 
> The `mdnsbranch` should be the same in both commands
> The git fetch... creates a branch, the git checkout... lets you use the newly created branch

Indeed, I didn't notice it either =D
I've done everything:
```
C:\Users\oao-a\Documents\Arduino\hardware\esp8266com\esp8266>git fetch origin pull/7859/head:mdnsbranch

C:\Users\oao-a\Documents\Arduino\hardware\esp8266com\esp8266>git checkout mdnsbranch
Switched to branch 'mdnsbranch'
M       libraries/SoftwareSerial

C:\Users\oao-a\Documents\Arduino\hardware\esp8266com\esp8266>git status
On branch mdnsbranch
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   libraries/SoftwareSerial (new commits)

no changes added to commit (use "git add" and/or "git commit -a")
```

I checked, there are no more errors) this can be considered a correction) Thank you.

Can you tell me how I can then go back to the master again?  (I'm new to git)

d-a-v, 2021-02-04:
> Can you tell me how I can then go back to the master again?

`git checkout master` ?

closed 2021-02-04 by d-a-v

whakru, 2021-02-04:
> `git checkout master` ?

 Thank you.

PR #7859 (2021-02-04): MDNS.end() also needs to release services
Fixes #7856