esp8266/Arduino #7857: after auto-reconnection attempts, memory leaks and crashes occur. Exception 29

whakru, 2021-02-02:
#### Platform

- Hardware:					[ESP8285 device]
- Core Version:			[arduino 1.8.13 esp Release 2.7.4]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

After losing the connection for unknown reasons, it can not restore the connection, and begins to lose memory (FreeHeap: ). What eventually comes to crash.
If I try to disable my router myself to lose connection, this does not happen.

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

```

### Debug Messages

```
15:12:49.956 ->   -  -  -  -  -  -  -  -  - Sync OK -  -  -  -  -  -  -  -  -  -  -  -
15:12:49.956 -> FreeHeap: 34504
15:12:54.942 -> FreeHeap: 34504
15:12:59.932 -> FreeHeap: 34504
15:13:03.026 -> state: 5 -> 2 (3c0)
15:13:03.026 -> rm 0
15:13:03.026 -> pm close 7
15:13:03.026 -> wifi evt: 1
15:13:03.026 -> STA disconnect: 3
15:13:04.023 -> reconnect
15:13:04.023 -> state: 2 -> 0 (0)
15:13:04.023 -> Интернета нет...... OFF MDNS! 
15:13:04.161 -> scandone
15:13:04.161 -> state: 0 -> 2 (b0)
15:13:04.945 -> FreeHeap: 35640
15:13:05.148 -> state: 2 -> 0 (2)
15:13:05.148 -> reconnect
15:13:05.148 -> wifi evt: 1
15:13:05.148 -> STA disconnect: 2
15:13:05.285 -> scandone
15:13:05.285 -> state: 0 -> 2 (b0)
15:13:06.276 -> state: 2 -> 0 (2)
15:13:06.276 -> reconnect
15:13:06.276 -> wifi evt: 1
15:13:06.276 -> STA disconnect: 2
15:13:06.413 -> scandone
15:13:06.413 -> state: 0 -> 2 (b0)
15:13:07.401 -> state: 2 -> 0 (2)
15:13:07.401 -> reconnect
15:13:07.401 -> wifi evt: 1
15:13:07.401 -> STA disconnect: 2
15:13:07.536 -> scandone
15:13:07.536 -> state: 0 -> 2 (b0)
15:13:08.523 -> state: 2 -> 0 (2)
15:13:08.523 -> reconnect
15:13:08.523 -> wifi evt: 1
15:13:08.523 -> STA disconnect: 2
15:13:08.658 -> scandone
[... 1295 lines cut ...]
0x402103fc: esp8266::MDNSImplementation::MDNSResponder::_udpAppendBuffer(unsigned char const*, unsigned int) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1096
0x401005b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x4021056d: esp8266::MDNSImplementation::MDNSResponder::_write8(unsigned char, esp8266::MDNSImplementation::MDNSResponder::stcMDNSSendParameter&) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\libraries\ESP8266mDNS\src\LEAmDNS_Transfer.cpp line 1254
0x401005b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x401005b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x401005b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x401005b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x401005b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x401005b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x4020779d: CFastLED::show(unsigned char) at C:\Users\oao-a\Documents\Arduino\libraries\FastLED-master\src\FastLED.cpp line 47
0x40207796: CFastLED::show(unsigned char) at C:\Users\oao-a\Documents\Arduino\libraries\FastLED-master\src\FastLED.cpp line 47
0x401005b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x4020657d: ClocklessController4, 20, 50, 30, (EOrder)66, 0, false, 50>::showPixels(PixelController(EOrder)66, 1, 4294967295u>&) at C:\Users\oao-a\Documents\Arduino\libraries\FastLED-master\src/platforms/esp/8266/clockless_esp8266.h line 50
0x402066dc: CPixelLEDController(EOrder)66, 1, 4294967295u>::show(CRGB const*, int, CRGB) at C:\Users\oao-a\Documents\Arduino\libraries\FastLED-master\src/controller.h line 409
0x401005b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x40207814: CFastLED::show(unsigned char) at C:\Users\oao-a\Documents\Arduino\libraries\FastLED-master\src\FastLED.cpp line 63
0x40201798: timerMinim::isReady() at C:\Users\oao-a\AppData\Local\Temp\arduino_build_545404\sketch/timerMinim.h line 26
0x401005b8: ets_post(uint8, ETSSignal, ETSParam) at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 177
0x402164af: __yield() at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_features.h line 92
0x40201891: wdtFeedGo() at D:\YandexDisk\Нужное\Arduino\b1_CLOCK_v1.4NEW/b1_CLOCK_v1.4NEW.ino line 302
0x402021d5: effectsTick() at D:\YandexDisk\Нужное\Arduino\b1_CLOCK_v1.4NEW/effectsTick.ino line 56
0x40206aca: loop() at D:\YandexDisk\Нужное\Arduino\b1_CLOCK_v1.4NEW/b1_CLOCK_v1.4NEW.ino line 324
0x4021659c: loop_wrapper() at C:\Users\oao-a\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266\core_esp8266_main.cpp line 197

```

whakru, 2021-02-03:
and I do not understand why the gap occurs.  it never connects after this break.  as you can see in the log, there is a constant cycle of connection attempts.

d-a-v, 2021-02-04:
In the Arduino IDE, can you select tools>"generic esp8285" then tools>"Espressif FW" then second entry (+119) and retry ?

whakru, 2021-02-05:
> In the Arduino IDE, can you select tools>"generic esp8285" then tools>"Espressif FW" then second entry (+119) and retry ?

I have updated to the latest version of gid, everything is fine.

d-a-v, 2021-02-05:
> I have updated to the latest version of git, everything is fine.

Are you using the default FW version (`nonos-sdk 2.2.1+100 (190703)`) ?

whakru, 2021-02-05:
> Are you using the default FW version (`nonos-sdk 2.2.1+100 (190703)`) ?

how do I find out in git?

whakru, 2021-02-05:
The problem described above started to recur again, I closed the post early.

d-a-v, 2021-02-05:
> how do I find out in git?

In the same Arduino IDE menu.

Can you retry then with the most recent version of the closed-source FW, selectable in the Arduino MENU (*generic board* esp8266/esp8285 has to be chosen) ?

whakru, 2021-02-05:
> In the same Arduino IDE menu.
> 
> Can you retry then with the most recent version of the closed-source FW, selectable in the Arduino MENU (_generic board_ esp8266/esp8285 has to be chosen) ?

Thank you I found this, yes it was there by default. I am now starting to test +119. As you advised me earlier:
> In the Arduino IDE, can you select tools>"generic esp8285" then tools>"Espressif FW" then second entry (+119) and retry ?

This will take some time. Since this happens at random time.

whakru, 2021-02-08:
> Can you retry then with the most recent version of the closed-source FW, selectable in the Arduino MENU (_generic board_ esp8266/esp8285 has to be chosen) ?

on the +119 version, it really works more stable. but all the same, chaotic reboots on the dog are learned randomly for 0-10 hours. (the reason I have not yet taken into account).
but it looks better than it was.