esp8266/Arduino #7865: ets Jan  8 2013,rst cause:4, boot mode:(1,6) wdt reset

28Pan, 2021-02-08:
#### Platform

- Hardware:					[ESP8266|WEMOS D1]
- Core Version: [2.4.2]
- Development Env:		[Arduino IDE]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[LOLIN(WEMOS) D1 R2 & mini ]
- Flash Size: 		[4MB]
- lwip Variant:   [v2 Lower Memory]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description
I've had this problem ever since I reuploaded the code to my board since it stopped sending the barcode to firebase. I did the same procedures I did when I first uploaded the code which was disconnecting the RX TX from the other board where the barcode scanner is connected then upload the code and connect is again. The problem is after the board connects to the wifi and I scan the barcode after that I get wtd reset. 

It seems that the error is because of this part 

Firebase.setString("Barcode",a);
 for(int i = 0; i <sizeof(data); i++)
    {
      data[i] = (char)0;
    }

Please help, I really need this for my project.

[image]

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <ESP8266WiFi.h>
#include<FirebaseArduino.h>
#include <SoftwareSerial.h>
SoftwareSerial s(3,1);

#define FIREBASE_HOST "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"                     
#define FIREBASE_AUTH "XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXX"   
String a;

char data[99];
const char *ssid =  "XXXXXXXXXX";     // ssid
const char *pass =  "XXXXXXXX";   //pass ng wifi

void setup() {

  delay(3000);
  s.begin(115200);
       delay(10);
               
       s.println("Connecting to ");
       s.println(ssid); 

       WiFi.begin(ssid, pass); 
       while (WiFi.status() != WL_CONNECTED) 
          {
            delay(500);
            s.print(".");
          }
      s.println("");
      s.println("WiFi connected"); 
       Firebase.begin(FIREBASE_HOST,FIREBASE_AUTH);

}

void firebasereconnect()
{
  Serial.println("Trying to reconnect");
  Firebase.begin(FIREBASE_HOST, FIREBASE_AUTH);
  }

void loop() {

  if(s.available())
  {
    s.readBytes(data, sizeof(data));
    a = data;
    a.trim();
    s.println(a);

  Firebase.setString("Barcode",a);
  for(int i = 0; i <sizeof(data); i++)
    {
      data[i] = (char)0;
    }
  delay(1000);
  }
  delay(1000);
  
  }

  
     if (Firebase.failed())
      {
      Serial.print("setting number failed:");
      Serial.println(Firebase.error());
      firebasereconnect();
      return;
      }

  
}

```

coelner, 2021-02-08:
My first concern: https://forum.arduino.cc/index.php?topic=41009.0
```
  for(int i = 0; i < (sizeof(date) / sizeof(char)); i++)
    {
      data[i] = (char)0;
    }
```

earlephilhower, 2021-02-08:
Please move this to esp8266.com or https://gitter.im/esp8266/Arduino per the template as this involves a general programming question, a 3rd party library, external hardware, and an MCVE which can't be run by others.  Closing here.

closed 2021-02-08 by earlephilhower