esp8266/Arduino #7869: Kernel runtime measurement

gsieben, 2021-02-09:
# This is a Feature Request, but not a Bug Report. 
I hope I am in the right place. (I searched like a madman and found nothing. I asked in a bunch of forums without any answer which aims the target. I hope it is not a stupid feature request and it already exists)

## Roughly:
I need within a certain time frame the processor utilization by the kernel. Because the kernel time is a pure blind for the user. 

## In Detail:
I have programmed a time synchronized application. It runs in a fixed frequency. The time frames have to be of the same size.  Like a display that rans under certain Herz (frames per seconds). 

The requirement of the application is that a maximum possible Frequency should be reached. But now I have a big problem, because the processor is not only there for my application. It also needs time for the kernel purposes. As known the kernel will run during the waiting time of the user application. But I have no glue about the need of the kernel in terms of runtime. In the end I need the load of the processor by the kernel. The load of my program can be calculated easily. But the utilization of the kernel is unknown. For such purposes I suggest the following
A feature which allows to measure the kernel time within a certain time frame very easily. 

## It could look like this:

long returnKernelTime(); 		// Returns the kernel time within this and the last request in nanoseconds. 

## Simple Example:

A user loop which running with one Herz (looptime 1000 milli seconds) would look like this: (for convenience in milli seconds)

```
returnKernelTime(); // The value is irrelevant outside a defined time frame (in terms of overflow it would make sense to set it to zero at the very first call)
loop {
   start_usertime=millis();
       .... do user stuff....
   end_usertime=milliss();

   user_runtime=end_usertime-start_usertime;
   delay(1000-user_runtime); // without overflow (here the kernel works but how long?)

   kernel_truntime=returnKernelTime(); // Kernel Laufzeit zwischen diesen und den letzten Aufruf
   processing_runtime=kernel_runtime+user_runtime
   idle_time=1000-processing_runtime;
   
   utilization=1000/processing_runtime = 100 // in percent
   idle=1000/idle_time * 100 // in percent
}
```

This would allow me to set the frequency to an optimal ratio. With this I could reach a nearly 100% processor load without wasting real waiting time, because I can only guess how long the system needs for kernel tasks.

d-a-v, 2021-02-09:
```
auto cycles = esp_get_cycle_count();
auto usec = clockCyclesToMicroseconds(cycle);
```
In the core API:
```
#define clockCyclesToMicroseconds(a) ( (a) / clockCyclesPerMicrosecond() )
#define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )
```

gsieben, 2021-02-09:
Thanks for the quick answer. Maybe my request was not clear enough. 

But your code simple calculates the micros seconds based on the cycle counts and CPU frequency which is already covered by micros() system function. 

But I need the runtime of the kernel (core) within a certain time frame to see which minimum time is needed by the kernel to handle system things like serial.print… wifi… and so on…

gsieben, 2021-02-09:
Maybe I'll say it in another way. Maybe that will make me more understandable. 

A user writes a program inside the loop. He knows that he must also wait at one or the other place, so that the kernel (The base system, the core firmware, the bios or whatever the thing is called) does its system tasks. But the user doesn't know how long he has to wait to let the system do its work. Is a microsecond enough? Is it a hundred, a thousand? Well, there is no answer to this question. Because it depends on which basic functions the user actually uses. If he uses only a Serial.print, the system will certainly need less computing time than if he uses Serial.print, WIFI, web server. You can only guess and hope that you have given the system enough time. In case of doubt you always wait too long. Otherwise the system will not work properly. 

But if you know the runtime of the system, you do not have to wait too much. No end user can really use the processor to 100% without the knowledge about the necessary kernel runtime. 

A function that simply waits until the system has completed its tasks would also be useful. Thus one would know also automatically the system run time. That the whole thing must happen within a certain frequency, because otherwise some things like the communication could run into the timeout, is a problem of the nature of things.

earlephilhower, 2021-02-09:
Due to WiFi, interrupts, and other stuff it's not a single number.  I'm not quite sure your way would work because of this...there's no way for your app to know, say, a WiFi router you're connected to just sent a beacon and the wifi libs need to do extra stuff.

I think it's a simpler problem than all this, really.  Add `delay(0)` and the SYS context will run to completion and immediately return to the main app.  So, no wasted waiting time at all.  No need to worry about exactly how long it takes.

gsieben, 2021-02-09:
aaa... I did not know about the `delay(0)`. This corresponds to the last suggestion of my last post. With the `delay(0)` the problem is indeed solved and my request is unnecessary. Thanks for that.

closed 2021-02-09 by gsieben

gsieben, 2021-02-09:
I will write a small example program for users like me. Because with `delay(0)` you can calculate all runtimes. So including the kernel time. Like an operating system. If you want you can do it with fixed frame rate like a monitor or with free frame rate like the graphic cards. 

At least I have not seen something like this before. Otherwise I would not have made the unnecessary feature request. Sorry for the stupid question and thanks for your patience. 

I'm sure this would be useful for others as well. Is there a special place for such a code? Otherwise, I'll just throw it out there in the forums.