esp8266/Arduino #7872: Erase from an empty list with a String object causes crash

TD-er, 2021-02-12:
I have mentioned it before in issues with the String class, but I finally found out what may be happening here.
Not sure if it is something that can be fixed in the String class or maybe in the std::list implementation.
So this is rather something of a discussion issue I guess.

I do have some structs which have `String` member variables in them.
When I add them to a std::list I occasionally get crashes, all of them in the destructor of `String`.

It took a lot of debugging, but I finally have a good feeling of what may be happening here.
This is mainly a concurrency issue as I act on this list from callback functions.

So what happens?
For example in a function I try to do something like this example code below.
The `myList` is a std::list of some struct which can be constructed with a `String` object and has a `String` as member.

```c++
void myFunction() {
  String myString = F("foo");
  myList.emplace_front(myString);
}

void processScan() {
  myList.clear();
  // Check to see if we can add stuff and if not, the list is empty
}

bool SomethingToDo() {
  if (myList.empty()) return false;
  if (myList.front().isUsable()) return true;
  // Some other code
  myList.pop_front(); // <==== crash
}
```

As you can see, I did check to see if the list was empty in `SomethingToDo()`, but since `processScan()` was called from a callback, it may have been cleared in the mean time.

A simple fix for this is something like this:
```c++
if (!myList.empty()) {
    myList.pop_front();
}
```

So I guess it would make sense to have this simple check in std::list on the `_M_erase` function.
Of course this does not fix concurrency issues when you try to erase using an iterator where the pointed element is erased.

I'm not sure if this does have a deeper underlaying problem, for example when deleting an element from a list, its pointer should be set to NULL or something.
I did find this mainly in the std::list implementation, but I would be really surprised if this is the only case where things may go wrong like this.

d-a-v, 2021-02-12:
Because we don't have tasks and muteces, you may try and do this:

```cpp
void SanityCheckCallMeNotFromCallBack ()
{
  if (myListShouldBeInvalidated) 
  {
    myList.clear();
    myListShouldBeInvalidated = false;
  }
}

// called from callback / asynchronously
void processScan() {
  myListShouldBeInvalidated = true;
}

void myFunction() {
  SanityCheckCallMeNotFromCallBack();
  String myString = F("foo");
  myList.emplace_front(myString);
}

bool SomethingToDo() {
  SanityCheckCallMeNotFromCallBack();
  if (myList.empty()) return false;
  if (myList.front().isUsable()) return true;
  // Some other code
  myList.pop_front(); // <==== crash
}
```

TD-er, 2021-02-12:
This `schedule_function` does not appear to exist for ESP32, so not sure if that's a good approach for this.
Or does have the ESP32 code something similar?

d-a-v, 2021-02-12:
It's not clear whether you have this issue on both arch.
You above code is for esp32 only ? `isUsable` does not exist on this repo.
This locking issue should be there on esp32 too.
They may not have scheduled functions.
The `bool` trick should work for both archs.
The other solution for esp32 should be to use a mutex around `myList` manipulation.

TD-er, 2021-02-12:
isUsable was an example of a function of the object I stored in the list.

The issue with the crash I'm seeing is now reproducible on ESP8266, but my code should run on both ESP8266 and ESP32, so preferrably not need lots of tricks to get it to compile on both platforms.

d-a-v, 2021-02-12:
> need lots of tricks to get it to compile on both platforms.

The `bool` way is compatible with both archs. I'm sorry if I called it a trick :)

Anyway this issue should exist on any archs.
Concurrent/asynchronous accesses to shared data must be protected by some kind of facility.

TD-er, 2021-02-12:
> Concurrent/asynchronous accesses to shared data must be protected by some kind of facility.

Yep, that's why I do not really expect an immediate fix in either String class or the std::list implementation.
But I was wondering if this insight could occur on more places in the code as the `_M_erase` function in `stl_list.h` is causing these kind of crashes mainly in the `String` class and not on other objects (I am aware of)
I do know my code is doing stuff I should not let it do and in my mind I can hear the deep sigh of my chef I had at a job a long time ago :)
Still I can't yet explain what causes just the String class to behave like this.
Only explanation I can think of is that some of the memory may still be present when it tries to destruct the object that no longer exists.

d-a-v, 2021-02-12:
The general rule is that ***any*** data structure (an int or a complex class) that are accessed to by more than one context (async callback, threads, whatever) *must be protected* by something like a mutex.

The scheduled function proposal ensures that the structure is only accessed from a single thread (CONT here).
The boolean version is a bit more intrusive and less general but does the same.
We don't have muteces in the esp8266 core / sdk.

> I do not really expect an immediate fix in either String class or the std::list implementation.

There won't be such thing in `std::list`. There might be some special kind of stl container that would implicitly implement an hidden mutex but even if this is the case, they wouldn't be usable in esp8266 because we don't currently have muteces.

> Still I can't yet explain what causes just the String class to behave like this.
> Only explanation I can think of is that some of the memory may still be present when it tries to destruct the object that no longer exists.

One can not expect what happens in non-protected concurrent accesses.
I would try and check if something fixes these random crashes, even if it implies some #ifdef.

TD-er, 2021-02-12:
> I would try and check if something fixes these random crashes, even if it implies some #ifdef.

For now the check to see if the list is empty before the `pop_front()` call does fix my crashes.

What it does in the background when I think I am deleting the front item I just checked... I don't know.
So for sure I must implement some kind of mutex using the bools you suggested or add some overhead by only adding new stuff in a callback to 1 list and only delete it in another when I process it.

earlephilhower, 2021-02-12:
Looks like this is an issue of accessing an unprotected complex object in multiple contexts without a mutex protecting it, and not a core issue.  As such, not a core issue.  Closing.

closed 2021-02-12 by earlephilhower