esp8266/Arduino #7879: LCD date+météo

rateur42, 2021-02-15:
#include <LiquidCrystal_I2C.h>
#include <ESP8266WiFi.h>
#include <ArduinoJson.h>
#include <EasyNTPClient.h>
#include <WiFiUdp.h>
#include <Timezone.h>

LiquidCrystal_I2C Afficheur_G(0x27, 20, 4);
WiFiUDP Udp_G; // Objet UDP permettant d'envoyer et recevoir des trames Wi-Fi selon le protocole UDP
EasyNTPClient ClientNtp_G(Udp_G, "pool.ntp.org"); // Objet NTP synchronisé avec "pool.ntp.org"
TimeChangeRule RegleHeureEteFrance_G = {"RHEE", Last, Sun, Mar, 2, 120}; // Règle de passage à l'heure d'été pour la France
TimeChangeRule RegleHeureHiverFrance_G = {"RHHE", Last, Sun, Oct, 3, 60}; // Règle de passage à l'heure d'hiver la France
Timezone ConvertirHeureFrance_G(RegleHeureEteFrance_G, RegleHeureHiverFrance_G);
const char* ssid     = "xxxxxxxxxx";                 // SSID of local network
const char* password = "xxxxxxxxxx";                    // Password on network
String APIKEY = "xxxxxxxxxxxxxxxxxxxxxxxxxxxxx";
String CityID = "xxxxxxx";          //Your City ID
bool id = false;
WiFiClient client;
char servername[] = "api.openweathermap.org";            // remote server we will connect to
String result;

    
void Affichagetemps()
{
  if (client.connect(servername, 80))
  { //starts client connection, checks for connection
    client.println("GET /data/2.5/weather?id=" + CityID + "&units=metric&APPID=" + APIKEY);
    client.println("Host: api.openweathermap.org");
    client.println("User-Agent: ArduinoWiFi/1.1");
    client.println("Connection: close");
    client.println();
  }
  else {
    Serial.println("connection failed");        //error message if no client connect
    Serial.println();
  }

  while (client.connected() && !client.available())
    delay(1);                                          //waits for data
  while (client.connected() || client.available())
  { //connected or data available
    char c = client.read();                     //gets byte from ethernet buffer
    result = result + c;
  }

                                       //stop client
  result.replace('[', ' ');
  result.replace(']', ' ');
  //Serial.println(result);
  char jsonArray [result.length() + 1];
  result.toCharArray(jsonArray, sizeof(jsonArray));
  jsonArray[result.length() + 1] = '\0';
  StaticJsonDocument<1024> doc;
  DeserializationError  error = deserializeJson(doc, jsonArray);

  if (error) {
    Serial.print(F("deserializeJson() failed with code "));
    Serial.println(error.c_str());
    return;
  }

  String location = doc["name"];
  String country = doc["sys"]["country"];
  float temperature = doc["main"]["temp"];
  int humidity = doc["main"]["humidity"];
  float pressure = doc["main"]["pressure"];
  int id = doc["id"];
  float Speed = doc["wind"]["speed"];
  int degree = doc["wind"]["deg"];
[... 358 lines cut ...]
3ffffe60:  00000008 0000e704 3ffff3ca 3ffff3c0  
3ffffe70:  40208a7c ffffe704 3ffff3b4 3ffff3ae  
3ffffe80:  0000000f ffffe704 3ffff3a9 00000320  
3ffffe90:  3ffeeef8 ffff0008 3ffff3a6 3ffffe8c  
3ffffea0:  3ffffe5c fffb0020 3ffff39e 422e34a2  
3ffffeb0:  00007372 0000410c 3ffff39a 40e068db  
3ffffec0:  4a000000 ffff690c 3ffff396 3ffffebc  
3ffffed0:  3ffffeac fffd0020 3ffff390 3f005246  
3ffffee0:  3ffff480 82fff558 0000007d 3ffffac0  
3ffffef0:  00303032 00383038 65003331 6e754c00  
3fffff00:  00006964 614d0000 00696472 4d000000  
3fffff10:  72637265 00696465 6475654a 00000069  
3fffff20:  6e655600 64657264 61530069 6964656d  
3fffff30:  00000000 6e6e6143 20007365 86766546  
3fffff40:  00070507 00000000 00000100 3ffee9a8  
3fffff50:  3ffffac0 4094cccd 44812000 002e3748  
3fffff60:  409428f6 00000118 a0000000 40129999  
3fffff70:  c0000000 4012851e 422e34a2 40206430  
3fffff80:  40207041 00002710 3ffee9a8 3ffee9a8  
3fffff90:  3fffdad0 00000000 3ffee968 40202ad1  
3fffffa0:  3fffdad0 00000000 3ffee968 40206548  
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------
but I also have another problem my clock does not refresh every second and I have to put it in the same place as the weather forecast can you help me please

d-a-v, 2021-02-16:
> can you help me please

Yes. Use the exception decoder, [it's in the doc](https://arduino-esp8266.readthedocs.io/en/latest/Troubleshooting/stack_dump.html).

You can then read the FAQ.

If you have a problem with *your application*, you can ask for help on community forum like esp8266.com, or stackoverflow.
If you think you have found a bug in this core, feel free to open an issue *without removing the issue template - which must be filled in* and we'll be happy to track it down.

d-a-v, 2021-02-16:
Look at the [NTP example](https://github.com/esp8266/Arduino/tree/master/libraries/esp8266/examples/NTP-TZ-DST), you should have the right time with it.

closed 2021-02-16 by d-a-v

rateur42, 2021-02-16:
ok thank you for this answer I'm going to research