esp8266/Arduino #7880: DHCP Server in STA+AP mode

DedkovArtem, 2021-02-16:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [X] I have filled out all fields below.

#### Platform

- Hardware:					[ESP-12]
- Core Version:			[Core:2.7.3-3-g2843a5ac=20703003]
- Development Env:		[Platformio]
- Operating System:		[Windows]

### Settings in IDE

- Module:  			[Nodemcu]
- Flash Mode:			[qio]
- Flash Size: 		[4MB]
- lwip Variant:   [STABLE-2_1_2_RELEASE]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200]

### Problem Description

The module operates in STA+AP mode.
The station connects to the router. 2 clients are connected to the software access point: another esp8266 and a smartphone (android).

When the first client connects, the DHCP server assigns it the address 192.168.4.100.
If the second client connects in the near future (3-5 minutes), it will be assigned the address 192.168.4.101.
If the second client disconnects and 10-15 minutes pass between reconnection, the second client will be assigned the address 192.168.4.100.
The same situation will happen if time passes between the connections of the first and second clients (10-15 minutes), then the second client will be assigned the address 192.168.4.100.

This behavior occurs only if the station is connected to the router, if the router is turned off, or the module is in AP mode, then the DHCP server assigns correct addresses to clients.

If one of the clients connected to the access point is a laptop (Windows), then the assignment of addresses to new clients occurs correctly, but after disconnecting the laptop, the same addresses are assigned again (when reconnecting).

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp

#include <arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiAP.h>
#include <Wire.h>

void setup()
{
    Serial.begin(115200);

	WiFi.mode(WIFI_STA);
    WiFi.begin("ssid", "superMegaPassword");

    while (WiFi.status() != WL_CONNECTED) 
    {
        delay(1000);
    }

    IPAddress apIP(192, 168, 4, 1);
    WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0));
    WiFi.softAP("tstESP_AP", "0123456789");
}

void SendClientsCmd() 
{
  unsigned char number_client;
  struct station_info *stat_info;
  
  struct ip4_addr *IPaddress;
  IPAddress address;
  int cnt=1;
  
  number_client = wifi_softap_get_station_num();
  stat_info = wifi_softap_get_station_info();
  
  Serial.print("Connected clients: ");
  Serial.println(number_client);

  while (stat_info != NULL)
  {
      IPaddress = &stat_info->ip;
      address = IPaddress->addr;

      Serial.print(cnt);
      Serial.print(": IP: ");
      Serial.print((address));
      Serial.print(" MAC: ");

      uint8_t *p = stat_info->bssid;
      Serial.printf("%02X:%02X:%02X:%02X:%02X:%02X", p[0], p[1], p[2], p[3], p[4], p[5]);

      stat_info = STAILQ_NEXT(stat_info, next);
      cnt++;
      Serial.println();
  }
}

void loop()
{
    SendClientsCmd();
    delay(5000);
}

```

### Debug Messages

Client 1 connected
1-2 minutes passed
Client 2 connected
```
Connected clients: 2
1: IP: 192.168.4.100 MAC: F4:**:**:**:**:03
2: IP: 192.168.4.101 MAC: 80:**:**:**:**:03
```
____________________________________________

Client 1 connected
10-15 minutes passed
Client 2 connected
```
Connected clients: 2
1: IP: 192.168.4.100 MAC: F4:**:**:**:**:03
2: IP: 192.168.4.100 MAC: 80:**:**:**:**:03
```
____________________________________________

(See previous debug post)
Client 3 connected (Windows laptop)
```
Connected clients: 3
1: IP: 192.168.4.100 MAC: F4:**:**:**:**:03
2: IP: 192.168.4.100 MAC: 80:**:**:**:**:03
3: IP: 192.168.4.101 MAC: AC:**:**:**:**:5F
```
____________________________________________

(See previous debug post)
android smartphone disconnected and connected again
```
Connected clients: 3
1: IP: 192.168.4.100 MAC: F4:**:**:**:**:03
2: IP: 192.168.4.101 MAC: AC:**:**:**:**:5F
3: IP: 192.168.4.102 MAC: 80:**:**:**:**:03
```
____________________________________________

(See previous debug post)
the laptop disconnected, and then (within a few seconds) the android smartphone reconnected
```
Connected clients: 3
1: IP: 192.168.4.100 MAC: F4:**:**:**:**:03
2: IP: 192.168.4.100 MAC: 80:**:**:**:**:03
```
____________________________________________

d-a-v, 2021-02-16:
This is an interesting finding.
DHCP server source code is moved from lwip2 repository to the main repository in the git version of the core.
There's a `DHCPS_DEBUG` that can be set to 1 in `LwipDhcpServer.cpp`.
Do you plan to make further tests with this issue ?

DedkovArtem, 2021-02-16:
Optimized the code. With the 2.7.4 release, the problem described in the first message still appears.
### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiAP.h>
#include "Ticker.h"

Ticker t;

void ShowClients() 
{
  unsigned char number_client;
  struct station_info *stat_info;
  
  struct ip4_addr *IPaddress;
  IPAddress address;
  int cnt=1;
  
  number_client = wifi_softap_get_station_num();
  stat_info = wifi_softap_get_station_info();
  
  Serial.print("Connected clients: ");
  Serial.println(number_client);

  while (stat_info != NULL)
  {
      IPaddress = &stat_info->ip;
      address = IPaddress->addr;

      Serial.print(cnt);
      Serial.print(": IP: ");
      Serial.print((address));
      Serial.print(" MAC: ");

      uint8_t *p = stat_info->bssid;
      Serial.printf("%02X:%02X:%02X:%02X:%02X:%02X", p[0], p[1], p[2], p[3], p[4], p[5]);

      stat_info = STAILQ_NEXT(stat_info, next);
      cnt++;
      Serial.println();
  }
}

void EventCb(System_Event_t *evt)
{
    switch (evt->event)
    {
    case EVENT_SOFTAPMODE_STACONNECTED:
	case EVENT_SOFTAPMODE_STADISCONNECTED:
        t.once_scheduled(3, ShowClients);
        break;
    }
}

void setup()
{
    Serial.begin(115200);

    wifi_set_event_handler_cb(EventCb);

    WiFi.mode(WIFI_STA);
    WiFi.begin("ssid", "superMegaPassword");

    Serial.println("");
    while (WiFi.status() != WL_CONNECTED) 
    {
        Serial.print(".");
        delay(1000);
[... 184 lines cut ...]
dhcps: handle_dhcp-> receive a packet
dhcps: handle_dhcp-> p->tot_len = 320
dhcps: handle_dhcp-> p->len = 320
dhcps: handle_dhcp-> parse_msg(p)
dhcps: (sint16_t)*optptr = 53
dhcps: (sint16_t)*optptr = 61
dhcps: (sint16_t)*optptr = 50
dhcps: DHCP_OPTION_REQ_IPADDR != 0 err
dhcps: (sint16_t)*optptr = 54
dhcps: (sint16_t)*optptr = 57
dhcps: (sint16_t)*optptr = 60
dhcps: (sint16_t)*optptr = 12
dhcps: (sint16_t)*optptr = 55
dhcps: (sint16_t)*optptr = 255
dhcps: return s.state = 4
dhcps: handle_dhcp-> DHCPD_STATE_NAK
udhcp: send_nak>>p->ref = 1
dhcps: send_nak>>pbuf_alloc succeed
dhcps: send_nak>>p->tot_len = 548
dhcps: send_nak>>p->len = 548
udhcp: send_nak>>free pbuf
dhcps: handle_dhcp-> pbuf_free(p)
Connected clients: 1
1: IP: 192.168.4.2 MAC: F4:**:**:**:**:03
```

d-a-v, 2021-02-16:
If you can't try the git version, there's the "stage version in platformio or an [unofficial snapshot release](https://d-a-v.github.io/esp8266/Arduino). Both point a recent version of git master.

DedkovArtem, 2021-02-17:
Installed Arduino Ide 1.8.13
Installed version 0.0.2: git: '2.7.2-698-gd2f471a33'.
Set DHCPS_DEBUG to 1.
Result: The problem disappeared. The time between client connections is more than 30 minutes. Clients are assigned addresses 192.168.4.2, 192.168.4.3. (Is this correct? As far as I understand, the addresses should be 192.168.4.100, 192.168.4.101)

### Debug Messages
```
SDK:2.2.2-dev(38a443e)/Core:2.7.2-698-gd2f471a33=20702698/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:c0b69df
bcn 0
del if1
mode : sta(50:**:**:**:**:06)
scandone

.scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 47
cnt
state: 5 -> 2 (4a0)
rm 0
.reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 56
cnt

connected with ZTE :-(, channel 4
dhcp client start...
...ip:192.168.100.24,mask:255.255.255.0,gw:192.168.100.1

mode : sta(50:**:**:**:**:06) + softAP(52:**:**:**:**:06)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
dhcps:dhcps_start->udp_recv function Set a receive callback handle_dhcp for UDP_PCB pcb_dhcps
add 1
aid 1
station: f4:**:**:**:**:03 join, AID = 1
dhcps: handle_dhcp-> receive a packet
dhcps: handle_dhcp-> p->tot_len = 308
dhcps: handle_dhcp-> p->len = 308
dhcps: handle_dhcp-> parse_msg(p)
dhcps: (sint16_t)*optptr = 53
dhcps: (sint16_t)*optptr = 57
dhcps: (sint16_t)*optptr = 55
dhcps: (sint16_t)*optptr = 255
dhcps: DHCPD_STATE_OFFER
dhcps: return s.state = 1
dhcps: handle_dhcp-> DHCPD_STATE_OFFER
udhcp: send_offer>>p->ref = 1
dhcps: send_offer>>pbuf_alloc succeed
dhcps: send_offer>>p->tot_len = 548
dhcps: send_offer>>p->len = 548
udhcp: send_offer>>free pbuf
dhcps: handle_dhcp-> pbuf_free(p)
dhcps: handle_dhcp-> receive a packet
dhcps: handle_dhcp-> p->tot_len = 308
dhcps: handle_dhcp-> p->len = 308
dhcps: handle_dhcp-> parse_msg(p)
dhcps: (sint16_t)*optptr = 53
dhcps: (sint16_t)*optptr = 57
dhcps: (sint16_t)*optptr = 50
dhcps: DHCP_OPTION_REQ_IPADDR = 0 ok
dhcps: (sint16_t)*optptr = 54
dhcps: (sint16_t)*optptr = 55
dhcps: (sint16_t)*optptr = 12
dhcps: (sint16_t)*optptr = 255
dhcps: return s.state = 3
dhcps: handle_dhcp-> DHCPD_STATE_ACK
udhcp: send_ack>>p->ref = 1
dhcps: send_ack>>pbuf_alloc succeed
dhcps: send_ack>>p->tot_len = 548
dhcps: send_ack>>p->len = 548
udhcp: send_ack>>free pbuf
dhcps: handle_dhcp-> pbuf_free(p)
Connected clients: 1
1: IP: 192.168.4.2 MAC: F4:**:**:**:**:03
add 2
aid 2
station: 80:**:**:**:**:03 join, AID = 2
dhcps: handle_dhcp-> receive a packet
dhcps: handle_dhcp-> p->tot_len = 308
dhcps: handle_dhcp-> p->len = 308
dhcps: handle_dhcp-> parse_msg(p)
dhcps: (sint16_t)*optptr = 53
dhcps: (sint16_t)*optptr = 61
dhcps: (sint16_t)*optptr = 57
dhcps: (sint16_t)*optptr = 60
dhcps: (sint16_t)*optptr = 12
dhcps: (sint16_t)*optptr = 55
dhcps: (sint16_t)*optptr = 255
dhcps: DHCPD_STATE_OFFER
dhcps: return s.state = 1
dhcps: handle_dhcp-> DHCPD_STATE_OFFER
udhcp: send_offer>>p->ref = 1
dhcps: send_offer>>pbuf_alloc succeed
dhcps: send_offer>>p->tot_len = 548
dhcps: send_offer>>p->len = 548
udhcp: send_offer>>free pbuf
dhcps: handle_dhcp-> pbuf_free(p)
dhcps: handle_dhcp-> receive a packet
dhcps: handle_dhcp-> p->tot_len = 320
dhcps: handle_dhcp-> p->len = 320
dhcps: handle_dhcp-> parse_msg(p)
dhcps: (sint16_t)*optptr = 53
dhcps: (sint16_t)*optptr = 61
dhcps: (sint16_t)*optptr = 50
dhcps: DHCP_OPTION_REQ_IPADDR = 0 ok
dhcps: (sint16_t)*optptr = 54
dhcps: (sint16_t)*optptr = 57
dhcps: (sint16_t)*optptr = 60
dhcps: (sint16_t)*optptr = 12
dhcps: (sint16_t)*optptr = 55
dhcps: (sint16_t)*optptr = 255
dhcps: return s.state = 3
dhcps: handle_dhcp-> DHCPD_STATE_ACK
udhcp: send_ack>>p->ref = 1
dhcps: send_ack>>pbuf_alloc succeed
dhcps: send_ack>>p->tot_len = 548
dhcps: send_ack>>p->len = 548
udhcp: send_ack>>free pbuf
dhcps: handle_dhcp-> pbuf_free(p)
Connected clients: 2
1: IP: 192.168.4.2 MAC: F4:**:**:**:**:03
2: IP: 192.168.4.3 MAC: 80:**:**:**:**:03
```

d-a-v, 2021-03-02:
> Is this correct? As far as I understand, the addresses should be 192.168.4.100, 192.168.4.101

According to SoftAP code, it should indeed be .100+

mhightower83, 2021-03-02:
@d-a-v 
The example does not check the return code from `WiFi.softAPConfig(...`.

And, this catches my eye:
> Result: The problem disappeared. Clients are ...

I also saw problems disappear with debug options in https://github.com/esp8266/Arduino/issues/7795#issue-774830679

d-a-v, 2021-03-02:
@mhightower83 Your issue was forgotten - Thanks for the heads-up ! I tagged it
So this issue might be duplicate of #7795.

DedkovArtem, 2021-03-02:
> @d-a-v
> The example does not check the return code from `WiFi.softAPConfig(...`.
> 
> And, this catches my eye:
> 
> > Result: The problem disappeared. Clients are ...
> 
> I also saw problems disappear with debug options in [#7795 (comment)](https://github.com/esp8266/Arduino/issues/7795#issue-774830679)

softAPConfig returns true

### [MCVE](https://stackoverflow.com/help/mcve) Sketch

```cpp
#include <arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WiFiAP.h>

void ShowClients() 
{
  unsigned char number_client;
  struct station_info *stat_info;
  
  struct ip4_addr *IPaddress;
  IPAddress address;
  int cnt=1;
  
  number_client = wifi_softap_get_station_num();
  stat_info = wifi_softap_get_station_info();
  
  Serial.print("Connected clients: ");
  Serial.println(number_client);

  while (stat_info != NULL)
  {
      IPaddress = &stat_info->ip;
      address = IPaddress->addr;

      Serial.print(cnt);
      Serial.print(": IP: ");
      Serial.print((address));
      Serial.print(" MAC: ");

      uint8_t *p = stat_info->bssid;
      Serial.printf("%02X:%02X:%02X:%02X:%02X:%02X", p[0], p[1], p[2], p[3], p[4], p[5]);

      stat_info = STAILQ_NEXT(stat_info, next);
      cnt++;
      Serial.println();
  }
}

void EventCb(System_Event_t *evt)
{
    switch (evt->event)
    {
      case WIFI_EVENT_SOFTAPMODE_DISTRIBUTE_STA_IP:
      case WIFI_EVENT_SOFTAPMODE_STADISCONNECTED:
      ShowClients();
      break;
      
    default:
        break;
    }
}

void setup()
{
    Serial.begin(115200);

    wifi_set_event_handler_cb(EventCb);

    WiFi.mode(WIFI_STA);
    WiFi.begin("ssid", "superMegaPassword");

    Serial.println("");
    while (WiFi.status() != WL_CONNECTED) 
    {
        Serial.print(".");
        delay(1000);
    }
    Serial.println("");

    IPAddress apIP(192, 168, 4, 1);

    if (WiFi.softAPConfig(apIP, apIP, IPAddress(255, 255, 255, 0)))
      Serial.println("softAPConfig: True");
    else
      Serial.println("softAPConfig: False");

    if (WiFi.softAP("tstESP_AP", "0123456789"))
      Serial.println("softAP: True");
    else
      Serial.println("softAP: False");
}

void loop()
{
    
}
```

### Debug Messages
```
SDK:2.2.2-dev(38a443e)/Core:2.7.2-698-gd2f471a33=20702698/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:c0b69df
bcn 0
del if1
mode : sta(50:**:**:**:**:06)
scandone

...scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 27
cnt
state: 5 -> 2 (4a0)
rm 0
.reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 13
cnt

connected with ZTE :-(, channel 4
dhcp client start...
....ip:192.168.100.24,mask:255.255.255.0,gw:192.168.100.1

mode : sta(50:**:**:**:**:06) + softAP(52:**:**:**:**:06)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
softAPConfig: True
dhcps:dhcps_start->udp_recv function Set a receive callback handle_dhcp for UDP_PCB pcb_dhcps
softAP: True
add 1
aid 1
station: 80:**:**:**:**:03 join, AID = 1
dhcps: handle_dhcp-> receive a packet
dhcps: handle_dhcp-> p->tot_len = 308
dhcps: handle_dhcp-> p->len = 308
dhcps: handle_dhcp-> parse_msg(p)
dhcps: (sint16_t)*optptr = 53
dhcps: (sint16_t)*optptr = 61
dhcps: (sint16_t)*optptr = 57
dhcps: (sint16_t)*optptr = 60
dhcps: (sint16_t)*optptr = 12
dhcps: (sint16_t)*optptr = 55
dhcps: (sint16_t)*optptr = 255
dhcps: DHCPD_STATE_OFFER
dhcps: return s.state = 1
dhcps: handle_dhcp-> DHCPD_STATE_OFFER
udhcp: send_offer>>p->ref = 1
dhcps: send_offer>>pbuf_alloc succeed
dhcps: send_offer>>p->tot_len = 548
dhcps: send_offer>>p->len = 548
udhcp: send_offer>>free pbuf
dhcps: handle_dhcp-> pbuf_free(p)
dhcps: handle_dhcp-> receive a packet
dhcps: handle_dhcp-> p->tot_len = 320
dhcps: handle_dhcp-> p->len = 320
dhcps: handle_dhcp-> parse_msg(p)
dhcps: (sint16_t)*optptr = 53
dhcps: (sint16_t)*optptr = 61
dhcps: (sint16_t)*optptr = 50
dhcps: DHCP_OPTION_REQ_IPADDR = 0 ok
dhcps: (sint16_t)*optptr = 54
dhcps: (sint16_t)*optptr = 57
dhcps: (sint16_t)*optptr = 60
dhcps: (sint16_t)*optptr = 12
dhcps: (sint16_t)*optptr = 55
dhcps: (sint16_t)*optptr = 255
dhcps: return s.state = 3
dhcps: handle_dhcp-> DHCPD_STATE_ACK
udhcp: send_ack>>p->ref = 1
dhcps: send_ack>>pbuf_alloc succeed
dhcps: send_ack>>p->tot_len = 548
dhcps: send_ack>>p->len = 548
udhcp: send_ack>>free pbuf
dhcps: handle_dhcp-> pbuf_free(p)
Connected clients: 1
1: IP: 192.168.4.2 MAC: 80:**:**:**:**:03
```

d-a-v, 2021-03-02:
@DedkovArtem Also when `DHCPS_DEBUG` is 0 ?

DedkovArtem, 2021-03-02:
> @DedkovArtem Also when `DHCPS_DEBUG` is 0 ?

#define DHCPS_DEBUG          0

### Debug Messages
```
SDK:2.2.2-dev(38a443e)/Core:2.7.2-698-gd2f471a33=20702698/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-48-g7421258/BearSSL:c0b69df
bcn 0
del if1
mode : sta(50:**:**:**:**:06)
scandone

...scandone
state: 0 -> 2 (b0)
.state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 14
cnt
state: 5 -> 2 (4a0)
rm 0
.reconnect
state: 2 -> 0 (0)
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 49
cnt

connected with ZTE :-(, channel 4
dhcp client start...
....ip:192.168.100.24,mask:255.255.255.0,gw:192.168.100.1

mode : sta(50:**:**:**:**:06) + softAP(52:**:**:**:**:06)
add if1
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
softAPConfig: True
softAP: True
add 1
aid 1
station: 80:**:**:**:**:03 join, AID = 1
Connected clients: 1
1: IP: 192.168.4.2 MAC: 80:**:**:**:**:03
```

PR #7905 (2021-03-03): AP & dhcp-server: fix uninitialized variables
Although legacy logic is greatly improvable, at least AP now seems to run like before, offering 192.168.4.100+ addresses, whether debug mode enabled or not.
Fixes #7880 
Fixes #7795 

The difference between debug mode enabled (not `DHCPS_DEBUG` but general debug options) had an effect on one of the uninitialized variables.
closed 2021-03-03 by d-a-v