esp8266/Arduino #7885: Missing operator +() implementation after : String: compatibility with 64 bits scalars (#7863)

blue-wind-25, 2021-02-18:
### Basic Infos

- [X] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [X] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [X] I have tested that the issue is present in current master branch (aka latest git).
- [X] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:					[All]
- Core Version:			[483519f852d37abbd4fc15063b60e9763ccc62c2]
- Development Env:		[N/A]
- Operating System:		[N/A]

### Problem Description

In WString.h there are new friend functions:

       friend StringSumHelper &operator +(const StringSumHelper &lhs, long long num);
       friend StringSumHelper &operator +(const StringSumHelper &lhs, unsigned long long num);

The function implementations seem to be missing from WString.cpp

earlephilhower, 2021-02-19:
@blue-wind-25 do you have a snippet that causes a compilation error?  Please post it so we have a good MCVE.

blue-wind-25, 2021-02-19:
@earlephilhower no, because those functions are not actually used by any of the Arduino esp8266 core or library code.

I follow the repository because I use it as a refererence in my customized library that use different built system.

When I browsed the Arduino commit change, I found that those functions are declared, but seems to be not defined anywhere. So, it is not an actual bug report, but more like an information. I am sorry if this is the wrong place for this kind of information.

earlephilhower, 2021-02-19:
No, it's a bug if we declare a function that's never implemented!  Thanks for the notice!  I was just hoping to have a premade test case to verify the fix.

blue-wind-25, 2021-02-19:
@earlephilhower You are welcome :-)

PR #7888 (2021-02-19): String: add missing long long operator implementations
fixes #7885
closed 2021-02-19 by earlephilhower