esp8266/Arduino #7886: Hard resetting via RTS pin...

SKGEzhil, 2021-02-18:
### Basic Infos

- [ ] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ ] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [ ] I have tested that the issue is present in current master branch (aka latest git).
- [ ] I have searched the issue tracker for a similar issue.
- [ ] If there is a stack dump, I have decoded it.
- [ ] I have filled out all fields below.

#### Platform

- Hardware:				[ESP-01]
- Core Version:			[latest]
- Development Env:		[Arduino IDE]
- Operating System:		[Ubuntu]

### Settings in IDE

- Module:  			[Generic ESP8266 Module]
- Flash Mode:			[DOUT]
- Flash Size: 		[1MB]
- lwip Variant:   [v2 Lower Memory]
- Reset Method:		[nodemcu]
- Flash Frequency:	[40Mhz]
- CPU Frequency:		[80Mhz]
- Upload Using:		[SERIAL]
- Upload Speed:   [115200] (serial upload only)

### Problem Description

I uploaded the blink sketch from examples. It is not working. While uploading it is showing that
```
Leaving...
Hard resetting via RTS pin...
```
I couldn't see any output in the serial monitor and the led was also not blinking

### Sketch

```cpp

/*
  ESP8266 Blink by Simon Peter
  Blink the blue LED on the ESP-01 module
  This example code is in the public domain

  The blue LED on the ESP-01 module is connected to GPIO1
  (which is also the TXD pin; so we cannot use Serial.print() at the same time)

  Note that this sketch uses LED_BUILTIN to find the pin with the internal LED
*/

void setup() {
  pinMode(LED_BUILTIN, OUTPUT);     // Initialize the LED_BUILTIN pin as an output
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, LOW);   // Turn the LED on (Note that LOW is the voltage level
  // but actually the LED is on; this is because
  // it is active low on the ESP-01)
  delay(1000);                      // Wait for a second
  digitalWrite(LED_BUILTIN, HIGH);  // Turn the LED off by making the voltage HIGH
  delay(2000);   
}

```

Gbro30, 2021-02-18:
After uploading the sketch, you have to ensure your device is restarted without GPIO0 connected to ground (flash mode).

SKGEzhil, 2021-02-18:
> 
> 
> After uploading the sketch, you have to ensure your device is restarted without GPIO0 connected to ground (flash mode). 
> 
> 

GPIO0 is not connected to ground while flashing. I just connected it at the start then removed while flashing. Then also it's not working. Any other solutions??

d-a-v, 2021-02-18:
At boot,
- gpio0 must be up,
- gpio2 must be up,
- gpio15 must be down.

JiriBilek, 2021-02-18:
Are you sure the upload finished ok? Switch on the checkbox in the IDE to be sure: _Preferences - Show verbose output during - upload_

earlephilhower, 2021-02-19:
This isn't related to the core.  Please move such general questions to https://esp8266.com or https://gitter.im/esp8266/Arduino .  Closing.

closed 2021-02-19 by earlephilhower