esp8266/Arduino #7893: Broken link Online Help in Arduino IDE

Bighoneypot, 2021-02-25:
After install esp8266 on Arduino Ide, link Online Help is broken.

[image]

and open this

[image]

Tech-TX, 2021-02-26:
A better screencap would have helped, like for instance one that included the URL you're being sent to.

That URL is coming from https://arduino.esp8266.com/stable/package_esp8266com_index.json, specifically
`          "help": {
            "online": "https://arduino-esp8266.readthedocs.io/en/2.7.3/"`

Looks like ReadTheDocs hasn't kept the older versions; weird, as I thought it did.
It also looks like there's a blank entry for the 2.7.4 version, so it's falling through down to the 2.7.3 online help.

d-a-v, 2021-02-26:
Thanks for reporting.
2.7.3 and 2.7.4 are bugfix releases and were not processed as other released.
It seems that we'll have to fix this for next times.
2.7.3 and 2.7.4 links are https://arduino-esp8266.readthedocs.io/en/2.7.3_a and https://arduino-esp8266.readthedocs.io/en/2.7.4_a.