esp8266/Arduino #7915: NoSuchFile_Nested to deep

xnordens, 2021-03-10:
Arduino:1.8.13 (Windows 10), Kort:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"
C:\Users\Peter\Arduino\libraries\ESP8266WiFi\src\ESP8266WiFiAP.cpp:40:28: fatal error: LwipDhcpServer.h: No such file or directory
 #include "LwipDhcpServer.h"
compilation terminated.
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).
I download the file from GitHub but where to put the missing file. I have tried to put the file in many libs but I always got To deep nested. The program is strippen down to only #include <ESP8266WiFi.h> and I cannot find out what's the problem. I have use this lib for long time and it has worked before. I am very frustrated. I have reinstalled both IDE  and the libraries

d-a-v, 2021-03-10:
Your installation is corrupted. This file appeared after release 2.7.4, only available with the git version until next release. You need to reinstall cleanly the esp8266 Arduino core.
You may first backup your libraries, then
https://forum.arduino.cc/index.php?topic=362507.0
then reinstall arduino then the esp8266 arduino core (v2.7.4 or git version).

closed 2021-03-10 by d-a-v

xnordens, 2021-03-11:
Thank you very much. I've been messing with this for a week. What should I do to prevent this from happening more than once?

d-a-v, 2021-03-11:
Can't say.

What should be remembered is when logic is challenged, a real and full start-over may be needed :)