esp8266/Arduino #7917: No such file or directory

xnordens, 2021-03-11:
Arduino:1.8.13 (Windows Store 1.8.42.0) (Windows 10), Kort:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, 115200, 4M (3M SPIFFS)"
D:\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.4.0-rc2\libraries\Hash\src\Hash.cpp:26:34: fatal error: bearssl/bearssl_hash.h: No such file or directory
 #include <bearssl/bearssl_hash.h>
                           
compilation terminated exit status 1 Error compiling for board NodeMCU 1.0 (ESP-12E Module).

I was in here before with (#7915) I reinstalled everything. and it was OK for a while but later I've got this above. I have put the the file in several dir but the same proproblem. Once I had it in two places and got Found dubble files. Used this...

d-a-v, 2021-03-11:
Try with the latest version of this core. You are using 2.4.0-rc2. Latest is 2.7.4.
The parameter is in the Arduino IDE board manager.

xnordens, 2021-03-11:
Now:
Arduino:1.8.13 (Windows Store 1.8.42.0) (Windows 10), Kort:"NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), 2, v2 Lower Memory, Disabled, None, Only Sketch, 115200"
d:/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Peter\AppData\Local\Temp\arduino_cache_666980\core\core_435d0f3e16965c2651d354c13b2b2311.a(core_esp8266_main.cpp.o): in function `__loop_end':
D:\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_main.cpp:185: undefined reference to `setup'
d:/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Peter\AppData\Local\Temp\arduino_cache_666980\core\core_435d0f3e16965c2651d354c13b2b2311.a(core_esp8266_main.cpp.o): in function `loop_wrapper()':

D:\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_main.cpp:189: undefined reference to `setup'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

xnordens, 2021-03-12:
After again I have unistalled 2.7.4. and reinstalled 2.7.4. I have these problems: 

d:/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Peter\AppData\Local\Temp\arduino_cache_670523\core\core_435d0f3e16965c2651d354c13b2b2311.a(core_esp8266_main.cpp.o): in function `__loop_end':
D:\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_main.cpp:185: undefined reference to `setup'
d:/documents/arduinodata/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-4-b40a506/bin/../lib/gcc/xtensa-lx106-elf/4.8.2/../../../../xtensa-lx106-elf/bin/ld.exe: C:\Users\Peter\AppData\Local\Temp\arduino_cache_670523\core\core_435d0f3e16965c2651d354c13b2b2311.a(core_esp8266_main.cpp.o): in function `loop_wrapper()':
D:\Documents\ArduinoData\packages\esp8266\hardware\esp8266\2.7.4\cores\esp8266/core_esp8266_main.cpp:189: undefined reference to `setup'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

xnordens, 2021-03-12:
It fixed. Simple problem. I had dimmed to much program rows in the file. void setup () was not included in the program.

closed 2021-03-12 by xnordens