esp8266/Arduino #7918: Limitting to TLS 1.2

manolodd, 2021-03-12:
Hi,

I've been analysing the wifi secure client looking for a way of limitting the use of TLS of the engine to TLS 1.2 only.

In this example from bear ssl, there is that option that allow the engine to be configured tu use only TLS versions between the specified range.

https://www.bearssl.org/gitweb/?p=BearSSL;a=blob;f=samples/custom_profile.c;h=e88d948e43ce05123e1d9af78651c3f6e754de8f;hb=f3ac21cb00646600475b02ecd2c592ee36e417da

But I am not able to find a similar method in wifi secure client. "Only" setCipher() is avilable but I cannot limit the protocol to be used.

Am I wrong? Is there any way of configurion the wifi secure client to use only TLS 1.2?

Best regards

earlephilhower, 2021-03-12:
There is no way presently for the core to specify this, however it is possible in BSSL to do so.

https://bearssl.org/apidoc/bearssl__ssl_8h.html#aca341c90ed49000ec1560101d6794e07

A PR would be appreciated, and looks relatively simple.  A new call, similar to setCiphers() would be needed.

manolodd, 2021-03-12:
Ok, thank you. I have found tls1.0 to tls 1.2 set by default in wifi secure client initialization.

I don't know if I am able to modify the code to do so, but i'll try.

Having tls 1.0 and 1.1 configured by default with no option to change can be insecure because of the known vulnerabilities of those versions.

Thank you again

PR #7920 (2021-03-15): Add setSSLVersion call to SSL object
Allow users to only allow specific TLS versions for connections with an
additional call in their app, similar to the setCiphers call.

Fixes #7918
commit 7475ba7f (2021-03-15): Add setSSLVersion call to SSL object (#7920)

* Add setSSLVersion call to SSL object

Allow users to only allow specific TLS versions for connections with an
additional call in their app, similar to the setCiphers call.

Fixes #7918

* Add SSL level options to WiFiServerSecure
closed 2021-03-15 by earlephilhower