esp8266/Arduino #7931: Set WiFi channel on ESP8266MOD

fabiuz7, 2021-03-18:
Hi,
sorry if this question is not fully appropriated, but on ESP8266MOD, if I try to set the wifi channel, it always remains 1 (I'm testing with a simple wifi scanner for android). If I try with the modern ESP-WROOM-02D or other non-packaged esp8266 and I can set whichever channel I prefer without any trouble. 

Is it a known problem?

Tech-TX, 2021-03-20:
You've deleted the issue template and ignored the requirements needed for anyone to help.  Without an MCVE (Minimal Complete Verifiable Example) there's little anyone can do but play a game of "20 Questions".

By the way, nearly everyone in the ESP world calls it an ESP-12, either -12E or 12F.  I had to google it to figure out what the heck an ESP8266MOD was.

The only thing I can see _somewhat_ mimics what you're describing.  When I take a canned example using WIFI.softAP to control an LED on a D1 Mini, the access point doesn't show up on my Android phone WiFi scanner when I add any of the optional variables to the function (int channel, bool hide SSID, int max connections), although it _does_ show on my laptop.  Trying variations on a theme, one time it used channel 8 (I'd previously connected that D1 Mini to my router on channel 8), even though I'd put channel 12 in the command.

closed 2021-03-21 by devyte

fabiuz7, 2021-03-21:
Sorry for the poor issue, but finally I can tell you that I was victim of a counterfeit ESP-12E. I have tested many other original modules and they can switch channel without any problem. The script to test this behavior is very simple:

```
#include <ESP8266WiFi.h>
#include <WiFiClient.h>

const char *ssid = "MYSSID123";
const char *password = "mypassword";
int channel = 6;

void setup() {
  Serial.begin(115200);
  Serial.print("Configuring access point...");
  WiFi.softAP(ssid, password, channel);
}

void loop() {}
```

Then you can check the actual channel with a laptop, smartphone or another ESP.