esp8266/Arduino #7937: Can't compile NTP-TZ-DST.ino

jov58, 2021-03-24:
Hi, 

I'm trying to compile [NTP-TZ-DST.ino](https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino), but for this line:

`settimeofday_cb(time_is_set);`

I get an error:

`no matching function for call to 'settimeofday_cb(void (&)(bool))'`

In coredecls.h I find:

```
using TrivialCB = std::function<void()>;
void settimeofday_cb (TrivialCB&& cb);
void settimeofday_cb (const TrivialCB& cb);
```

No callback function with bool parameter is provided. I'm using PlatformIO with esp8266 v 2.6.3.

I can't figure out how to compile this sketch, am I missing something here?

d-a-v, 2021-03-24:
You probably mixed up several versions of the core in your installation.
Try to reinstall. If it persists, completely remove the esp8266 arduino core ("arduino15"/packages/esp8266).

devyte, 2021-03-24:
Closing due to user error.

closed 2021-03-24 by devyte

jov58, 2021-03-24:
What is a user error and why is this closed?

mcspr, 2021-03-24:
Amending the above, [link should be for the 2.7.4 tag / branch](https://github.com/esp8266/Arduino/blob/2.7.4/libraries/esp8266/examples/NTP-TZ-DST/NTP-TZ-DST.ino)

PIO refers to its own version -> https://github.com/platformio/platform-espressif8266/releases, but it is Core 2.7.4 as seen in the build logs:
```
> pio run
Processing stable (platform: espressif8266; board: d1_mini; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/d1_mini.html
PLATFORM: Espressif 8266 (2.6.3) > WeMos D1 R2 and mini
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 3.20704.0 (2.7.4) <-------------------
 - tool-esptool 1.413.0 (4.13)
 - tool-esptoolpy 1.30000.201119 (3.0.0)
 - toolchain-xtensa 2.40802.200502 (4.8.2)
```
*(...my earlier e-mail reply got lost...)*

jov58, 2021-03-24:
Anyhow, after using the updatedd link mcspr provided, I tried that and it gives me the correct date and time.
Thx for pointing me in the right direction!

d-a-v, 2021-03-24:
> > PLATFORM: Espressif 8266 (2.6.3) > WeMos D1 R2 and mini
> > Amending the above, link should be for the 2.7.4 tag / branch

> after using the updated link provided by mcspr, I tried that and it gives me the correct date and time

It appears that current core maintainers do not understand these statements which apparently successfully help PIO users 👀 😅 

Luckilly we speak of the same source code at least we have that in common 😎 😇

mcspr, 2021-03-26:
> It appears that current core maintainers do not understand these statements which apparently successfully help PIO users 👀 😅

> Luckilly we speak of the same source code at least we have that in common 😎 😇

Both system, sadly, obfuscate the build process in some random places. There's also a github issue at play with the versioning, but that's another question altogether.

And I really don't get the maintainer position then, PIO integration was around here for 4y and no-one even looked at it to understand it at least on the base level :(

d-a-v, 2021-03-26:
> no-one even looked at it to understand it at least on the base level :(

I can tell for sure that *at least* two of them (out of 3) tried at least one time to work with PIO.
I guess that the complexity is too high where a single makefile can make it, or the guys are too old.
The official Arduino IDE is supported and the PIO python helper is also updated when necessary.
As you know maintainers do that in their free time. Any contributor is welcome to improve documentation about anything, including everything that is related with PIO.

PIO folks are *also* official maintainers on this repository btw (they can merge PR and they do sometimes). 

All requests that would ease PIO environment have been welcome and helped.
The active 3 maintainers are just *unable* to help with everyday PIO issues.
The esp8266 arduino gitter is the right place to ask about this, some of the users are really aware about how PIO work.

@mcspr please point us to the versioning issue is there's one.

mcspr, 2021-03-26:
The original answer pointed to the wrong version, specifically b/c of the misunderstanding of the PIO platforms versioning vs. the Arduino Core, creating another series of misunderstandings :) Not the first time I've seen it for both platform and Core issues, this is not a Core specific or a maintainer problem. And please do not take this as an attack, it's just an observation on the state of things

It does feel like more of a translation issue, both languages of the build tools and the people themselves :) Arduino IDE builder is just the another complicated beast, maybe a more familiar one.

d-a-v, 2021-03-26:
@mcspr are you saying that, in OP,
> I'm using PlatformIO with esp8266 v 2.6.3.

is not really what *I* understand ?

> And please do not take this as an attack, it's just an observation on the state of things

Sorry, that was not meant. We do answer to issues because that is our duty when we accepted the "job".
If there's someone we can count on with the ability to answer to PIO-user questions, I agreee that it would help alot with this issue tracker.

d-a-v, 2021-03-26:
> @mcspr please point us to the versioning issue if there's one.

Well sorry @jov58  [I think](https://github.com/platformio/platform-espressif8266/releases/tag/v2.6.3) I just [got it](https://github.com/esp8266/Arduino/releases/tag/2.6.3) 😓

mcspr, 2021-03-26:
Hence my point, you don't see Core version in the VS Code PlatformIO (which is the most common case), the version on the forefront is of the platform files aka build tools. There's also an examples tab which is just found out about, which will retrieve the example code that comes with the Core tarball, but it is not really apparent unlike the Arduino alternative.

Maybe it is for the best to delegate PIO questions to the community with the PIO experience? You may also close the question and refer to either https://community.platformio.org/ or platform-espressif8266?