esp8266/Arduino #7946: OTA stopped working with new router

SteveRMann, 2021-03-26:
https://github.com/esp8266/Arduino/issues/new
----------------------------- Delete below -----------------------------
----------------------------- Delete above -----------------------------

### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [ x I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] I have filled out all fields below.

#### Platform

- Hardware:				Wemos D1 Mini, NodeMCU
- Core Version:			2.7.4
- Development Env:		Arduino IDE
- Operating System:		Windows

### Settings in IDE

IDE settings
 Board: "LOLIN(WEMOS) D1 R2 & mini"
 Upload Speed: "921600"
 CPU Frequency: "80 MHz"
 Flash Size: "4MB (FS:2MB OTA:-1019KB)"
 Debug port: "Disabled"
 Debug Level: "None"
 IwIP Variant: "v2 Lower Memory"
 VTables: "Flash"
 Exceptions: "Legacy (new can return nullptr)"
 Erase Flash: "Only Sketch" (I also tried Sketch + WiFi settings)
 SSL Support: "All SSL ciphers (most compatible)"
 Port: "esp8266-fb00a5 at 192.168.1.76"

### Problem Description

I have a project on a Wemos D1 Mini that has been running for a year or so with occasional updates by OTA.
The device, by the way, receives and displays data over MQTT, so its WiFi is working. Except for OTA.

This morning I did another update- or I tried- but OTA fails with:
[ERROR]: No response from device

I spent the morning trying to identify if anything in my sketch could have caused this, but nothing looked suspicious. I tried reducing my program to a simple test, but still no OTA. Next, I tried running the BasicOTA example sketch. Again, no connection. I tried on a new Wemos D1 Mini and a new NodeMCU board. No connection.

ESPHome, however, does upload OTA.

The only thing that has changed on my network is that I replaced my router two weeks ago. The old router did not do WPA (yes, it's that old) and I couldn't use the latest Tasmota or ESPHome without WPA2.

###  Sketch

#include <ESP8266WiFi.h>
#include <ESP8266mDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>

#ifndef STASSID
#define STASSID "your-ssid"
#define STAPSK  "your-password"
#endif

const char* ssid = STASSID;
const char* password = STAPSK;

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }

  // Port defaults to 8266
  // ArduinoOTA.setPort(8266);

  // Hostname defaults to esp8266-[ChipID]
  // ArduinoOTA.setHostname("myesp8266");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA.onStart([]() {
    String type;
    if (ArduinoOTA.getCommand() == U_FLASH) {
      type = "sketch";
    } else { // U_FS
      type = "filesystem";
    }

    // NOTE: if updating FS this would be the place to unmount FS using FS.end()
    Serial.println("Start updating " + type);
  });
  ArduinoOTA.onEnd([]() {
    Serial.println("\nEnd");
  });
  ArduinoOTA.onProgress([](unsigned int progress, unsigned int total) {
    Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
  });
  ArduinoOTA.onError([](ota_error_t error) {
    Serial.printf("Error[%u]: ", error);
    if (error == OTA_AUTH_ERROR) {
      Serial.println("Auth Failed");
    } else if (error == OTA_BEGIN_ERROR) {
      Serial.println("Begin Failed");
    } else if (error == OTA_CONNECT_ERROR) {
      Serial.println("Connect Failed");
    } else if (error == OTA_RECEIVE_ERROR) {
      Serial.println("Receive Failed");
    } else if (error == OTA_END_ERROR) {
      Serial.println("End Failed");
    }
  });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}

void loop() {
  ArduinoOTA.handle();
}
### Debug Messages

```
Arduino: 1.8.13 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

Executable segment sizes:
IROM   : 281812          - code in flash         (default or ICACHE_FLASH_ATTR) 
IRAM   : 27828   / 32768 - code in IRAM          (ICACHE_RAM_ATTR, ISRs...) 
DATA   : 1276  )         - initialized variables (global, static) in RAM/HEAP 
RODATA : 1212  ) / 81920 - constants             (global, static) in RAM/HEAP 
BSS    : 25472 )         - zeroed variables      (global, static) in RAM/HEAP 
Sketch uses 312128 bytes (29%) of program storage space. Maximum is 1044464 bytes.
Global variables use 27960 bytes (34%) of dynamic memory, leaving 53960 bytes for local variables. Maximum is 81920 bytes.

16:31:32 [ERROR]: No response from device
16:31:32 [ERROR]: No response from device

```

SteveRMann, 2021-03-28:
Here's an update.
Apparently the router isn't the problem. I downloaded the IDE to an old laptop and uploaded BasicOTA by USB and then again using OTA- It works as expected.
Tried it on a second PC- still working.

So OTA isn't working from my main PC. The fail is apparently just coincident with my router upgrade which is why I suspected the router.

Should I delete and reinstall the IDE and the esp8266 core, or does anyone think the problem is somewhere else?

earlephilhower, 2021-03-28:
I bet the firewall is blocking your main PC.  The router change caused Windows to think it was a new network, and  the firewall settings aren't permitting incoming connections on the needed ports.

In any case, closing as this is not a core issue.  Good luck!

closed 2021-03-28 by earlephilhower

SteveRMann, 2021-03-28:
You're probably right that it isn't a core issue.
God I hate Windows....
It's odd that an identical Windows 10 PC initiates OTA just fine, but at least you have given me something to investigate.

SteveRMann, 2021-03-29:
SOLVED
It turns out that my network type had been changed to "Public" (I'm not sure why I would have done that). Making the network type "Private" fixed OTA.  The clue was that turning off the public firewall allowed OTA to complete.