esp8266/Arduino #7950: Running out of heap (and thus crashing) during filesystem update will corrupt the filesystem

timkoers, 2021-03-29:
### Basic Infos

- [x] This issue complies with the [issue POLICY doc](https://github.com/esp8266/Arduino/blob/master/POLICY.md).
- [x] I have read the documentation at [readthedocs](https://arduino-esp8266.readthedocs.io/en/latest) and the issue is not addressed there.
- [x] I have tested that the issue is present in current master branch (aka latest git).
- [x] I have searched the issue tracker for a similar issue.
- [x] If there is a stack dump, I have decoded it.
- [x] I have filled out all fields below.

#### Platform

- Hardware:					ESP-12
- Core Version:			2.6.3
- Development Env:		PlatformIO
- Operating System:		Windows

### Settings in IDE

- Module:  			nodemcuv2
- Flash Size: 		4MB
- CPU Frequency:		160MHz
- Upload Using:		Serial
- Upload Speed:   921600

### Problem Description

When the ESP is (almost) out of heap memory, and crashes during an update, the filesystem or files get corrupted, sometimes rendering the internal webserver useless.

## How to reproduce
Upload the standard filesystem + sketch via USB. 
Open the serial monitor, drag the filesystem image onto the upload button. 
Before you click Upload to fs, make sure to visit `ESP_IP/fill_memory` a couple of times until the free heap memory reaches around 7000 bytes. 
Click the upload to fs button, you might want to refresh the `ESP_IP/fill_memory` page one or two times during the upload, but it should also work without that.

Watch the ESP crash due to the free heap size available and after a reboot, you'll see that the LittleFS is corrupted.

Please note that the filesystem reports that index.html is 3994 bytes both before and after the update, so the filesystems file table is still intact.

Due to the OTA partition of the ESP, I don't think that this will happen during a Flash upgrade, but please feel free to test.

## Sketch

```cpp

#include <Arduino.h>

#include <AsyncJson.h>
#include <ESP8266WiFi.h>
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>

void setup() {

// Connect to wifi
WiFi.begin(SSID, PASS);
server.begin();

    // Route for root / web page
    server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
      request->send(LittleFS, "/index.html");
    });

    server.on("/fill_memory", HTTP_GET, [](AsyncWebServerRequest *request) {
      // Fill up the memory
      Serial.println("Filling memory");
      fillMemory = true;
      request->send(200, "Ok");
    });

    server.on(
[... 869 lines cut ...]
    <br>
    <div style="display:inline-block">
        <button class="button" onclick="upload_ota('')" style="float:left; margin-right:5px;">Upload</button>
        <button class="button" onclick="upload_ota('SPIFFS')" style="float:right; margin-left:5px;">Upload to
            FS</button>
    </div>

<script>    
    function upload_ota(type) {
        var typeArgument = type != "" ? "?SPIFFS" : "";

        var xhttp = new XMLHttpRequest();
        xhttp.onreadystatechange = function () {
            //if (this.readyState == 4 && this.status == 200) {
            //    // TODO: Check how many bytes have been transmitted and calculate the progress based on that
            //    // State object.
            //    var data = JSON.parse(this.responseText);
            //    ota_finished();
            //} else {
            //    document.getElementById("overlay").style.display = "none";
            //}
        };
        xhttp.open("POST", `/update${typeArgument}`, true);
        xhttp.se����������������������������������������������������������������������������������������������������������������������������������������������������������</script></body>
```

earlephilhower, 2021-03-29:
Closing as this is expected behavior.  It overwrites the FS directly since in may cases the FS is way larger than the standard OTA space.

Use the `ATOMIC_FS_UPDATE` define in the OTA to make it use the OTA space, but then your largest FS will only be the size of the OTA space.

timkoers, 2021-03-29:
@earlephilhower The ESP continues to function with 2000 bytes of free memory, until the update starts and is almost completed. I agree that this is the expected behaviour, but a check if there is enough free heap memory to even process the update is a must in my opinion.

Pablo2048, 2021-03-29:
I don't think so - you can always check the heap space before doing the update in your code, but according to your log it seems like you "eat the memory" when the update was in progress, so it's practically impossible to avoid this crash...

timkoers, 2021-03-29:
@Pablo2048 Nope, the crash also occured on a system where I did not eat memory and the memory is only been eaten once. Running the Updater asynchronously creates the scenario that heap usage could be changed during the update. This is something that the library should account for as it can potentially brick (end user) devices

Pablo2048, 2021-03-29:
So ok, lets say that we have the following scenario - the update is running, half of the filesystem is already overwritten (on the fly) and now the heap disappeared - we just not have enough memory. What do you thing has to be done? The filesystem is already corrupted (half overwriten). The only way I can see is by using @earlephilhower suggestion - ATOMIC_FS_UPDATE - when download can not be finished, filesystem cannot be overwriten. Unfortunately the maximum size of the filesystem is the free OTA area, so if you have 2M/2M you are out of luck.

Pablo2048, 2021-03-29:
Ehm, and the crash you have posted seems to be in the AsyncWebServer - he is unable to allocate some strings when he tries to add some headers...

earlephilhower, 2021-03-29:
It's impossible to guarantee no memory overflow while doing OTA.  There's a lot going on in the system with Wifi, external IP connections (as shown in your crash), etc.  So that is why if you really care you use the ATOMIC option.

Also, you can use `gzip` to compress your filesystem w/ATOMIC (not with live update).  So the compressed filesystem needs to fit in the OTA area but can decompress to a much larger one (i.e. an empty LittleFS 3MB FS is <100KB compressed IIRC).

timkoers, 2021-03-29:
> 
> 
> So ok, lets say that we have the following scenario - the update is running, half of the filesystem is already overwritten (on the fly) and now the heap disappeared - we just not have enough memory. What do you think has to be done? The filesystem is already corrupted (half overwriten). The only way I can see is by using @earlephilhower suggestion - ATOMIC_FS_UPDATE - when download can not be finished, filesystem cannot be overwriten. Unfortunately the maximum size of the filesystem is the free OTA area, so if you have 2M/2M you are out of luck.

Yeah I got you, sorry, misunderstood your last post.

> It's impossible to guarantee no memory overflow while doing OTA. There's a lot going on in the system with Wifi, external IP connections (as shown in your crash), etc. So that is why if you really [...]

Is the atomic behaviour guaranteed when using decompression, so that nothing will be changed if the ESP crashes when decompressing?

earlephilhower, 2021-03-29:
ATOMIC uses the same stuff as the program OTA and runs right at bootup before any app code is run.  So it's guaranteed to be atomic *except in the case of power failure*.  That's one niggle that's left, and it applies to FS and app.

devyte, 2021-03-30:
To clarify:
When the FS is being updated with atomic, the eboot sector isn't touched. If power fails during decompression, I think that on next boot up the decompression should just restart, so for this case you can't end up with a corrupted system.
For app update with atomic, if the eboot sector changed in the firmware (which is super rare) AND you get a power failure before the 4KB eboot sector is finished updating, then you get a bricked device. The chance of that happening is very very low. But yes, it's a niggle that is still pending, i. e. @davisonja's #6538.

timkoers, 2021-03-30:
> So ok, lets say that we have the following scenario - the update is running, half of the filesystem is already overwritten (on the fly) and now the heap disappeared - we just not have enough memory. [...]

Yeah I got you, sorry, misunderstood your last post.

> It's impossible to guarantee no memory overflow while doing OTA. There's a lot going on in the system with Wifi, external IP connections (as shown in your crash), etc. So that is why if you really [...]

Is the atomic behaviour guaranteed when using decompression, so that nothing will be changed if the ESP crashes when decompressing?

> So ok, lets say that we have the following scenario - the update is running, half of the filesystem is already overwritten (on the fly) and now the heap disappeared - we just not have enough memory. [...]

Would it be an idea to add something to LittleFS/Updater that checks if the update was interrupted due to a crash or something?
Store a flag in the RTC memory for example?

Pablo2048, 2021-03-30:
> Would it be an idea to add something to LittleFS/Updater that checks if the update was interrupted due to a crash or something?
> Store a flag in the RTC memory for example?
This can be done in your application code - I'm strongly against using the RTC area internally by some library without some sort of central management. For example I'm using portion of RTC area for counting unexpected crashes and forcing emergency mode when the application crashes too often to avoid bricking the device. So yes - you can use the RTC memory to mark unsuccessful update, but it wont survive power failure (EEPROM would be better?).