esp8266/Arduino #7964: "Blink"-Sketch is not running on my D1 Mini Pro

Mathias2803, 2021-04-06:
I was able to flash several "D1 Mini V3"  with different Sketches and I never had a problem. But due to the external antenna I decided to get a "D1 Mini **Pro**".

I am able to flash the sketch on the board without error messages, **but then nothing is happening**.

I even tried the "Blink" example code, but even with this program I had no success.
When I push the reset button with the connected serial output I get following message:

"
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16 
tail 0
chksum 0x55
csum 0x55
csum err
ets_main.c 
"

Here you can see my Arduino IDE settings:
[image]

Can anybody help me what I am doing wrong?

Thanks a lot and kind regards Mathias
****

d-a-v, 2021-04-06:
Try a full flash erase ("Erase flash" entry).
Also check with esptool.py that your flash is 16MBytes.

Mathias2803, 2021-04-07:
Hello, thanks for your feedback.

The esptool.py gives out following information, so the flash size seems to be 16MB:
[image]

I also made a a full flash erase as you can see in my settings now:
[image]

But after the restart still the same error message and even the example skretch "Blink" is not running:
[image]

What else can I do to get it running?

Thanks a lot, kind regards Mathias

Jason2866, 2021-04-08:
Try a slower upload speed. I have several devices not working with 921600.
You can try with 115200

Mathias2803, 2021-04-09:
I have changed the upload speed to 115200 but still the same error message at the end.

[image]

Jason2866, 2021-04-09:
There is a high chance that the board is just defect.

d-a-v, 2021-04-09:
Another test would be to try another *flash mode* by selecting the "esp8266 generic board".
D1 is by default DIO, you could try DOUT.

devyte, 2021-04-09:
Try also:
- without FS (the ingo above shows 14MB FS)
- selecting 4MB flash size (your flash chip may be fake despite being detected as 16MB)

Tech-TX, 2021-04-10:
I'd bet money on the Flash mode David mentioned above. :+1:   I wasn't aware that it defaulted to DIO; I thought most of them were DOUT.  Since there are probably more cheap clone D1 Mini Pro boards than official Lolin boards, maybe I should do a PR to switch that to DOUT.  The error message is a CRC error, which  happens if the chip is in a mode that it can't support at 26MHz.  I have a clone D1 Mini Pro V1.x and a couple of real Lolin D1 Mini Pro V2.x boards that work with the current defaults, but that's no guarantee that a clone with cheap Flash from _Wo Fat Electronics_ will work.

Jason2866, 2021-04-10:
It seems new crappy boards are out https://github.com/arendst/Tasmota/discussions/11615

d-a-v, 2021-04-10:
@Mathias2803
Can you provide a photo of your flash chip?
Can you use `esptool.py` to identify it ?

@Jason2866 
I can't seem to understand whether the issue you mention came to a conclusion ?
It seems it's not the flash chip since it has been replaced.
What version of the core is used in these tries ?

@Tech-TX 
The DIO configuration for d1 mini  [was made by](https://github.com/esp8266/Arduino/pull/1309/files#diff-c3a7c63d53cdec402cf559c1707ce5ef639a7e59e575f41ac5c8f5941e806a9fR596) @wemos 
If a change has to be made, what about
- adding the flash mode menu for wemos boards (with DIO by default)
- adding a new board labeled "wemos d1 mini clone" "wemos d1 mini pro clone" with DOUT

Jason2866, 2021-04-10:
@d-a-v  i dont think it is the flash chip. Imho the board design is crappy at all.
Core is 2.7.4.x (we added some backports)

Mathias2803, 2021-04-10:
Thanks for all your support. I will try all later. But here is the photo from my board as good as I can do it.

[image]
[image]

Mathias2803, 2021-04-10:
This is the chip I have ordered:
https://a.aliexpress.com/_msEoiOT

Tech-TX, 2021-04-10:
Hi, @d-a-v !  I'm willing to do anything that you gentlemen think will help, David.  I'll let you folks set the direction.   As I'd mentioned above, Wemos doesn't make the D1 Mini Pro any longer, and the authentic Lolin D1 Mini Pro are rare since they cost 2x what a clone board does.  Simpler than adding another board type is allowing the Flash mode to be selectable for the D1 Mini Pro with the initial default set to DOUT.  That will raise the fewest issues with "my board doesn't work..."  I have a couple of boards marked WEM*S (cheap clones), so the name on the board could easily be misleading to newbies.

I'd thought that the default Flash mode was changed to DOUT as it works with nearly all Flash brands, fewer questions & issues.  The only weird part I've heard of (haven't seen myself) was Macronix MX25L3205D which only works in DIO mode, not DOUT.  That Flash was on a clone D1 Mini that one person was having issues with.  The Winbond parts I have work in all 4 modes;  I bought 50 of their 4MB parts in case I needed to swap out cheap Flash. ;-)

I was wrong earlier, the [Lolin D1 Mini Pro](https://www.aliexpress.com/item/32724692514.html?spm=a2g0o.store_home.productList_2559240.pic_1) boards I have are v2.0.0, not v3.x.

@Mathias2803 Those photos look like a _really_ cheap clone.  I hadn't seen anyone use an unbranded ceramic antenna yet; all of the boards I'd seen use the [Rainsun antenna](https://lcsc.com/product-detail/Antennas_Rainsun-microwave-Tech-AN9520-245_C99665.html).  That likely means the _rest_ of the parts on the board are also floor sweepings.  The Flash chip has marking of a Micron 16MB Flash although the chip looks shiny, and the real Micron chips I've seen are a matte finish.  That's the hallmark of a rebranded IC, so no telling what the Flash truly is.  The LDO voltage regulator has the marking used by a Torex XC6102, a 300mA 3.3V regulator (a little underpowered for ESP8266) but again, there's no honor among thieves so it may be a cheap Chinese clone part that only runs 150mA.  If the LDO is underrated, there's a [quick low-power test](https://github.com/esp8266/Arduino/issues/7055#issuecomment-581183276) you can run that will boot with a 150mA regulator:

```cpp
#include <user_interface.h>

#define _R (uint32 *)PERIPHS_RTC_BASEADDR
#define valRTC 0xFE000000  // normal value during CONT (modem turned on)

RF_PRE_INIT() {
  *(_R + 4) = 0;  // value of RTC_COUNTER for wakeup from light/deep-sleep (sleep the modem)
  system_phy_set_powerup_option(2);  // stop the RFCAL at boot
  wifi_set_opmode_current(NULL_MODE);  // set Wi-Fi to unconfigured, don't save to flash
  wifi_fpm_set_sleep_type(MODEM_SLEEP_T);  // set the sleep type to Forced Modem Sleep
}

void preinit() {
  *(_R + 4) = 0;  // sleep the modem again after another initialize attempt before preinit()
  wifi_fpm_open();  // enable Forced Modem Sleep
  wifi_fpm_do_sleep(0xFFFFFFF);  // enter Modem Sleep mode
  // won't go into Forced Modem Sleep until it sees the delay() in setup()
}

void setup() {
  *(_R + 4) = valRTC;  // Force Modem Sleep engaged, set the modem back to 'normal' briefly
  delay(1);  // and sleep the modem at 18mA nominal substrate current
  Serial.begin(74880);  // bootloader baud rate so you can also see the boot message
  pinMode(LED_BUILTIN, OUTPUT);
}

void loop() {
  Serial.println("I'm alive!");
  digitalWrite(LED_BUILTIN, !digitalRead(LED_BUILTIN));
  delay(500);
}
```
You can either monitor the serial port at 74880 baud or watch the blinky LED to see that it booted correctly.

Jason2866, 2021-04-10:
My first action on all wemos mini xyz boards is replacing the onboard 3.3 vreg.
I never! received a board with a 500mA vreg.

Tech-TX, 2021-04-10:
I must have lucked out, compared to other people.  2 years ago I bought [10 D1 Mini clones](https://i.imgur.com/Zob5QuV.jpg) from [Alice](https://www.ebay.com/str/txhangelectronic), and they all worked perfectly.  I'd already seen the thread at Reddit about people getting 150mA LDOs, so I checked mine.  Hmmmm.  Marked 4B29, the only hits I got for that were for 2.9V LDOs, but these ran 3.29V.  Under load, they dropped down to 3.26V at 450mA, so I felt they were OK.  I bought 40 more immediately and got the same batch.  They have the DOIT ESP-12F modules, no WiFi problems whatsoever.  I generally cut the RF power by several dBmV as it overdrives my router.  I've since pulled the can on one of the DOIT modules and reverse-engineered most of the schematic, and it looks bog-standard.  I couldn't measure 3 of the caps (too low a value, and I lost one of 'em removing it) but the rest was average for the standard circuit.  They also go in & out of Deep Sleep without issues.  

What I see different between my D1 Minis and Mathias' D1 Mini Pro boards is that mine have 0805 10uF MLCC caps on the input and output of the LDO.  His _doesn't_.  0603 or smaller > 1uF caps are junk: you're lucky if you get 30% of the rated capacitance at 70% of the working voltage.  Whoever did my boards = reasonable, his manufacturer = idiot.

Mathias2803, 2021-04-10:
Thanks a lot for your feedback Tech-TX. I don't have a deep electronic knowledge, can you show me on a picture where I need to have a look on next time?
Where is the best place to order a better working one?
Thanks a lot, Mathias

Tech-TX, 2021-04-10:
Hi, Mathias!  The only D1 Mini Pro boards I can truly recommend are the real boards from Lolin that I linked above.  Everything else I've seen was poor quality design, poor parts, or both.  I have a couple of boards very much like yours that I bought [here](https://www.ebay.com/itm/D1-Mini-Pro-16-Bytes-Module-ESP8266-Series-WiFi-Wireless-Antenna-WeMos-Connector/222703207266) last year, although there's no guarantee that he still has the same quality boards.  The Flash chip on the ones I got is a 16MB Winbond, which is one of the better Chinese brands.  It has the same poor bypass around the LDO, though.  I had some 10uF 0805 caps, so I bodged them in place of the (2) 0603 bypass caps around the LDO.  It worked fine before I did the mod to the bypass caps, but I'm cautious with power.

Did the low-current blink sketch I listed above work for you?  That _only briefly_ needs a maximum of 50mA, the rest of the time it's running around 35mA (or less) substrate current for the ESP8266.  If your LDO is dropping out, that should boot.

Mathias2803, 2021-04-10:
Thanks d-a-v for his tip. I have changed to DIO and it is running now.

Thanks a lot!!

[image]

Mathias2803, 2021-04-10:
Hey Tech-Tx,
I guess you see that I could get my board running, BUT I don't wanna have a board with bad quality where I cannot relay on the functionality later. So my next order should be a better one.
Can you explain me the problem on my board?
And if I wanna order the "real" ones, how do I recognice them? Is this the correct link from you? It has a different boad size, is this correct? (https://www.aliexpress.com/item/32724692514.html?spm=a2g0o.store_home.productList_2559240.pic_1)
Thanks a lot for your support, kind regards Mathias

Mathias2803, 2021-04-10:
I also run your "Low-Power-Test" but I'm not sure if this helps anything to check the quality?

Here is the result:

[image]

Kind regards Mathias

Tech-TX, 2021-04-11:
Cool!  The problem was that your Flash wouldn't run in DIO mode. (**You nailed it, @d-a-v**)  It probably won't run in the Quad modes either.  At least it's not power related, if it runs a normal 'blink' sketch.  Mine is for a special case where you don't have enough power.

Yes, the Ali link is authentic Lolin boards.  They use Winbond Flash since it's very reliable.  They also use a 600mA LDO regulator, which is excellent.  Build quality like that is why they cost 3X what the cheap clones cost.  However, they *work*.  The ugly truth of clones is that you rarely get the same thing twice, and the next batch will invariably be worse than the last as they shave pennies off of the cost.  If you're only buying a few boards for learning, then I'd always recommend an authentic board and not a clone.  If you're buying tens of boards, it's harder to choose... I look VERY closely at the photos and then ask if the board is EXACTLY like the photo (not a representative sample).  I recognize good parts, though that's tough with _anything_ made in China.  Can I get a schematic for it?  I've worked in electronics & microprocessors since ~ 1976, so I have experience.  Recognizing good vs. poor components isn't something I can teach in a paragraph or three; the same goes for board layout, and rules of thumb for component selection.

Mathias2803, 2021-04-12:
Thanks a lot for your support.

Should I close the issue due to the fact that is is solved or is this done by an admin after there have been modifications in Arduino IDE?

Kind regards Mathias

d-a-v, 2021-04-15:
@devyte 
Do you think we should do something about the Arduino IDE and the default flash mode ?
Users may not be aware about official boards named `LOLIN`, good clones named `WEMOS`, and bad clone named whatever.
The board in the IDE is "`LOLIN (WEMOS)`", and it is not immediately obvious to the user whether a wrong choice has been made or by whom, and nowhere is advised that the generic board has to be selected and DOUT chosen as flash mode, at least for a try.
Before this was closed there were several choices:
- rename current `LOLIN (WEMOS)` to `LOLIN` using `DIO` and add `WEMOS(clone)` using `DOUT`,
- change `LOLIN (WEMOS)` from using `DIO` to `DOUT`,
- add the flash mode menu to `LOLIN(WEMOS)`.

devyte, 2021-04-15:
In general, my thinking is that there are too many variants out there combining weird flash chips under cloned names, and we can't individualize them all in the board selection.
Having said that, I like your idea: we could have "clone" boards that default tonthe slowest most compatible flash mode.

Reopening.

devyte, 2021-04-15:
About the options you listed:
> rename current LOLIN (WEMOS) to LOLIN using DIO and add WEMOS(clone) using DOUT

I don't think this is a good idea. The current board name was requested by someone at Lolin. 

> change LOLIN (WEMOS) from using DIO to DOUT

This would degrade all users of the real boards. 

> add the flash mode menu to LOLIN(WEMOS).

This sounds like a good idea.

I see the following possibilities:
- Simply add the flash mode menu to the LOLIN(Wemos) board
- Leave the current LOLIN(Wemos) board intact and add a new board LOLIN/Wemos clone that uses DOUT
- A combination of both: leave the current LOLIN intact, and add a new LOLIN/Wemos clone board that has the flash mode menu included, DOUT mode by default

PR #7989 (2021-05-15): boards: wemos d1 clones: new entry with menus for flash mode
Link to official boards is provided indicating that they are preferred.

fixes #7964 

*edit*: that is [3rd option (combination of both)](https://github.com/esp8266/Arduino/issues/7964#issuecomment-820443180)
closed 2021-05-15 by earlephilhower