esp8266/Arduino #7966: ESP8266httpUpdate FW+FS

ripper121, 2021-04-08:
Hi,
is it possible to do a WebUpdate via one Bin file which combines Firmware and Filesystem Bin?

I use the "Exported Sketch Binary" from Arduino and the "ESP LittleFS Data Upload" binary for the Filesystem.

So I get two files:
SketchName.ino.generic.bin (My Firmware)
SketchName.mklittlefs.bin (My Filesystem)

When I upload them trough the Arduino IDE everything works fine.
If I use the SketchName.ino.generic.bin for the ESP8266httpUpdate, the Spiffs get deleted.

My (Sonoff TH16) 1MB (8Mbit) Flash is parted to "FS:256KB OTA:~374KB"

With the "flash_download_tool_v3.8.5" I can combine two binary files into one.
How do I need to set the Flash Adresses to get one working FW Update Binary based on my settings?
Is it even possible to upload Sketch+Spiffs in one file?

d-a-v, 2021-04-08:
> Is it even possible to upload Sketch+Spiffs in one file?

It is a feature currently not provided by the esp8266 arduino core.
But yes that is technically possible.

ripper121, 2021-04-09:
You explained what I have already written...

Could you please say which start addresses I need to choose for my settings to get a working combined FW?

ripper121, 2021-04-09:
From the boards.txt for "1MB (FS:256KB OTA:~374KB)"

`generic.menu.eesz.1M256=1MB (FS:256KB OTA:~374KB)
generic.menu.eesz.1M256.build.flash_size=1M
generic.menu.eesz.1M256.build.flash_size_bytes=0x100000
generic.menu.eesz.1M256.build.flash_ld=eagle.flash.1m256.ld
generic.menu.eesz.1M256.build.spiffs_pagesize=256
generic.menu.eesz.1M256.upload.maximum_size=761840
generic.menu.eesz.1M256.build.rfcal_addr=0xFC000
generic.menu.eesz.1M256.build.spiffs_start=0xBB000
generic.menu.eesz.1M256.build.spiffs_end=0xFB000
generic.menu.eesz.1M256.build.spiffs_blocksize=4096`

For my SketchName.ino.generic.bin (368.640 Bytes) file I use Address: 0x00000
For my SketchName.mklittlefs.bin (262.144 Bytes) file I use Address: 0xBB000
When I Click on the CombineBin (Selected DoNotChgBin) it creates a target.bin (1.028.096 Bytes) with SUCCESS

Now I use the target.bin for the ESP8266httpUpdate and then it says: 
`Update: target.bin
sleep disable
LmacRxBlk:1
ERROR[4]: Not Enough Space
`

So what do I wrong?

ripper121, 2021-04-09:
Got it now working.
I used: "1MB (FS:192KB OTA:~406KB)"
For my SketchName.ino.generic.bin (368.640 Bytes) file I use Address: 0x00000
For my SketchName.mklittlefs.bin (196.608 Bytes) file I use Address: 0xCB000
Got a target.bin (1.028.096 Bytes) which I compressed with 7z to target.bin.gz  (348.160 Bytes)

The compressed version was now working with the ESP8266httpUpdate

closed 2021-04-09 by ripper121